SpenceKonde/megaTinyCore #131: UART communication issue on Attiny412 - maybe megaTinyCore retailed?

Remni1, 2019-12-28:
Hello, I have problem with communication between Attiny412 and PMS7003 using UART. There seems to be some synchronization problem. I wrote about it here: [https://www.avrfreaks.net/forum/attiny412-cant-communicate-correctly-uart-device-pms7003](url)
With new chip is seems to work once in 20 trials, but only use 20MHz or 10MHz frequency. Could it be related to megaTinyCore UART implementation? On other chips, for example ESP8266, it seem to works perfectly.

Remni1, 2019-12-29:
I think that this line in UART.cpp can cause problems:
`baud_setting /= (1024 - abs(sigrow_val));`
User "curtvm" on avrfreaks forum calculated that this line results in wrong baud register value, and the correct one should be:
`baud_setting /= 1024;`
After chaining this line my problem seems to be solved.

SpenceKonde, 2019-12-29:
Interesting - thanks for this.

There is also a closely related issue in that regardless of whether the
16mhz or 20mhz oscillator is in use, and regardless of the voltage, it
reads the 16mhz 5v correction from the sigrow.

commit c2c406ba (2019-12-30): I think that this fixes #131
closed 2019-12-30 by SpenceKonde