SpenceKonde/megaTinyCore #136: TCA0 high frequency output (500khz)

inf1981, 2020-01-16:
I was very happy to find the new configuration option for millis() to use another timer. I'm currently trying to get a 500khz signal output on PA5. The only problem is that I’m not sure how to use TCA0 correctly, so some help is appreciated. I configured the timer and the maximum I can get is around 320khz if the main clock is at 20 Mhz. Code looks like this:

```c++
void setup() {
  PORTA.DIRSET = PIN5_bm; // use PA5 as an output
  PORTA.OUT &=~PIN5_bm; // write PA5 low

  TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV2_gc    /* System Clock / 2 */
                      | 1 << TCA_SINGLE_ENABLE_bp; /* Module Enable: enabled */
  TCA0.SINGLE.PER = 0;
  TCA0.SINGLE.CTRLA |= TCB_ENABLE_bm;
  bitSet(TCA0.SINGLE.INTCTRL, 0);

}

ISR(TCA0_OVF_vect) {
  PORTA.OUTTGL = PIN5_bm; //toggle PA5
}

void loop() {}
```

I did something similar on an Attiny84, but somehow it worked. Unfortunately, I can't find my old code. Any ideas? I guess the toggle is maybe too slow...

SpenceKonde, 2020-01-16:
Dont try to generate pwm with an isr. Do it by using output compare, like
hardware pwm. I would set divider to 1, period to 39 (so 40 clocks),
compare to 19 (so 20 clocks, 50%duty cycle with TCA0.PER=39), single slope
mode. Will take some meditation on the TCA chapter; while the megaavr
peripherals are way better than the classic ones, the writing quality in
the datasheets leave something to be desired.

 My instinct is that the datasheet js now written by tech writers who dont
have a deep understanding of the hardware (if you work as a programmer at a
largish company, and have worked with the documentation people on the
customer facing docs, you probably know what I mean, this phenomenon is
ubiquitous in tech industry). Be glad you are using TCA, not TCD - both are
poorly documented, but TCA is at least simple once you understand it.

The core configures TCA0 for split mode to get 6 pwm channels; in general
when you take over a timer, since the core configures the timer for built
in functionality, you cant trust that the relevant registers will be in
their power on reset state - set all the registers explicitly to the
desired values in setup, ie, with =, not |= and &=. Once you've set a
register to a known state (probably in setup), assuming you dont call any
core functions that change them, you can use |= and &=.

The code you used for the t84 is of very limited use, the timers on the
megaavr parts are very different from classic avrs. Be aware that changing
the divider on TCA will hose tone and serve, because the timer input to TCB
is prescaled by the same factor, but the core assumes it has not been
changed by the sketch (we cant spare the flash for this on small parts).

Hope this was helpful. I do not have time to provide hands on help with
writing the code - am hopelessly behind on dozens of urgent issues.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

inf1981, 2020-01-17:
Thank you for giving me some hints. I had trouble with the timers before, that’s why I had this hack with an ISR. So I followed your advice but couldn’t get it working, till I discovered in the datasheet of the Attiny 3217 that the PA5 can only be used in split mode for waveform output: “Bit 5 – Write this bit to '1' to select alternative output pin for TCA0 waveform output 5 in Split mode. Not applicable when TCA in normal mode.”.
So, after finding this information I just configured the timer in split mode with this code:
```c++
void setup() { 
  TCA0.SPLIT.CTRLD = TCA_SPLIT_SPLITM_bm;  // Enable split mode
  TCA0.SPLIT.CTRLB = TCA_SPLIT_HCMP2EN_bm; // Enable all comparators in split mode
  TCA0.SPLIT.HPER = 39; // Set the period
  PORTA.DIRSET = 1 << 5; //PA5
  TCA0.SPLIT.CTRLA =
    TCA_SPLIT_CLKSEL_DIV1_gc | // divide sys. clock by 1
    TCA_SPLIT_ENABLE_bm;
   TCA0.SPLIT.HCMP2 = 20; // W05 / PA5
}

void loop() {}
```
And now it works like expected (It is not at exactly 500khz, my scope shows 497.5khz, but that is fine for me). Thank you, a lot, for your help / hints, that was what I needed. So, if anyone is trying to do the same or similar, above code works. You can also add more outputs in split mode.

closed 2020-01-17 by inf1981

SpenceKonde, 2020-01-17:
Great! Glad you got it sorted! 

A 0.5% error is well within expected process variation on the internal oscillator, so that is not surprising.