SpenceKonde/megaTinyCore #140: Possibly broken library with recent update?

veganism, 2020-02-11:
[RDA5807M](https://github.com/csdexter/RDA5807M)

Using the Arduino IDE version 1.8.10.

Installed megaTinyCore from the board manager.

**1.1.4** compiles the sketch and operates fine on my attiny402 project.

**1.1.5** compiles fine but then the MCU hangs on the first i2c operation `radio.begin(RDA5807M_BAND_WEST);`.

Is this one of those edge-cases that is mentioned in issue #130? I'm aware that not everyone has the capabilities to test this particular library and hardware/software setup, so I just figured I would report the issue and see if it's something plainly obvious that I'm missing. Any info would be appreciated.

SpenceKonde, 2020-02-13:
I don't see anything in there that would suggest a problem related to #130 (my understanding is that it would be a compile error in that case).

One way to be certain would be to simply undo that change, and see if it fixes it - it's this one: https://github.com/SpenceKonde/megaTinyCore/pull/130/commits/f6e6c15055730590abcc6f5acb218574f499821c (ignore the subsequent mess of changes

If you installed from board manager (if a github install, I am sure you don't need me to tell you where it is) , it will be under (assuming windows): C:/Users/yourusername/Arduino15/Local/packages/megaTinyCore/hardware/megaavr/1.1.5/libraries/Wire

veganism, 2020-02-13:
Thanks for the idea. I attempted with a fresh repo pull and unfortunately it was hanging the same as before. Then I reset to before the commit you mentioned and it was working again - after I commented out TWI_SlaveInit from `Wire.cpp` as otherwise the sketch would be too big for the 4KB of space. _( I don't believe this has any relevance, however )_

I actually stumbled across something interesting while messing with all this, that might shed some more light on the issue. I was using 1MHz clock speed before - my project is battery powered, and the 402 is in sleep modes 99% of the time, so I figured the lowest clock speed would be fine. I bumped it up to 8MHz and tried again and surprisingly it worked! So the <=1.1.4 core "allowed" i2c to operate from 1MHz, while the >=1.1.5 core requires >1MHz. So for now, I'm just keeping the project with 4MHz and everything is gravy, plus the new core saves a lot of space which is greatly appreciated. I guess the mystery is solved for now, as long as the whole "Wire freaks out on 1MHz" thing is to be expected.

veganism, 2020-02-14:
The different behavior happens exactly at https://github.com/SpenceKonde/megaTinyCore/commit/690d36b0ea3e52badb248f43a710619bec449f61

It appears that I'm probably doing something incredibly stupid. Changing `Stream` back to `HardwareI2C` makes everything work again at 1MHz with the latest git version. Without knowing a whole lot about it, I'd guess it's a limitation (right?)

SpenceKonde, 2020-02-18:
I am baffled as to how this has chaned behavior here - it absolutely should not have caused a change in behavior that ONLY occurs at low clock speed. Suspect this is some weird timing issue that only manifests at the low clock speed, and somehow changing HardwareI2C to stream had an effect on execution time that, at 1MHz was enough to change behavior of that device.

I will check on a basic I2C device and make sure I don't see the same thing.

SpenceKonde, 2020-03-12:
After a long, non-specific illness that has finally lifted from me, I'm trying to again tackle these issues. I realized those boards were super cheap, so I ordered a 5-pack of them, though it'll be a while before I get them (though shipping out of china has been really picking up again, and I think due to smaller volumes, it's getting here much faster than they've been estimating). Under $3 for 5 small, easy to store parts seems pretty good to me.

veganism, 2020-03-12:
Yep, it is a super cheap and pretty well-featured FM radio module. It only lacks documentation and i2s audio output (as far as I can tell anyways). 

I'm still thinking this issue is related to my hardware, because I pieced this project together as one of my first prototypes before I knew better. I've spent the past month learning how to use C and avr-gcc directly, but I haven't gotten around to experimenting with I2C yet. Seems a little daunting, especially because I don't have an oscilloscope. Just the other day, I had an aha moment when I got CCL to work and understood something of it.

I've been planning to figure out I2C when I piece together an improved version of my radio, when I get some free time of my own. Then I might understand more about what's going on here, at least. For now, I'm pretty sure it's on my end so wouldn't be surprised at all if everything works fine for you.

SpenceKonde, 2020-10-03:
No comment; please reopen if additional information is available.

closed 2020-10-03 by SpenceKonde