SpenceKonde/megaTinyCore #141: Disable DAC output -> File not found

LordJakson, 2020-02-12:
Hi
I just installed the megaTinyCore the first time.

I noticed if I disable the DAC output in the Arduino/Tools Menu the compiler print out a Error:

`avr-g++: error: {build.dacvref}: No such file or directory`

Is this normal or did I something wrong?

Arduino 1.8.10, megaTinyCore 1.1.5

SpenceKonde, 2020-02-12:
Thanks, that would be a bug

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-02-13:
Was broken in 14-pin non-optiboot board def (and menu option was missing for 20-pin part optiboot def).

commit 756a73ec (2020-02-13): Fix #141

Correct copy-paste errors in boards.txt
closed 2020-02-13 by SpenceKonde