SpenceKonde/megaTinyCore #143: High sleep mode current draw?

JasXSL, 2020-02-14:
I'm testing on Tiny804 and it's drawing more than 1mA in sleep mode with the following code at 5V. Any idea why it's drawing so much?

Wiring is pretty simple. 0.1uF cap between vcc and gnd. An LED on IO5 with a 1k resistor, and that's it.

```
#include <avr/sleep.h>
void setup() {
	set_sleep_mode(SLEEP_MODE_PWR_DOWN);  /* Set sleep mode to POWER DOWN mode */
	sleep_enable();                       /* Enable sleep mode, but not going to sleep yet */

	pinMode(5, OUTPUT);
	digitalWrite(5, LOW);
	delay(100);
	digitalWrite(5, HIGH);

	sleep_cpu();
}

void loop() { 
  	digitalWrite(5, LOW);              /* This is never reached */
}
```

Board settings:

- BOD disabled
- EEPROM Retained
- Clock speed 10mhz
- DAC vref 0.55V
- TinyNeoPixel: Porta
- millis enabled
- Voltage for uart: 5v
- Uart: tx4, rx5
- i2c 8/9
- Bootloader burned

JasXSL, 2020-02-14:
Looks like the issue is with the ADC, setting all pins to outputs uses significantly lower current. In the previous tiny models you could do `ADCSRA &= ~ bit(ADEN)` to turn it on/off entirely. Any idea how that's done with the new library?

SpenceKonde, 2020-02-18:
Try:

ADC.CTRLA&=~ADC_ENABLE_bm; 

Does that do anything to power consumption while sleeping?

That's the analogous setting to enable/disable the ADC. There may be other peripherals that will need to be shut down too - I haven't looked into what does and doesn't make a difference. 

ADC0.CTRLA |= ADC_ENABLE_bm;

will turn it back on. 

Also - I assume you are testing without any external parts that could be burning power? (regulators for example will waste power, even if their input is not connected and the chip is powered through other source)

veganism, 2020-02-19:
@SpenceKonde can the core differ from the datasheet on what's available in different power modes?

JasXSL, 2020-02-19:
No difference, each input adds a substantial amount of current draw. At 5 inputs active it settles at about 500uA. Vs setting all outputs making current draw so low my instruments are having a difficulty detecting it.

SpenceKonde, 2020-05-11:
Saaaaay.... were you by any chance testing this without a full power cycle between when you programmed it using jtag2updi, and when you tested the current draw? 

Because I'm currently mucking around with the jtag2updi source code, and it never actually shut down the UPDI interface, so - by my reading of the datasheet - it would keep the oscillator from ever turning off in sleep mode until a full powercycle (ie, disconnect from power and wait a minute or two, or (with power disconnected) short power+ground briefly to discharge the decoupling caps on the board)... I'll be releasing a better version of jtag2updi Real Soon Now that fixes things like this...

JasXSL, 2020-05-31:
even with a full power cycle I'm still seeing 1mA current draw on a bare 1604, gonna experiment some more and see if I can figure something out

DaleMitchell, 2021-01-26:
@JasXSL high current consumption in sleep mode is expected if any pins are floated. Yes, the difference is between microamps to milliamps. Internal pull-ups for any pins floating should be sufficient. I've found external pull-down resistors to be slightly more efficient by nanoamps.

SpenceKonde, 2021-01-27:
Interesting - @DaleMitchell that would mean that these parts deal several orders of magnitude more poorly with pins left floating than the classic AVRs! Correct or am I misunderstanding?

DaleMitchell, 2021-01-31:
I was able to get current consumption closer to 100 nA by connecting an LED's anode to an IO pin (1k resistor connected to LED cathode and ground), and also a 100k pull-down resistor connected to the IO pin.

```
#include <avr/sleep.h>
void setup() {

	// set all unused pins to INPUT_PULLUP, or
	// use external pull-up or pull-down resistors
	pinMode(5, OUTPUT);
	digitalWrite(5, HIGH);
	delay(100);
	digitalWrite(5, LOW);
	pinMode(5, INPUT);

	power_all_disable();
	set_sleep_mode(SLEEP_MODE_PWR_DOWN);
	sleep_enable();
	sleep_cpu();
}

void loop() { 
  	digitalWrite(5, LOW);              /* This is never reached */
}
```

@SpenceKonde - excessive current consumption in sleep mode due to floating pins is about equal between the new AVRs and classic AVR. Obviously, the more pins that are left floating, the higher the current draw.

SpenceKonde, 2021-01-31:
Thanks - since I think we've figured out the key factor, at long last, I'm going to close this one.

closed 2021-01-31 by SpenceKonde