SpenceKonde/megaTinyCore #145: Example of using 16-bit PWM using analogWrite()

Nbelles, 2020-02-21:
Hello,

I am very new to the MegaTinyCore community as I just purchased a lot of ATtiny’s. I have been going through the list of features listed in the main welcome page and one of those is the PWM bit rates. 

I have spent some time trying to figure out how to use 16-bit PWM but to no avail. If someone has the time, I would love to see an example of how to switch between the different timers in Arduino to change the range of analogWrite.

Thanks!

SpenceKonde, 2020-02-24:
It is worth noting that analogWrite can *never* be used for 16-bit PWhere, on any board.

If you need 16-bit PWM, you would need to:

Set millis timer to use TCD (in next release the dependence of TCB's on the TCA prescaling for millis/micros/Servo/Tone will be removed) 

Set timer A to not operate in split mode (hence 3 pwm channels instead of 6)

Configure timer A as you desire for the frequency/etc.

Write your own analogWrite16Bit() function for setting the duty cycle. 

The whole thing is similar to taking over Timer1 on the classic AVRs, just with a slightly more complicated peripheral.

SpenceKonde, 2020-02-24:
I do think an example of this would be a useful thing for people to see though; I have a lot on my plate; if you haven't done it by the time I have a chance, I will do it and put the code in the examples.

flodmotorgrodan, 2020-03-10:
**Strange 16-bit PWM problem using ATtiny402.**

I am trying to configure simple 16-bit single slope PWM with adjustable TOP and CMP values. 
[ATtiny202/402 manual "Single-Slope PWM Generation" page 192](http://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny202-402-AVR-MCU-with-Core-Independent-Peripherals_and-picoPower-40001969A.pdf#page=192)  

I have moved millis() compiler option to TCB. And I can see the millis() works fine and is affected only by prescaler setting. A prescale of :64 seems to generate accurate ms counts.

The strange thing is the PWM TCA0 behaves like it counts 8-bit.

With a prescale of :1024 and 0xFFFF as top the lowest output frequency is 1.28kHz. 

Writing to low byte of the compare register TCA0.SINGLE.CMP0L has no effect on the PWM output. Output changes only when TCA0.SINGLE.CMP0H is written.

How can it be it counts 8-bit when setting PWM mode using this code:

	TCA0.SINGLE.CTRLB = 0 << TCA_SINGLE_ALUPD_bp       /* Auto Lock Update: disabled */
	| 1 << TCA_SINGLE_CMP0EN_bp    /* Compare 0 Enable: enabled */
	| 0 << TCA_SINGLE_CMP1EN_bp    /* Compare 1 Enable: disabled */
	| 0 << TCA_SINGLE_CMP2EN_bp    /* Compare 2 Enable: disabled */
	| TCA_SINGLE_WGMODE_NORMAL_gc; /* Set PWM normal Mode (WGMODE[000])  */

PWM turns on/off with the _1 << TCA_SINGLE_CMP0EN__bp above.

Writing 16-bit PWM _Periode_ and _Top_ like this:

	TCA0.SINGLE.PER = 0xFFFF;
	TCA0.SINGLE.CMP0 = 0x01FF;
 
, but the lower byte is kind of ignored??

Using the buffered registers seems kind of irresponsive and only direct write works:

	TCA0.SINGLE.PERBUF = 0xFFFF;
	TCA0.SINGLE.CMP0BUF = 0x01FF;

PWM output appears on pin PA3

Looks like it is in 8-bit split mode but how?

flodmotorgrodan, 2020-03-11:
This code is auto-generated using the excellent on-line software tool [Start.ATMEL](https://start.atmel.com/) 

Does not help. PWM still counts only high byte of TCA0.SINGLE.PER and TCA0.SINGLE.CMP0

When PER and CMP0 set in small range 0..FF no PWM is output. Puzzling!

```
int8_t PWM_0_init()
{

	TCA0.SINGLE.CMP0 = 0x00f0; /* Setting: 0x00f0 */

	TCA0.SINGLE.CTRLB = 0 << TCA_SINGLE_ALUPD_bp            /* Auto Lock Update: disabled */
	                    | 1 << TCA_SINGLE_CMP0EN_bp         /* Setting: enabled */
	                    | 0 << TCA_SINGLE_CMP1EN_bp         /* Setting: disabled */
	                    | 0 << TCA_SINGLE_CMP2EN_bp         /* Setting: disabled */
	                    | TCA_SINGLE_WGMODE_SINGLESLOPE_gc; /*  */

	TCA0.SINGLE.CTRLC = 1 << TCA_SINGLE_CMP0OV_bp    /* Setting: enabled */
	                    | 0 << TCA_SINGLE_CMP1OV_bp  /* Setting: disabled */
	                    | 0 << TCA_SINGLE_CMP2OV_bp; /* Setting: disabled */

	TCA0.SINGLE.PER = 0x00ff; /* Top Value: 0x00ff */

	TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV1_gc /* System Clock */
	                    | 1 << TCA_SINGLE_ENABLE_bp /* Module Enable: enabled */;

	return 0;
}
```

SpenceKonde, 2020-03-11:
During initialization (the init() function, which is called internally before setup()) the core configures TCA0 to split mode in order to support 8-bit PWM via analogWrite() on 6 pins - so you need to use single mode, you need to be sure to turn off split mode first ( TCA0.CTRLD=0; ). I thought I mentioned that TCA0 is configured in split mode in the section on timer usage; will review docs and make sure of it.

SpenceKonde, 2020-03-11:
It's also worth noting that this applies to any peripheral used by the core for anything. Don't assume that the core hasn't already set up a given peripheral so that calls to functions that manipulate it within the sketch can be efficient, and so that the code for these calls can be made efficient. If your initial attempts at configuring a peripheral don't behave as expected, print out all the registers associated with that peripheral - in a few cases, even printing them after you think you've set them turns out to reveal a problem, because you might have missed that a certain register requires a protected write or cannot be written under certain conditions (unlike classic AVRs, the megaavr parts have a bunch of registers with such restrictions).

flodmotorgrodan, 2020-03-11:
@SpenceKonde ,

Thank you for this great library. The documentation is excellent. 

Writing `TCA0.SINGLE.CTRLD = 0;` to revert from split mode did the trick!

Here is the code used from setup() to finally make TCA 16-bit PWM work on ATtiny402 using output PA7 configured for output.

In the example 20.2 MHz/64/65535 resulted in 4.8Hz 50% duty-cycle 

```
//PWM timer TCA 16-bit - Requres millis() on TCB or other timer
int8_t PWM_A_init()
{
	//Timer TCA will be taken over and used for 16-bit PWM 
	//Read about Arduino PWM functions affected:
	//https://github.com/SpenceKonde/megaTinyCore#pwm-support
	//For millis(), micros() options and timer use please read: 
	//https://github.com/SpenceKonde/megaTinyCore#millismicros-options-and-timer-use
	
	//Reverting from Arduino core TCA timer split mode
	TCA0.SINGLE.CTRLD = 0;

	TCA0.SINGLE.CTRLB = 1 << TCA_SINGLE_ALUPD_bp            /* Auto Lock Update: disabled */
	| 1 << TCA_SINGLE_CMP0EN_bp         /* Setting: enabled */
	| 0 << TCA_SINGLE_CMP1EN_bp         /* Setting: disabled */
	| 0 << TCA_SINGLE_CMP2EN_bp         /* Setting: disabled */
	| TCA_SINGLE_WGMODE_SINGLESLOPE_gc; /*  */

	//PWM output pins override - disable
	TCA0.SINGLE.CTRLC = 0 << TCA_SINGLE_CMP0OV_bp    /* Setting: enabled */
	| 0 << TCA_SINGLE_CMP1OV_bp  /* Setting: disabled */
	| 0 << TCA_SINGLE_CMP2OV_bp; /* Setting: disabled */
	
	 TCA0.SINGLE.INTCTRL = 0 << TCA_SINGLE_CMP0_bp /* Compare 0 Interrupt: disabled */
	| 0 << TCA_SINGLE_CMP1_bp /* Compare 1 Interrupt: disabled */
	| 0 << TCA_SINGLE_CMP2_bp /* Compare 2 Interrupt: disabled */
	| 0 << TCA_SINGLE_OVF_bp; /* Overflow Interrupt Enable: disabled */

	//Set max periode time and 50% duty cycle
	TCA0.SINGLE.PERBUF = 0xffff; /* Top Value */
	TCA0.SINGLE.CMP0BUF = 0x7fff; /* Periode Value */		
		

	TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV64_gc | 1 << TCA_SINGLE_ENABLE_bp /* Timer Module Enable */;

	return 0;
}
```

As You mentioned users have to read the documentation:
[#pwm-support](https://github.com/SpenceKonde/megaTinyCore#pwm-support)
[#millismicros-options-and-timer-use](https://github.com/SpenceKonde/megaTinyCore#millismicros-options-and-timer-use)

SpenceKonde, 2020-03-19:
I have added a guide, because so many people were asking me about this.

https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/TakingOverTCA0.md

closed 2020-03-19 by SpenceKonde

elektroboard, 2023-07-07:
> @SpenceKonde ,
> 
> Thank you for this great library. The documentation is excellent.
> 
> Writing `TCA0.SINGLE.CTRLD = 0;` to revert from split mode did the trick!
> 
> Here is the code used from setup() to finally make TCA 16-bit PWM work on ATtiny402 using output PA7 configured for output.
> 
> In the example 20.2 MHz/64/65535 resulted in 4.8Hz 50% duty-cycle
> 
> ```
> //PWM timer TCA 16-bit - Requires millis() on TCB or other timer
> int8_t PWM_A_init()
> {
> 	//Timer TCA will be taken over and used for 16-bit PWM 
> 	//Read about Arduino PWM functions affected:
> 	//https://github.com/SpenceKonde/megaTinyCore#pwm-support
> 	//For millis(), micros() options and timer use please read: 
> 	//https://github.com/SpenceKonde/megaTinyCore#millismicros-options-and-timer-use
> 	
> 	//Reverting from Arduino core TCA timer split mode
> 	TCA0.SINGLE.CTRLD = 0;
> 
> 	TCA0.SINGLE.CTRLB = 1 << TCA_SINGLE_ALUPD_bp            /* Auto Lock Update: disabled */
> 	| 1 << TCA_SINGLE_CMP0EN_bp         /* Setting: enabled */
> 	| 0 << TCA_SINGLE_CMP1EN_bp         /* Setting: disabled */
> 	| 0 << TCA_SINGLE_CMP2EN_bp         /* Setting: disabled */
> 	| TCA_SINGLE_WGMODE_SINGLESLOPE_gc; /*  */
> 
> 	//PWM output pins override - disable
> 	TCA0.SINGLE.CTRLC = 0 << TCA_SINGLE_CMP0OV_bp    /* Setting: enabled */
> 	| 0 << TCA_SINGLE_CMP1OV_bp  /* Setting: disabled */
> 	| 0 << TCA_SINGLE_CMP2OV_bp; /* Setting: disabled */
> 	
> 	 TCA0.SINGLE.INTCTRL = 0 << TCA_SINGLE_CMP0_bp /* Compare 0 Interrupt: disabled */
> 	| 0 << TCA_SINGLE_CMP1_bp /* Compare 1 Interrupt: disabled */
> 	| 0 << TCA_SINGLE_CMP2_bp /* Compare 2 Interrupt: disabled */
> 	| 0 << TCA_SINGLE_OVF_bp; /* Overflow Interrupt Enable: disabled */
> 
> 
> 	//Set max periode time and 50% duty cycle
> 	TCA0.SINGLE.PERBUF = 0xffff; /* Top Value */
> 	TCA0.SINGLE.CMP0BUF = 0x7fff; /* Periode Value */		
> 		
> 
> 	TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV64_gc | 1 << TCA_SINGLE_ENABLE_bp /* Timer Module Enable */;
> 
> 	return 0;
> }
> ```
> 
> As You mentioned users have to read the documentation: [#pwm-support](https://github.com/SpenceKonde/megaTinyCore#pwm-support) [#millismicros-options-and-timer-use](https://github.com/SpenceKonde/megaTinyCore#millismicros-options-and-timer-use)

Hi. 
I use Attiny3226. I try your codes above. The PWM output comes from the PB0 pin. But I want to use PA4 for PWM output. 
What should I do for it.
Thanks.

SpenceKonde, 2023-07-07:
From the tools -> PWM pins menu, choose an option that includes PA4./

elektroboard, 2023-07-07:
Where is this option. I don't find
[image]

SpenceKonde, 2023-07-07:
Should be in latest released version

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

elektroboard, 2023-07-07:
> Should be in latest released version
> […](#)
> ____________ Spence Konde Azzy’S Electronics New products! Check them out at tindie.com/stores/DrAzzy GitHub: github.com/SpenceKonde ATTinyCore: Arduino support for almost every ATTiny microcontroller Contact: ***@***.***

I use latest versions 2.1.1 for Arduino IDE and 2.6.4 for MegaTinyCore.

SpenceKonde, 2023-07-07:
Yes, it was added in 2.6.8, the most recent version, that option wasn't exposed in the older version.

elektroboard, 2023-07-08:
Please share image of menu for pen