SpenceKonde/megaTinyCore #147: Problems compiling a deep_sleep program when adding SoftwareSerial

saysphilippe, 2020-02-23:
Hi!,
I'm trying to build a solution that wakes up by an interrupt, and sends a radio signal, and then goes back to sleep again, until next interrupt. When running the sleep sketch alone without SoftwareSerial, it works fine, and running the sending sketch alone works fine as well, but when adding the SoftwareSerial lib into the deep sleep sketch it won't compile. 
[1614_deep_sleep_HC-12.ino.txt](https://github.com/SpenceKonde/megaTinyCore/files/4241277/1614_deep_sleep_HC-12.ino.txt)

[image]

SpenceKonde, 2020-02-23:
This is correct and expected behavior. SoftwareSerial sets up interrupt handlers for all ports (it uses the pin change interrupts to detect the incoming serial data (since it can be used on any pin, it has to set up ISRs for all of them)), but you are also defining your own handler for the same interrupt. Thus, the compiler reports that you have two interrupt handlers defined for the same interrupt. This is one of the problems with the SoftwareSerial library - it grabs all the PORT interrupts (also known as PCINTs on classic AVRs, where there are two types of pin interrupts - megaavr's don't have separate INT0/INT1/etc interrupts), so that it can work on any pin. There's no mechanism for the library to recognize which port the pin you're using for RX is in and only set up the ISR for that port.

This limitation is present on official boards too, both classic and megaavr (though more painful for megaavr, as classic avr has the INTn interrupts in addition to the PCINTs) It is one of the many reasons that SoftwareSerial sucks, (and also the reason why I have that weird TinySoftSerial on ATTinyCore which uses a fixed pin and a different interrupt for chips without a hardware serial port - losing all the PCINTs to get a serial port is a very steep price to pay!)

Are you actually using hardware serial port (Serial) and still need another one, hence SoftwareSerial?

If you do actually need a second serial port, you could make a copy of SoftwareSerial library, rename it (both the folder and the SoftwareSerial.* files themselves), and in your modified copy, comment out the ISRs for all ports except the one with the pin you want to use, and then use the modified version of the library.

closed 2020-02-23 by SpenceKonde

saysphilippe, 2020-02-23:
Hi, thanks for a quick and good response. It sounds like that would work as a solution yes, to copy the serial and comment out all other ports.

SpenceKonde, 2020-02-23:
Glad that helped!

Of course, another option is just to use a chip with more hardware serial
ports like the ATtiny1634 or ATmega328PB... I absolutely despise software
serial ports (wanting a small, cheap AVR with two serial ports is actually
why I first started working on my cores for ATTiny parts!), though neither
of those is a megaavr.

saysphilippe, 2020-02-23:
I completely agree, what I really love about the Attiny series is the power consumption, so this is only a snag, when I have good support from you guys! I finally broke another problem I have had with the freaking ESP8266/32 with SoftwareSerial as well. You need to use the espsoftwareserial library instead. I'm trying to use radio as much as possible where I don't have wifi access, range or because of power consumption, and you would like to receive it and communicate by wifi where you have steady power.

saysphilippe, 2020-02-23:
Or, another simple solution is to use the http://gammon.com.au/Arduino/SendOnlySoftwareSerial.zip, https://forum.arduino.cc/index.php?topic=112013.0 which works perfect!