SpenceKonde/megaTinyCore #153: Support:  Cannot get Serial to emit bytes.

robshep, 2020-03-20:
Hi,

I'm not sure of where to reach out with support-type questions. please feel to delete and redirect elsewhere.

I can't get `Serial` to emit bytes.  
I've only done simple ops _like_ this, using pin2-TX and pin3-RX ...

```
void setup() { Serial.begin(); }
void loop(){ Serial.println("Hello etc"); }
```

...but I can't see the bytes coming out of the serial line.  
I'm wondering if there are any parameters to change (timers etc) that may affect this?

* I haven't adjusted any of the sub-menu params from _default_
* I've tried `flush`, and different baud rates.
* The FTDI USB adapter is known working - I just swapped the RX/TX/GND lines from a Nano which emits fine using same sketch. 

Any pointers or things to try are gratefully RX.

SpenceKonde, 2020-03-20:
This code:

```
void setup() { Serial.begin(); }
void loop(){ Serial.println("Hello etc"); }

```
Does not even compile for me, because it is missing the baud rate that needs to be passed to Serial.begin()  - same as a normal Arduino... You're saying that compiles for you? o_o

Assuming you are actually passing a baud rate to Serial.begin and just forgot to include it on your post, what version of the Arduino IDE, and what version of the core?

robshep, 2020-03-20:
Yes, I meant to illustrate just "typical" basic serial output - nothing fancy. 
IDE: 1.8.12
megaTinyCore: 1.1.6
Thanks

This is my actual code, just more cruft than I wanted to express.

```
#define LED PIN_PA3

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED, OUTPUT);
  Serial.begin(9600);                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                    
  while(!Serial){
    digitalWrite(LED, HIGH);   // turn the LED on (HIGH is the voltage level)
    delay(75);                       
    digitalWrite(LED, LOW);    // turn the LED off by making the voltage LOW
    delay(75);                       
  }

  digitalWrite(LED, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(5000);                       
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.println("on");
  Serial.flush();
  delay(1000);                       
  digitalWrite(LED, LOW);    // turn the LED off by making the voltage LOW
  Serial.println("off");
  Serial.flush();
  delay(1000);                       
}
```

SpenceKonde, 2020-03-20:
That code works for me!

Are you looking at the right pins?  
Every time a pin is referred to by a number, that is the logical or "Arduino" pin number, not the physical pin number from counting pins on the package.. 

From your saying pins 2 and 3, it sounds like one of the 8-pin parts; refer to the pinout chart here:
https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/ATtiny_x12.md

TX of your adapter should connect to RX of tiny (ie, logical pin 3, PA1), RX of adapter to TX of tiny (logical pin 2, PA2), and grounds should be connected. 

There is also a tools submenu to select which pair of pins are used for Serial - make sure you've got 2 and 3 selected there.

And the LED is blinking right?

robshep, 2020-03-20:
Thanks for the support.
I should have mentioned it's a 412.

OK, I have using TXD/PA6/HW2  (connected to FTDI/RX) 
and RXD/PA7/HW3 (conneted to FTDI/TX)

I'd assumed that TXD&RXD were the hardware serial ports and (on the dia) TXD^1 & RXD^1 were the optional swaps, I'm sorry, I must have discounted ^1 options and did not think to eval this.

I'll try this wiring later and report back.

What is RXD and TXD (without ^1) on the diagram?

Thanks for the support

Rob

SpenceKonde, 2020-03-20:
The default for the *part* PA6, PA7 (Arduino pins 0 and 1, physical pins 2 and 3). The default for the *core* is PA1, PA2 (Arduino pins 2 and 3, physical pins 4 and 5). I know there was some historical reason for this, and I was annoyed about it afterwards (but by then, I had piles of boards with the FTDI header wired to the alt pins!) - I forget exactly what. I think it had something to do with the original garbage datasheet they released. 

When the megaavr tinies were first shipping, the preliminary datasheets they paired them with were full of inaccuracies (among other things, they only listed one PWM-capable pin on the 8-pin parts...). They also shipped a considerable quantity of 402's with the wrong signature bytes... The product launch was less than graceful, probably because it happened right as microchip was taking them over - particularly for the 8-pin parts.

The pins in use are shown in the tools submenu. In the next release, I will add the pin in Pxn notation next to the pin numbers.

SpenceKonde, 2020-03-20:
(note - that change to the menus will only be for the 8-pin parts, because the way that alt/standard options, and the physical and logical pin numbers line up conspires to make it particularly confusing there. Deploying it to all parts and all remappable interfaces would clutter the menu and hurt readability.

closed 2020-03-20 by SpenceKonde

robshep, 2020-03-20:
Works. Thanks again.

SpenceKonde, 2020-04-16:
FYI - 2.0.0 changes behavior here: The menu option is removed entirely, and you now use Serial.pins() or Serial.swap() to do pin-swapping (and the default behavior on the 8-pin parts no longer swaps the pins on Serial by default)