SpenceKonde/megaTinyCore #154: settings millis()/micros() is unclear for me

avandalen, 2020-03-20:
Hi Spence,
The setting millis()/micros () is a long complicated story. To keep it simple, just tell what to do for compatibility, so that old sketches will still work.
With enabled (default), I got this message:
#error "No millis timer selected, but not disabled - can't happen!".
With set to disabled, old sketches go well. So maybe set disabled as default?
greetings from Albert
stay healthy..

SpenceKonde, 2020-03-20:
Unoh, what board were you using? And is that with 1.1.7?

SpenceKonde, 2020-03-20:
If you are getting that with default on 1.1.7 that is a critical bug in the core which requires an immediate fix and new release.

SpenceKonde, 2020-03-20:
Issue fixed, rolling 1.1.8

SpenceKonde, 2020-03-20:
A #endif was on the wrong side of the #else that reports invalid timer selection

closed 2020-03-20 by SpenceKonde

SpenceKonde, 2020-03-21:
Sorry about that.... 1.1.8 is now available for manual and board manager installation.

avandalen, 2020-03-21:
thank you very much

commit 3c268422 (2020-03-20): Fix #154