SpenceKonde/megaTinyCore #155: Problem with some EEPROM libraries

avandalen, 2020-03-21:
Hi Spence,
I have troubles using some EEPROM libraries, such as the EEPROMex.h library:
[https://github.com/SpenceKonde/megaTinyCore](url)
It doesn’t work for the ATtiny3217, the error is:
'NVM_STATUS' was not declared in this scope

Note that the standard Arduino library EEPROM.h still works fine, see this sketch:

```
#include <EEPROM.h>  
void setup() 
{ Serial.begin(9600);
  while(!Serial);
  float f = 123.456f;  
  int eeAddress = 0;   
  EEPROM.put(eeAddress, f);
  EEPROM.get(eeAddress, f);
  Serial.println(f, 3);   
}
void loop() 
{
}
```

So, I think it is not a big compatiblity issue.

I have looked for a solution, but I don’t have enough knowledge of the Arduino platform. 
Maybe you can talk to Thijs Elenbaas what to do.

SpenceKonde, 2020-03-22:
Wow. The actual bug is in <avr/eeprom.h> - their definition of eeprom_is_ready() refers to registers and bits within the register that don't exist.... 

I can't do anything to my core to fix it, but I put in a PR to the library maintainer to fix it: https://github.com/thijse/Arduino-EEPROMEx/pull/31
However it's not clear if he's actively maintaining that library anymore... Anyway - the change shown in that PR is the workaround needed to make it work (well, compile, at least). 

I also submitted a PR to the EEPROMWearLevel library, which I think you also reported... https://github.com/PRosenb/EEPROMWearLevel/pull/9

closed 2020-03-22 by SpenceKonde

avandalen, 2020-03-22:
I think other libraries should only use the standard EEPROM.h library, which works fine with the ATtiny3217 (see example). 
The library WearLeveling uses the standard EEPROM.h library and works fine:
https://github.com/chayanforyou/WearLeveling/tree/master/WearLeveling

avandalen, 2020-03-22:
The problem is solved, I think, when we use the standard library EEPROM.h in
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src\EEPROM.h

instead of AVR Libc EEPROM.h here:
https://github.com/Patapom/Arduino/blob/master/Libraries/AVR%20Libc/avr-libc-2.0.0/include/avr/eeprom.h
Where is this library located on my PC? I can’t find it.

SpenceKonde, 2020-03-22:
The issue with Pete Rosen's EEPROMWearLevel library is unrelated to the eeprom.h library - the issue with *that* library is that he was manually manipulation the EEPROM registers for finer control over the write process, but he only wrote code to work with classic avrs not megaavrs. See the PR I submitted over on his library for the correction for that library - though he seems to have a problem with where I put the architecture-dependent code, so it's up to him to take the next steps there. I made the library work *shrug* 

And as I noted above, I also submitted a fix to this EEPROM library to work around this bug in the standard avr/eeprom.h library. 

EEPROM.h is an arduino library included with the core. It actually does use the avr/eeprom.h library, just not the one bad macro. eeprom.h is located in the avr-gcc package that gets installed using board manager - assuming you're on windows, it ends up in C:\Users\(your username)\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr  (note that AppData is a hidden folder) 

The files that are part of avr-libc are *really* supposed to work; they're distributed with the compiler, FFS! It's very surprising when we see stuff like this... THEY are the "standard" library - EEPROM.h is an arduino-specific one that's just a wrapper around eeprom.h

avandalen, 2020-03-23:
Hi Spence

Thank you
you know a lot of the Arduino..

stay healthy

Albert

From: Spence Konde (aka Dr. Azzy) 
Sent: Sunday, March 22, 2020 11:17 AM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] Problem with some EEPROM libraries (#155)

The issue with Pete Rosen's EEPROMWearLevel library is unrelated to the eeprom.h library - the issue with that library is that he was manually manipulation the EEPROM registers for finer control over the write process, but he only wrote code to work with classic avrs not megaavrs. See the PR I submitted over on his library for the correction for that library - though he seems to have a problem with where I put the architecture-dependent code, so it's up to him to take the next steps there. I made the library work shrug

And as I noted above, I also submitted a fix to this EEPROM library to work around this bug in the standard avr/eeprom.h library.

EEPROM.h is an arduino library included with the core. It actually does use the avr/eeprom.h library, just not the one bad macro. eeprom.h is located in the avr-gcc package that gets installed using board manager - assuming you're on windows, it ends up in C:\Users(your username)\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr (note that AppData is a hidden folder)

The files that are part of avr-libc are really supposed to work; they're distributed with the compiler, FFS! It's very surprising when we see stuff like this... THEY are the "standard" library - EEPROM.h is an arduino-specific one that's just a wrapper around eeprom.h