SpenceKonde/megaTinyCore #157: java.lang.NullPointerException ATtiny1606

hardeman, 2020-03-24:
Hello,
I'm trying to program the ATtiny1606 with USBtinyISP. Compiling works fine (the most basic program) but when uploading is about to start I get this error: 

`
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:295)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:90)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2047)
	at java.lang.Thread.run(Thread.java:748)
`  

I tried almost everything that's there to find on the great interwebs, but nothing lead to the desired result. Do you have any idea if this could be an error in the board package?

SpenceKonde, 2020-03-25:
You cannot use an ISP programmer to program the megaavr parts. You must use
a UPDI programmer. The ide generates those errors when a programmer that
cannot be used is selected. If there is a way to suppress programmers that
are not applicable to the selected board, I dont know what it is.

A '328p based board can be used as a UPDI programmer. See:
https://github.com/SpenceKonde/megaTinyCore/blob/master/MakeUPDIProgrammer.md

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

hardeman, 2020-03-25:
Oke, my bad. Thank you for the info.

closed 2020-03-25 by hardeman