SpenceKonde/megaTinyCore #159: add wrapper around using optiboot to write to flash

SpenceKonde, 2020-04-02:
The version of optiboot used appears to support the application calling the bootloader to write to the flash; we should provide a user-friendly wrapper around this.

MCUdude, 2020-05-20:
I did write a simple proof of concept sketch for this some years ago. You can find it here:
https://github.com/MCUdude/MightyCore/blob/master/avr/libraries/Optiboot_flasher/examples/SerialReadWrite/SerialReadWrite.ino

SpenceKonde, 2021-04-03:
Deferring to 2.4,.0

MCUdude, 2021-04-03:
Not sure if you're interested, but I've created a wrapper library around optiboot.h that makes it much easier to store data like strings, structs and variables:

https://github.com/MCUdude/MightyCore/blob/master/avr/libraries/Optiboot_flasher/examples/Flash_put_get/Flash_put_get.ino

It is written for classic AVRs and requires a flash buffer (that acts as a memory pool). I think very little work is needed to make it work with tinyAVRs too.

SpenceKonde, 2021-04-18:
Examples from your Optiboot_flash library compile. Is there any documentation on this library other than the comments in the cpp files, which are clearly meant to be parsed by some automatic documentation tool...

SpenceKonde, 2021-04-19:
And it is in! Thank you, and thanks for the readme!

closed 2021-04-19 by SpenceKonde