SpenceKonde/megaTinyCore #161: Support for modified jtag2updi sketch for HV programming

SpenceKonde, 2020-04-02:
If this works, will also allow UPDI to be fused as Reset and GPIO for non-optiboot definitions.

Floessie, 2020-04-09:
I've been following dlloyd's adventures on the [forum](https://forum.arduino.cc/index.php?topic=669577.45) since you mentioned it in #150 and I'm wondering if their code is under version control somewhere. Can you give me a hint?

Dlloydev, 2020-04-12:
I've created a pull request [here](https://github.com/SpenceKonde/jtag2updi/pull/1#issue-402287190).
Please see [README_HV.md](https://github.com/Dlloydev/jtag2updi/blob/master/README_HV.md)

Dlloydev, 2020-05-11:
An update and findings with jtag2updi from ElTangas ... his latest version (6 days ago) that disables UPDI on exit. I found that I needed to remove the fast double-break from the HV enable sequence, as this would cause communication to fail. The previous version (4 months ago) from ElTangas and the version from here would work perfectly. The solution was simple, but I haven't figured out why?

Previous (full) sequence: Power-cycle, HV pulse, double-break @4MHz, sync @4MHz.
New (full) sequence: Power-cycle, HV pulse, sync @4MHz.

With HV programming, I have used 3 different points in the programming timeline to insert new functions, each having each having different advantages and purpose:

1. POR (Power-up)
2. Just prior to jtag2 sign-on
3. Just after jtag2 sign-off and final commands

I'm guessing the cause would be something with this in jtag2updi.cpp:

// Change emulator mode to signal that the UPDI unit must be turned off
JTAG2::PARAM_EMU_MODE_VAL = JTAG2::emul_mode::EMULATOR_MODE_UNKNOWN;

SpenceKonde, 2020-05-11:
Interesting - so he caught the thing about not disabling UPDI on exit? I just found that this morning - I've been up to my elbows in this fixing it's miserable instability. Almost done with a version that can recover from communication problems with both the host and target...But ofc we're working from slightly different starting points, it seems...

Dlloydev, 2020-05-11:
Yeah, by chance, I happened to cloned his firmware just a few hours after, I think your copy was just previous on the same day. No problem with the different starting points ... when your new version is ready I'll just start from a clone of it to add and test the HV programming functions.  Wow, really looking forward to you're upcoming version!

Dlloydev, 2020-05-11:
Wondering if its possible to reduce the "wasted?" delays between the end of TX (firmware-->PC) and the next RX (PC-->firmware). The delays don't improve, even after the baudrate changes from 19200 to 115200. Different serial adapters (CH340G, FT231X, MCP2200) make no difference. The delays vary from 3ms to 17ms, I guess would be a combination of the USB-serial bridge and avrdude. It looks like there's room for up to 40% improvement in programming time without changing any of the communication speeds. I've highlighted the [delays here](https://i.imgur.com/Uzmq9Su.png) in translucent red.

Dlloydev, 2020-05-18:
Awesome new version! The performance boost is quite noticeable. Thank you for your hard work for this overhaul!
Previous version from ElTangas, 1.02 seconds for a simple sketch upload. Same sketch now takes 0.76 seconds (25% quicker). I've added the HV programmer commands to my repo and run some tests for all UPDI pin configurations ... all works well.

SpenceKonde, 2020-10-27:
This support is in as of 2.1.0

closed 2020-10-27 by SpenceKonde