SpenceKonde/megaTinyCore #164: EEPROM library issues

MCUdude, 2020-04-05:
I just had a look at your custom EEPROM library, and it's a brilliant idea to make access to the user row fields "above" the 256 byte EEPROM field.

However, I've discovered a few issues. First, the pointers in `nvm_read_byte` and `nvm_write_byte ` should be 16-bit pointers, not 8. The two parameters in `nvm_write_byte` are also missing datatypes. I believe idx should be uint16_t and the second one should be uint8_t?

The uint8_t* pointer on [line 62](https://github.com/SpenceKonde/megaTinyCore/blob/8a2a053e79c06df55d99035fe40520b80c39d8c0/megaavr/libraries/EEPROM/src/EEPROM.h#L62) should also be removed I believe.

BTW @SpenceKonde I sent you a PM on the Arduino forum. Would be nice if you could reply 🙂

MCUdude, 2020-04-05:
Oh, here's my suggestion. I've tested in on an ATmega4809.

```c
#define nvm_read_byte(idx)  *(uint16_t *)((idx & 0xFF) | ((idx & 0x100) ? USER_SIGNATURES_START : EEPROM_START))

void nvm_write_byte(uint16_t idx, uint8_t dat) {
    *(uint16_t*)((idx & 0xFF) | ((idx & 0x100) ? USER_SIGNATURES_START : EEPROM_START)) = dat;
    uint8_t oldSREG=SREG;
    cli();
    _PROTECTED_WRITE_SPM(NVMCTRL.CTRLA, NVMCTRL_CMD_PAGEERASEWRITE_gc);
    SREG=oldSREG;
    while (NVMCTRL.STATUS & NVMCTRL_EEBUSY_bm);
}
```

SpenceKonde, 2020-04-05:
Thanks for the help on this. I literally spent 4-6 hours on learning pointers and it didn't seem to leave me much better than before.

MCUdude, 2020-04-05:
Well, I have to say I spend some time before I truly understood what the code actually did. It's pretty clever! But yeah, pointers are tricky (but powerful!).

SpenceKonde, 2020-04-05:
Wait, no, your code is wrong, and mine is right, aside from the fact that just converted it from a define to a function and forgot to add the datatypes... We're writing a byte at a time, not two bytes, so the pointers should be pointers to a uint8_t - if we make pointers to a larger datatype, and treat them as such, we'll be scribbling the high byte over the next memory address (not the current address - that gets the low byte as AVR is little endian like most architectures, hence why your testing looked okay) And the high byte is always 0x00 because we're writing a byte into two bytes.

```
#include <EEPROM.h>

void setup() {
  delay(1000);
  Serial.begin(57600);
  deletetestdata();
  showHex(EEPROM.read(0));
  showHex(EEPROM.read(1));
  showHex(EEPROM.read(2));
  showHex(EEPROM.read(3));
  showHex(EEPROM.read(4));
  showHex(EEPROM.read(5));
  showHex(EEPROM.read(6));
  showHex(EEPROM.read(7));
  Serial.println();
  EEPROM.write(2,0xBA);
  EEPROM.write(3,0xBE);
  EEPROM.write(0,0xCA);
  EEPROM.write(1,0xFE);
  Serial.println("Should be CAFEBABE right?");
  showHex(EEPROM.read(0));
  showHex(EEPROM.read(1));
  showHex(EEPROM.read(2));
  showHex(EEPROM.read(3));
  Serial.println();
  Serial.println("And the byte after should still be FF?");
  showHex(EEPROM.read(4));
}

void deletetestdata() {
  uint16_t addr=EEPROM_START;
  byte * dataptr = (byte *) addr;
  for (byte x=0;x<8;x++){
    *dataptr++=0xFF;
  }
  cli();
  _PROTECTED_WRITE_SPM(NVMCTRL.CTRLA, NVMCTRL_CMD_PAGEERASE_gc);
  sei();
  while (NVMCTRL.STATUS &NVMCTRL_EEBUSY_bm); 
}
void showHex (const byte b) {
  char buf [3] = { ((b >> 4) & 0x0F) | '0', (b & 0x0F) | '0', 0};
  if (buf [0] > '9')
    buf [0] += 7;
  if (buf [1] > '9')
    buf [1] += 7;
  Serial.print(buf);
}
void loop() {
  /* Empty loop */
}
```

```
FFFFFFFFFFFFFFFF
Should be CAFEBABE right?
CAFE00BE
And the byte after should still be FF?
00
```

With my code, after adding datatypes:
```
FFFFFFFFFFFFFFFF
Should be CAFEBABE right?
CAFEBABE
And the byte after should still be FF?
FF
```

closed 2020-04-05 by SpenceKonde

SpenceKonde, 2020-04-06:
I think the key point where you got off course was here: 
> should be 16-bit pointers, not 8.
*(uint8_t *)address
is a pointer *to* a uint8 - not an 8-bit pointer. All pointers (on AVR devices that don't have too much memory to address with 16 bits - on the larger parts, things get weird)  are 16-bit pointers; that's implied by it being a pointer.

I *am* going to steal your nicer formatted code though. You always make your code look prettier than I do.