SpenceKonde/megaTinyCore #165: Steal the solution for swapping pins from MegaCoreX

SpenceKonde, 2020-04-05:
Okay, I admit it, you (@MCUdude ) were right, and I was wrong. This is a better method and more extensible. Reduce the number of submenus, so the logic of which pin mapping to use for these functions . 
Also no more DAC submenu, instead a DAC.setRef() or something. 
Also have some ideas to improve DAC conversion speed. 
Also, PinMode type will be renamed to PinMode_t because otherwise PinMode() isn't a compile error, which isn't user-friendly - I'm embarassed how long it took me to see it in a sketch submitted to me as a bug.... Hopefully this isn't going to break anything else, but what were the arduino people thinking?!

MCUdude, 2020-04-05:
Please, borrow as much as you need! Compatibility across cores is always a good thing.

I also think that pinMode/digitalRead/digitalWrite is pretty messed up. The official megaAVR core is the *only* "official" core that implements it and no one likes it. I've been wanting to ditch the "new" implementation and do it the "old fashion way" where the code just works. This core isn't even compatible with any official Arduino hardware, so (lucky) you have nothing to lose in terms of compatibility. MegaCoreX, on the other hand, supports the UNO Wifi and the Nano Every, so I might break something if I rewrite these functions. But, really, I have nothing to lose. Users that use my core aren't really switching between MegaCoreX and the official core all the time (I think). It's designed towards "non-official" hardware, just like this core.

SpenceKonde, 2020-04-05:
Yeah - I replied on forums - I've got lots to talk about!

closed 2020-04-11 by SpenceKonde