SpenceKonde/megaTinyCore #168: Return pinMode, digitalWrite, digitalRead to working with numbers, not enums

SpenceKonde, 2020-04-07:
The official arduino megaavr core developers made a widely panned decision to replace HIGH/LOW and the different pin modes, which were previously #defines with enums. This change provided no benefit, and broke large volumes of code (because digitalWrite'ing 1 or 0 was ubiquitous . I had not noticed this during internal testing because I usually use direct port writes... I apologize for taking this long to become aware of this new behavior.

2.0.0 will return to the classic behavior - the "new" behavior on official

MCUdude, 2020-04-07:
[Here's my implementation](https://github.com/MCUdude/MegaCoreX/commit/fb93094d6e149b625d3b0c98f9544a959b1cd87b). However, the latest Arduino megaavr core (1.8.6) has this somewhat fixed while still keeping the enums. But to get their fix working you will basically have to wrap all files in the api folder around `namespace arduino`.

SpenceKonde, 2020-04-08:
With their implementation, does if (digitalRead(pin)) work like it did in the old days? Never bothered to test....

SpenceKonde, 2020-04-08:
Changing back to the old implementation also saves 36 bytes of flash, and speeds up execution:

function| clocks | percent improvement
-- | -- | --
digitalWrite(OUTPUT) | 5 | 4.00%
digitalWrite(INPUT) | 5 | 3.38%
digitalRead() | 0 | 0.00%
analogRead() | 0 | 0.00%
pinMode(OUTPUT) | 2 | 5.48%
pinMode(INPUT) | 3 | 4.72%
pinMode(INPUT_PULLUP) | 3 | 4.96%

MCUdude, 2020-04-08:
Well, it's a no-brainer then!

SpenceKonde, 2020-04-08:
Control | 0 | 0.00%
-- | -- | --
digitalWrite(OUTPUT nonPWM) | 5 | 6.58%
digitalWrite(INPUT nonPWM) | 5 | 5.05%
digitalWrite(OUTPUT) | 5 | 4.00%
digitalWrite(INPUT) | 2 | 1.73%
analogWrite() | 5 | 3.38%
digitalRead() | 0 | 0.00%
analogRead() | 0 | 0.00%
pinMode(OUTPUT) | 2 | 5.48%
pinMode(INPUT) | 3 | 4.72%
pinMode(INPUT_PULLUP) | 3 | 4.96%
  |   |  
Flash | 57 | 1.09%
SRAM | 0 | 0.00%

A bit more data

SpenceKonde, 2020-04-08:
No idea why analogWrite is faster now!

closed 2020-04-10 by SpenceKonde