SpenceKonde/megaTinyCore #171: Is there a reason I can't return an int8_t instead of an int from digitalRead?

SpenceKonde, 2020-04-11:
It seems like an int8_t would be a better choice here - there's no benefit from the larger type, as we only ever return 0 or 1 (heck, we don't even return -1 in the event that they pass a bogus pin!)

SpenceKonde, 2020-04-11:
2.0.0 will do this; I'll change it back if people complain. 

I'd change it to return a -1 if people pass something that's not a pin, except that people probably expect to it to return 0 so it evaluates to false - it wouldn't even impose a flash/speed penalty...

SpenceKonde, 2020-04-11:
Actually, no, I'm just going to do that too, and change it back if people complain. I think giving people an indication of an error, even if they may not know to look for it, is better than returning something that's indistinguishable from correct output.

SpenceKonde, 2020-04-11:
I also removed turnOffPWM from digitalRead(). I couldn't think of a reason that doing that made any sense whatsoever.

MCUdude, 2020-04-11:
I used int because that's what the official core used. But like you said, int8_t or uint8_t should be fine. I should probably change this too. From a portability standpoint, int8_t/uint8_t, int16_t/uint16_t and int32_t/uint32_t are much better rather than short, int, long and so on. When I'm writing code I always use these typedefs.

SpenceKonde, 2020-04-11:
Yeah, something like a few weeks ago I was doing a bunch of work on the core and realized I really ought to be using those types, not the other names for those types. And I;ve since been trying to stick to that. One advantage to the other names that I have noticed, though, is that it seems harder to typo the names.

Floessie, 2020-04-16:
> From a portability standpoint, int8_t/uint8_t, int16_t/uint16_t and int32_t/uint32_t are much better rather than short, int, long and so on. When I'm writing code I always use these typedefs.

Absolutely. And as you mention portability, there's also a reason for the [`u?int_fast(8|16|32|64)_t`](https://en.cppreference.com/w/c/types/integer) types. IMHO, `u?int_fast8_t` should be the one used in the official core.

SpenceKonde, 2020-05-03:
Okay, this was implemented in 2.0.0, so closing this.

closed 2020-05-03 by SpenceKonde