SpenceKonde/megaTinyCore #172: DACVREF is not defined

CommonGrounds, 2020-04-12:
analogWrite() on the DAC pin gives us digital output.

It seems like defining DACVREF with default value in Arduino.h gives us 
acceptable output ( tried on attiny 412 and 3216 ) 

#define DACVREF INTERNAL0V55

SpenceKonde, 2020-04-12:
Thank you for reporting this. 

That proposed fix is entirely the wrong solution. The DACVREF define is to be removed from the core entirely - clearly I missed a bunch of spots, with the result being that the DAC is unusable, will be pushing out a fix for this shortly. Please see the version 2.0.0 release announcement and description of the DAC in the updated readme; the DAC reference voltage is now set with the new DACReference() function.

SpenceKonde, 2020-04-17:
fixed by 2b37418004c9dc4f63e3ef441d7315ee50b7edc3

closed 2020-04-17 by SpenceKonde