SpenceKonde/megaTinyCore #173: Compile fails for Tiny814 with ADC reference External

EclipseBoom, 2020-04-16:
Same code works in v1.1.9

in v2.0.0  error is 

`In file included from C:\Users\Chris\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.0\cores\arduino\wiring_private.h:31:0,

                 from C:\Users\Chris\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.0\cores\arduino\wiring_analog.c:25:

C:\Users\Chris\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.0\cores\arduino\wiring_analog.c: In function 'analogReference':

C:\Users\Chris\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.0\cores\arduino\Arduino.h:60:25: error: 'ADC_REFSEL_VREFA_gc' undeclared (first use in this function); did you mean 'ADC_REFSEL_VDDREF_gc'?

     #define EXTERNAL    ADC_REFSEL_VREFA_gc

                         ^

C:\Users\Chris\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.0\cores\arduino\wiring_analog.c:34:8: note: in expansion of macro 'EXTERNAL'

   case EXTERNAL:

        ^~~~~~~~

C:\Users\Chris\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.0\cores\arduino\Arduino.h:60:25: note: each undeclared identifier is reported only once for each function it appears in

     #define EXTERNAL    ADC_REFSEL_VREFA_gc

                         ^

C:\Users\Chris\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.0\cores\arduino\wiring_analog.c:34:8: note: in expansion of macro 'EXTERNAL'

   case EXTERNAL:

        ^~~~~~~~

exit status 1
Error compiling for board ATtiny1614/1604/814/804/414/404/214/204.
`

SpenceKonde, 2020-04-16:
Woah.... broken since 1.1.10...

Good catch!

The reason it didn't show up on older versions is that they were broken and
never provided EXTERNAL reference option for anything.

And I hadn't read the datasheets for the lower spec parts as closely as I
should have... I thought that all the 1-series parts supported external
voltage reference. But no - only the "good" ones (1614, 1616, 1617, 3216,
3217) have it! And now that my code was actually using EXTERNAL reference
as an option, it was checking for any 1-series parts (see you can't check
for whether ADC_REFSEL_VREFA_gc is defined because it's part of an enum,
not a define - that was what I'd done wrong before - it was never "defined"
so nothing had external reference usable). I habitually used the
1614/3216/3217 datasheets as reference...

SpenceKonde, 2020-04-17:
Okay, this is fucked up as hell. The headers don't agree with the datasheets about which parts do and don't have the external reference. 

I think what I'm going to do is is provide two #defines, EXTERNAL and EXTERNAL_EXPERIMENTAL - EXTERNAL on parts that I know have it, and EXTERNAL_EXPERIMENTAL for ones that - who knows if they have it or not?!

SpenceKonde, 2020-04-17:
Fixed by a148b90e4aa24091709f27f9559731ebffe3be63

closed 2020-04-17 by SpenceKonde