SpenceKonde/megaTinyCore #175: 16/8 MHz does not work on ATTiny402

pgwthf, 2020-04-19:
If I set the CPU Frequency to 16 Mhz, the ATTiny actually runs at 20 and if I set it to 8 it runs at 10. Apparently the  FREQSEL[1:0] fuse bits are not correctly set.

I am programming the Tiny through a Nano, as per this guide: 
https://www.electronics-lab.com/project/getting-started-with-the-new-attiny-chips-programming-the-microchips-0-series-and-1-series-attiny-with-the-arduino-ide/

Is this a bug, or am I doing something wrong?

SpenceKonde, 2020-04-19:
Did you remember to do burn bootloader after changing the clock speed from
20mhz derived speed to 16mhz derived speed? As with all arduino third party
hardware packages and described in the readme, fuses are only set when
doing "burn bootloader"

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-04-20:
Closing because cannot reproduce; this does not look like an issue with the core (likely what I mentioned above regarding forgetting to set fuses)

SpenceKonde, 2020-04-20:
I will still see comments here, though - please don't hesitate to update with further questions or confirmation that you have it working now.

closed 2020-04-20 by SpenceKonde

pgwthf, 2020-04-20:
Just tried to burn bootloader and that works. Thanks!
Sorry, I must have skipped that part in the readme.