SpenceKonde/megaTinyCore #177: Buffer size in wire lib to large

LordJakson, 2020-04-24:
Is there a reason why the buffer size got boosted to 128 bytes?
On my 404 just including the wire.h and calling Wire.begin exceeds the available memory!
16 or 32 byte are enough in most cases.

`#define BUFFER_LENGTH 128`

insteadof 

```
#if ((RAMEND - RAMSTART) < 1023)
#define BUFFER_LENGTH 16
#else
#define BUFFER_LENGTH 32
#endif
```

SpenceKonde, 2020-04-24:
Ugh, this is a serious bug! Happened because I pulled in the MegaCoreX
version of the library to get the pinswapping stuff and didnt realize the
other changes that got brought in...

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-05-03:
Fixed by d06361241df7020c9a4c83e1b666058403148cfe

closed 2020-05-03 by SpenceKonde