SpenceKonde/megaTinyCore #180: Cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

trusleymike, 2020-05-02:
Arduino IDE gives this error:

  Cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

with megaTinyCore installed when trying to compile for the ATtiny1614.

The code works when compiled for an ATmega328P but fails for the ATtiny1614:

    digitalWrite (LED,!digitalRead(LED));                 // Toggle LED

(LED is defined on IDE pin 3, PA7.)

Workaround:

    if (digitalRead(LED) == HIGH) digitalWrite (LED, LOW);  //Alternative: toggle LED
    else digitalWrite (LED, HIGH);

SpenceKonde, 2020-05-02:
Are you using latest version of core? (2.0.1 ) Will check this shortly - all the PinStatus and PinMode madness was supposed to have been removed with 2.0.0, (also significantly improved performance of those functions and reduced flash use) - maybe one was left behind?

trusleymike, 2020-05-02:
Quite right - fixed in 2.0,1.  Sorry to have bothered you.

SpenceKonde, 2020-05-02:
Great!

(just in time for 2.0.2, which will be coming out in an hour or two...)

What is sort of embarrassing is that I never noticed that until someone asked me about it (just after 1.1.10 release - at which point I realized a v2 was needed, because that garbage had to be removed..., because I so frequently use direct port writes instead of the arduino pin manipulation functions....

closed 2020-05-02 by SpenceKonde

trusleymike, 2020-05-26:
Spence

Couple of compile time #elif problems with the latest version in sys.h.

E.g.

  #elif defined (__AVR_ATmega_Zero__ || __AVR_DA__)
  
should be

  #elif defined (__AVR_ATmega_Zero__) || defined (__AVR_DA__)

Best regards

Mike Lake

SpenceKonde, 2020-05-26:
Thanks, I'll fix those shortly, like I said is untested on those parts -
any issues should be silly stuff like that. Any more complicated issues
should be addressed at a future time by someone else. I am trying to get
that code merged into the upstream so that my fork will not contain any
changes other than folder structure.

But what does this have to do with "Cannot convert 'bool' to 'PinStatus'
for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)"

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com