SpenceKonde/megaTinyCore #183: ATtiny406 not able to utilize I2C

MuzammilPatel, 2020-05-03:
ISSUE -   I am trying to use I2C pins on ATtiny406 to operate OLED SSD1306B 128*32 display using Tiny4kOLED library. 

By default Tiny4kOLED lib. uses wire.h, from my understanding as I am using megaTinyCore it using wire.h from the package(megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src). 

While verifying it shows errors as follows-
`Arduino: 1.8.9 (Windows 10), Board: "ATtiny3216/1616/1606/816/806/416/406, ATtiny406, 8 MHz, 1.8V, Disabled, Disabled, EEPROM retained, Enabled (default timer), Closer to 3v"

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp: In member function 'bool TwoWire::pins(uint8_t, uint8_t)':

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:67:24: error: 'PORTMUX_TWI0_bm' was not declared in this scope

         PORTMUX.CTRLB|=PORTMUX_TWI0_bm;

                        ^~~~~~~~~~~~~~~

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:67:24: note: suggested alternative: 'PORTMUX_TCB0_bm'

         PORTMUX.CTRLB|=PORTMUX_TWI0_bm;

                        ^~~~~~~~~~~~~~~

                        PORTMUX_TCB0_bm

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:73:25: error: 'PORTMUX_TWI0_bm' was not declared in this scope

         PORTMUX.CTRLB&=~PORTMUX_TWI0_bm;

                         ^~~~~~~~~~~~~~~

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:73:25: note: suggested alternative: 'PORTMUX_TCB0_bm'

         PORTMUX.CTRLB&=~PORTMUX_TWI0_bm;

                         ^~~~~~~~~~~~~~~

                         PORTMUX_TCB0_bm

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:79:25: error: 'PORTMUX_TWI0_bm' was not declared in this scope

         PORTMUX.CTRLB&=~PORTMUX_TWI0_bm;

                         ^~~~~~~~~~~~~~~

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:79:25: note: suggested alternative: 'PORTMUX_TCB0_bm'

         PORTMUX.CTRLB&=~PORTMUX_TWI0_bm;

                         ^~~~~~~~~~~~~~~

                         PORTMUX_TCB0_bm

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp: In member function 'bool TwoWire::swap(uint8_t)':

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:113:24: error: 'PORTMUX_TWI0_bm' was not declared in this scope

         PORTMUX.CTRLB|=PORTMUX_TWI0_bm;

                        ^~~~~~~~~~~~~~~

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:113:24: note: suggested alternative: 'PORTMUX_TCB0_bm'

         PORTMUX.CTRLB|=PORTMUX_TWI0_bm;

                        ^~~~~~~~~~~~~~~

                        PORTMUX_TCB0_bm

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:119:25: error: 'PORTMUX_TWI0_bm' was not declared in this scope

         PORTMUX.CTRLB&=~PORTMUX_TWI0_bm;

                         ^~~~~~~~~~~~~~~

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:119:25: note: suggested alternative: 'PORTMUX_TCB0_bm'

         PORTMUX.CTRLB&=~PORTMUX_TWI0_bm;

                         ^~~~~~~~~~~~~~~

                         PORTMUX_TCB0_bm

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:125:25: error: 'PORTMUX_TWI0_bm' was not declared in this scope

         PORTMUX.CTRLB&=~PORTMUX_TWI0_bm;

                         ^~~~~~~~~~~~~~~

C:\Users\Muzammil\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp:125:25: note: suggested alternative: 'PORTMUX_TCB0_bm'

         PORTMUX.CTRLB&=~PORTMUX_TWI0_bm;

                         ^~~~~~~~~~~~~~~

                         PORTMUX_TCB0_bm

exit status 1
Error compiling for board ATtiny3216/1616/1606/816/806/416/406.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

On doing some research I found that the wire.h file is not able to locate iotn406.h(which is only in \arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\iotn406.h )which contains all the PORTMUX defs. When I manually enter **#define PORTMUX_TWI0_bm  0x10**
in Wire.cpp (megaTinyCore\hardware\megaavr\2.0.1\libraries\Wire\src\Wire.cpp) file it is able to run but now I am getting **R_AVR_13_PCREL against undefined symbol `serialEventRun()'** which is related to not specified RX and TX pin.

The above Issue only started after I have upgraded to 2.0.1, previously I was able to select the I2C pins and RX/TX in Arduino/Tools/... which worked all well. 

Here is a simple OLED code that I am testing-
`# define __AVR_ATtiny406__
#include <Wire.h>
//#include <TinyWireM.h>
#include <Tiny4kOLED.h>

#define SDA 8
#define SCL 9

void setup() {
//  Enable PORT pins Manually
//  PORTB.DIR = PIN0_bm | PIN1_bm;  // sets PA3 and PA6 as an output
//  Wire.swap(1);
//  Wire.pins(8, 9);
  Wire.pins(SDA, SCL);
  Wire.begin();
 
//  oled.begin();
  oled.setFont(FONT8X16);
  // Clear the memory before turning on the display
  oled.clear();
  // Turn on the display
  oled.on();
  // Switch the half of RAM that we are writing to, to be the half that is non currently displayed
  oled.switchRenderFrame();
}

void loop() {
  updateDisplay();
  delay(50);
}

void updateDisplay() {
  // Clear the half of memory not currently being displayed.
  oled.clear();
  oled.setCursor(0, 1);
  // Write text to oled RAM (which is not currently being displayed).
  oled.print(F("ms: "));
  // Write the number of milliseconds since power on.
  oled.print(millis());
  // Swap which half of RAM is being written to, and which half is being displayed.
  // This is equivalent to calling both switchRenderFrame and switchDisplayFrame.
  oled.switchFrame();
}
`

Any help would be much appreciated.
Thanks
Muzammil

SpenceKonde, 2020-05-03:
Holy crap - I was in the process of creating a new release - I am investigating this now

MuzammilPatel, 2020-05-03:
Thanks @SpenceKonde.

SpenceKonde, 2020-05-03:
@MuzammilPatel can you please link to the exact version of that library you are using? I will see if I can reproduce your problem with the 2.0.2 code that I was literally minutes from releasing when this issue came through

MuzammilPatel, 2020-05-03:
@SpenceKonde,
megaTinyCore v2.0.1 [https://github.com/SpenceKonde/megaTinyCore](url)
Tiny4KOLED v1.5.1 [https://github.com/datacute/Tiny4kOLED](url)

I had installed both using Arduino library and board manager.

SpenceKonde, 2020-05-03:
I've reproduced the issue with the missing define. 

Do not define SCL and SDA. Those defines are provided by the core (but that's not the cause of the issue)

Investigating why it's missing.,

MuzammilPatel, 2020-05-03:
Initially, I didn't define SDA and SCL but was testing if that might work.

SpenceKonde, 2020-05-03:
Okay, I've found the issue....

MuzammilPatel, 2020-05-03:
Awesome, What is it?

SpenceKonde, 2020-05-03:
Okay, fix for this is being checked in - the pins_arduino.h files assumed that the 0-series and 1-series parts all had the same pin remapping options (also, the alt and normal I2C pins were backwards on 1-series).

That said, you still reach the same error you got (your fix was the wrong way to resolve it, but it wound up with you in the same place). As stated in the compiler bugs section of the Readme, you can get R_AVR_13_PCREL errors from the linker when you don't have enough space for the sketch instead of a sensible error message. The problem is that the sketch, when compiled, is like 4.5k, and you only have 4k of flash. (Considering the price of the 3216, I really wonder why people end up playing with a 406...)

2.0.3 (not 2.0.2, which is coming in a few minutes) will contain the fix for the terrible error message (and the bad eeprom.h as well) - I have the modified toolchain giving correct error message in my test environment, but I don't have it in a deployable form. There's another thing I need to get set up for that before I generate the new toolchain packages.

closed 2020-05-03 by SpenceKonde

MuzammilPatel, 2020-05-03:
@SpenceKonde, thanks. 
I had a few 406 given by one of mate to test few I/O options for his next project.
Anyways glad to see its been fixed.

SpenceKonde, 2020-05-03:
You should definitely try to score some 3216s - those are amazing chips. Actually - if I could recommend a part to you without spoiling the surprise, I would totally recommend something even more awesome that will be supported real soon now (tm)...

MuzammilPatel, 2020-05-03:
I definitely try 3216s. Waiting for your awesome upcoming supports..... :)

SpenceKonde, 2020-05-03:
Upcoming? 3216 is already supported by this core (and much better tested - vast majority of my testing is done with the 1614 and 3216 parts (the 1-series parts with 16k or more of flash have a few "bonus" features - external ADC reference, the weird second ADC, the second type B timer). A bunch of the recent fixes - including this one - were for cases where I had assumed that lower-spec parts worked the same as the top-end ones that I used in my testing.

SpenceKonde, 2020-05-03:
It's worth noting that a big part of the reason that these problems exist is that we don't have any sort of CI (automated testing) to compile-check a set of test sketches for every part.

MuzammilPatel, 2020-05-03:
I can feel the pain of testing. Sometimes its just a nightmare.

commit 468aa971 (2020-05-03): Correct I2C pins on 20-pin parts, check that pins swappable, don't assume

This fixes #183