SpenceKonde/megaTinyCore #186: EEPROM erased when programming

pgwthf, 2020-05-04:
Regardless of the "EEPROM (not) retained" setting, EEPROM is always erased when programming.

I change the setting to EEPROM retained and burn bootloader 

I made a test sketch that reads an EE address and then writes a value to it.

When I upload this sketch it writes a value to EEPROM and reads 0xFF (which is correct)
When I upload this sketch again, it reads 0xFF again (incorrect, it should be the new value).
But if I power-cycle, all EEPROM is retained and the new value is correctly read.

System:
OSX 10.14.6
Arduino 1.8.12
MegaTinyCore 2.0.2
Processor: ATtiny402
Programmer: Arduino Nano / JTAG2UPDI

SpenceKonde, 2020-05-07:
Thanks for reporting. I will investigate - at first glance though, it looks like it should be setting the fuses correctly. You did burn bootloader with recent version of core, right? IIRC it wasn't *that* long ago that I fixed this, so if you burned bootloader a while back, that would explain it....

pgwthf, 2020-05-07:
Hi Spence, immediately before reporting I downloaded the latest MegaTinyCore 2.02 and burnt bootloader (several times, with and without setting "retained"). Or are there other packages that should be updated, besides the ones I mentioned in the 1st post?

SpenceKonde, 2020-05-07:
No. that's the one - I was just concerned that you may have burned bootloader a while ago setting it to be retained, then updated, and *then* noticed that uploading didn't retain eeprom, having never reburned bootloader. 

I am currently up to my elbows in a different (exciting and promising) project, once I have my new version of jtag2updi working, I'll take a look at this

pgwthf, 2020-05-07:
No worries, I know what the issue is, so I can work around it.
Thanks for all the good work anyway!

SpenceKonde, 2020-05-07:
I will say that I am somewhat alarmed to hear that it doesn't work as is though - like, I don't see an obvious reason for it not to be working

SpenceKonde, 2020-09-28:
Moved this to next release. Have been getting complaints from more than one other person. I have a sense of forboding about this though... Cause like, the fuse *is* being set correctly. So why ain't it doing what the datasheet says it should?

My current most likely theory is - see there are 2 ways to do a chip erase - one is via a UPDI "key", and the other is by using UPDI to send that command via NVMCTRL. What if one of those ignored that fuse?

SpenceKonde, 2020-10-03:
This seems to be working now. There were some issues in EEPROM library. I saw some whack shit with EEPROM address 0, but after some changes I cant reproduce that anymore. I did not EVER see the issue with the EESAVE setting being ignored. Note that you must burn bootloader to change it.

closed 2020-10-03 by SpenceKonde