SpenceKonde/megaTinyCore #190: ATtiny402w doesn't seem to work with "bad signature" chips

iosengineer, 2020-05-26:
Although the definitions in the `avrdude.conf` file are correct for the ATtiny402 and ATtiny402w, the only way I can get my "bad signature" chips to work is to replace the ATtiny402 signature with the ATtiny402w signature directly. It's like avrdude isn't getting the `w` part of the chip identifier.

Issue exists still in version 2.0.2, installed fresh today.

SpenceKonde, 2020-05-26:
Thanks, will investigate...

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-06-02:
Thanks, this has been corrected and fix will be in 2.0.4 - boards.txt never actually referenced the attiny402w part.

closed 2020-06-02 by SpenceKonde