SpenceKonde/megaTinyCore #191: ATtiny1614 at internal 8MHz clock speed

dpetican, 2020-05-28:
Is there a way to run the ATtiny1614 at 8MHz using an internal oscillator?  I've read the datasheet and its not clear to me.  If its possible I can't discern what fuse bits to set to what to do this.  If I run off of an external 8MHz do I have to set any fuse bits?  Thanks.

SpenceKonde, 2020-05-28:
Yes, it's very easy!
Select the 3216/1616/... from tools -> board menu. Select the 8 mhz
internal option from the tools-> clock speed menu, and then do "burn
bootloader". When uploading, be sure to use the same setting for click
speed.

Background:
These parts use a different scheme for clock speed than the classic avr's -
a fuse sets whether the base oscillator speed is 16 or 20 mhz. It always
starts with that prescaled by 6, but the prescaler can be changed at
runtime; megaTinyCore does this (in init(), which is run at startup before
setup is called) to get the selected clock speed. See the readme for more
information.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-05-28:
Ah, and these parts do not support external crystal. This core does not
currently support external *clock* (which these new parts do support) -
this may be supported in a future version. The internal oscillator is much
better on these parts vs classic avr's though, so there is rarely a need
for a more accurate external clock source.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

dpetican, 2020-05-28:
Why do I need to burn the bootloader?  I use the Atmel Studio tool to set the clock freqency.

Going to try again with your method. I think I read that if I want 8MHz the internal clock has to 16MHz and for 10MHz it has to be 20MHz.  Is this correct?

SpenceKonde, 2020-05-28:
That is correct. If you want to use the atmel took to set clockspeed, go
ahead, but burn bootloader does the exact same thing without using external
tools (unless you select the "optiboot" board option, no bootloader is
involved, it just sets fuses - I agree that this convention is somewhat odd)

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

dpetican, 2020-05-28:
All I get is an error message when I try to burn the bootloader.  I can access the chip using Atmel Studio.

`Arduino: 1.8.10 (Windows 10), Board: "ATtiny1614/1604/814/804/414/404/214/204, ATtiny1614, 8 MHz, 1.8V, Disabled, Disabled, EEPROM retained, Enabled (default timer), Closer to 5v"

java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:329)
	at processing.app.Editor.lambda$handleBurnBootloader$65(Editor.java:2394)
	at java.lang.Thread.run(Thread.java:748)
Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

SpenceKonde, 2020-05-28:
Did you have jtag2updi or another supported updi-capable programmer
selected? That error indicates that you had a non-updi programmer selected.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

dpetican, 2020-05-28:
Derp.  I forgot to change the mode.  I was previously programming another part in ISP mode with the same programmer.  Also, I've found that sometimes I have to plug/unplug the programmer because it stops responding.  I got it to work.

Actually I probably had it working before but didn't realize it.  I'm running the ATtiny1614 on a variable power supply (ie battery that runs out).  I have a circuit that detects when the power goes below 3.2V.  The mcu then prints out a message and turns itself off by unlatching the power supply so as not to drain the battery further.  However, it turns out that some time is required for the serial messages to print before I unlatch the power.  Doh.  It looked like it wasn't running below ~3.2V at 8MHz.  I changed the code and verified that indeed it does work at very low voltages.  Thanks for the help.

closed 2020-05-28 by dpetican

SpenceKonde, 2020-05-28:
Use Serial.flush() to wait for everything in the serial buffer to be sent.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com