SpenceKonde/megaTinyCore #193: BAUD rate settings for 20MHz vs 16MHz - please check

EclipseBoom, 2020-06-01:
I've been getting serial transmission errors when I try to use 16MHz clock speed.  Exact same settings at 20MHz work perfectly.  AtTiny 1616M, 115200 8N1 settings.

I'm using IRDA mode on the UART which is super picky about pulse lengths and spacing so a small error might scrape by on Asynch mode which won't on IRDA mode.

Anyway I'm just requesting that you verify the Baud rate register setting calculations for 16MHz.

BTW thanks a bunch for all of your hard work on this core.  It is really helpful to use these newer very small and cool chips.

SpenceKonde, 2020-06-01:
Did you remember to "burn bootloader" when switching from 20MHz to 16MHz?
The choice of20 vs 16MHz base oscillator frequency is controlled by fuses,
so "burn bootloader" is required, following the convention set by other
cores, and as documented in the README.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

EclipseBoom, 2020-06-01:
Using UPDI programming

SpenceKonde, 2020-06-01:
And you did choose "burn bootloader" to set the fuses? Uploading does not
set fuses, in keeping with the longstanding conventions of arduino cores,
so if fuses need to be changed, you must still do "burn bootloader" even
when using a board definition which does not use a bootloader (in that
case, fuses are set, but no bootloader is uploaded).

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-06-02:
And, if it's not clear, I am asking these questions because your issue does not reproduce for me. 

Unless I set it for 16MHz without doing "burn bootloader" to set the OSCCFG fuse, in which case it is expected that timing and baud rates will be wrong.

EclipseBoom, 2020-06-04:
You are correct.  Sorry I didn't carefully RTFM.  It IS in the readme that you have to burn bootloader if you want to change the fuses.  

I'm not using Optiboot so I didn't want to lose the FLASH space to a bootloader that I won't be using.  Also, I didn't want the startup delay but I also see now that a power-on reset doesn't invoke the bootloader so that won't be an issue for me.

I guess you can close this as a duh! "Never Mind" resolution.

Thanks for taking a look at it anyway.

SpenceKonde, 2020-06-05:
Unless you select the "Optiboot" board definition, there is no bootloader - "burn bootloader" just sets the fuses and erases the flash. 

If you do not require use of a bootloader, do not use the "(Optiboot)" board definition. Either way, fuses will only be set when you do "burn bootloader". This is in keeping with the widely established Arduino conventions. It is unfortunate that the Arduino IDE named that menu option "burn bootloader", as it does not accurately represet what it does. "Burn bootloader (if any) and set fuses" is what it does.

closed 2020-06-05 by SpenceKonde