SpenceKonde/megaTinyCore #194: Attiny1617 no PWM on PC1 and PC0

xpix, 2020-06-03:
Hallo, 

i'm trying to run a DRV8837 motor driver with an Attiny1617, unfortunately the analogWrite() function doesn't really seem to provide a PWM signal on the pins. I use megaavr 2.0.2

I have at 0 - 129 LOW and from 130 HIGH, this was unfortunately no PWM signal on the oscilloscope. Here my test script for it:

void setup() {
}

void loop() {
  analogWrite(13, 130);
  analogWrite(12, 130);
}

[image]

How can I get a clean square wave signal?

Greetings from Germany & great work!
Frank

xpix, 2020-06-04:
Ok, founded ... sorry, but the user has to switch over IDE->Tools->millis to TB1. :)

closed 2020-06-04 by xpix

SpenceKonde, 2020-06-04:
Hmm, are you using latest version of megaTinyCore? The dependence on TCD0
not being used for millis was removed a while ago.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

xpix, 2020-06-04:
As described, I am using version 2.0.2 or is there still versions available under versions? Where should I look up?

```
Using board 'atxy7' from platform in folder: C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\megaTinyCore\hardware\megaavr\2.0.2
Using core 'arduino' from platform in folder: C:\Users\Frank\OneDrive\Dokumente\ArduinoData\packages\megaTinyCore\hardware\megaavr\2.0.2
```