SpenceKonde/megaTinyCore #196: CRC error during installation

harry-finch, 2020-06-09:
Hi, 

I get a CRC error when I try to install megaTinyCore through the board manager in Arduino IDE. I tried older versions with the same result. I use Windows 10 pro and the latest version of Arduino IDE and tried on another computer on the same network also with Win 10 pro with the same error.

However, I tried with my MacBook on another network with a slightly older version of Arduino IDE and there it worked...

This is puzzling... any ideas what could be the issue here?

Many thanks!

SpenceKonde, 2020-06-13:
Can you provide the full error output? Which file is it saying there is a CRC error with? 

Can you access that file on that network via the browser?

harry-finch, 2020-06-16:
Here's the error I get:
`CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:105)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:113)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more`

I'm not sure which file is the problem...

SpenceKonde, 2020-07-05:
CNR - this is a transient network issue or unrelated problem with some other component.

closed 2020-07-05 by SpenceKonde