SpenceKonde/megaTinyCore #197: wiring_shift.c bad code size and speed

apollo2mond, 2020-06-11:
Hi, 
sometimes c&p is not the best idea, arduino's shiftin/shiftout function is one example. These functions are horrible inefficient especially for small machines because of the use of variable shifts like (1<<i).
I propose a replacement with the functions below which significantly have better code size and speed! Keep going!

// proposed by dk4ug
uint8_t shiftIn(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder) {
	uint8_t val = 0;
	for (uint8_t = 0; i != 8; ++i) {
		digitalWrite(clockPin, HIGH);
		if (bitOrder == LSBFIRST)
			val = (val>>1) | (digitalRead(dataPin)<<7);
		else
			val = (val<<1) | digitalRead(dataPin);
		digitalWrite(clockPin, LOW);
	}
	return val;
}

void shiftOut(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t val) {
	for (uint8_t i = 0; i != 8; i++)  {
		if (bitOrder == LSBFIRST)
			digitalWrite(dataPin, val & 0x01), val >>= 1;		
		else
			digitalWrite(dataPin, !!(val & 0x80)), val <<= 1;

		digitalWrite(clockPin, HIGH);
		digitalWrite(clockPin, LOW);
	}
}

SpenceKonde, 2020-07-05:
Yeah, these functions are an abomination in their current form - those are better. But digitalWrite()/digitalRead() are slowing it down a lot (i suspect far more than the issues you flagged up) (#205)

SpenceKonde, 2020-07-20:
Your changes save 24b of flash, and probably a small amount of processing time. I'm not sure it's worth putting in though.
Its still ~22us per trip through the loop on the wasteful digitalWrite/digitalRead functions at 20MHz... so it's (off the cuff estimate here) probably like 180-200 us per byte, lol... 

fastDigitalIO likely doesn't help here (depends on if LTO is smart enough to figure it out) 

A much faster implementation would be to look up the bitmasks of the two pins, and a pointer (as volatile uint8_t*) to the IN register of the PORTx register for datapin in shiftin, and the OUTSET of the clockpin and datapin for shiftout (using VPORT doesn't buy you anything unless the LTO can realize they're compiletime known and optimize to SBI/CBI - and in that situation we'd get it for free when fast digital IO came in). Then you assign the clock bitmask to OUTSET to bring it high, OUTSET+1 (OUTCLR) to bring it low, and for shiftin, shift val, if(*datpinptr&datbitmask){val|=bitOrder?128:1;}.... the up-front time-cost of fetching those values is on the scale of a single digitalR/W each, and then you get each write done in like 2 clock cycles... 

Assuming it's comparable to digitalWrite (ie, 2x155=310 clocks up front to stage it (though it's faster), each trip through the loop would be flying by at 2 per write/read (LDD form 4 is 2 clocks with an offset argument)... a few for the logic - maybe 20 clocks per loop, 1us, so ~8us per byte, and 15us looking up those variables instead of relooking them up every time we call digital read/write....

SpenceKonde, 2020-07-20:
```c
uint8_t shiftIn(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder) {
  uint8_t val = 0;
  uint8_t dat_bit_mask = digitalPinToBitMask(dataPin);
  uint8_t clk_bit_mask = digitalPinToBitMask(clockPin);
  if (dat_bit_mask ==NOT_A_PIN || clk_bit_mask==NOT_A_PIN) return 0; //can't do this without valid pins.
  volatile uint8_t* dat_in_reg=((volatile uint8_t*)digitalPinToPortStruct(dataPin))+0x08; //PORTx.IN
  volatile uint8_t* clk_outset_reg=((volatile uint8_t*)digitalPinToPortStruct(clockPin))+0x05; //PORTy.OUTSET 
  for (uint8_t i = 0; i != 8; i++) {
    *clk_outset_reg=clk_bit_mask; //set the bit
    if (bitOrder == LSBFIRST)
      val = (val>>1) | (*dat_in_reg&dat_bit_mask)?128:0;
    else
      val = (val<<1) | !!(*dat_in_reg&dat_bit_mask);
    *(clk_outset_reg+1)=clk_bit_mask; //hopefully optimized to ST with offset, option 4 under ST/STD. 
  }
  return val;
}

void shiftOut(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t val) {
  uint8_t dat_bit_mask = digitalPinToBitMask(dataPin);
  uint8_t clk_bit_mask = digitalPinToBitMask(clockPin);
  if (dat_bit_mask ==NOT_A_PIN || clk_bit_mask==NOT_A_PIN) return; //can't do this without valid pins.
  volatile uint8_t* dat_outset_reg=((volatile uint8_t*)digitalPinToPortStruct(dataPin))+0x05; //PORTx.IN
  volatile uint8_t* clk_outset_reg=((volatile uint8_t*)digitalPinToPortStruct(clockPin))+0x05; //PORTy.OUTSET 
  for (uint8_t i = 0; i != 8; i++)  {
    if (i) *(clk_outset_reg+1)=clk_bit_mask; //move this call up here so it will do the loop counting first
    if (bitOrder == LSBFIRST) {
      *(dat_outset_reg+((val&1)?1:0))=dat_bit_mask;
      val >>= 1;   
    }
    else{
      *(dat_outset_reg+((val&0x80)?1:0))=dat_bit_mask; 
      val <<= 1;
    }
    *clk_outset_reg=clk_bit_mask; 
    // immediately writing it low worried me, because the pulse would only be 2 clocks, I think... that's too fast.
    // so put the writing it low at top of loop, so the loop test happens between them, stretching it out a little bit
    //
  }
  *(clk_outset_reg+1)=clk_bit_mask; // and finally bring it low again... after the final loop test...
}
```

I worry about whether my implementation will be *too* fast though, such that some things that people shiftout to wouldn't be able to catch it....

SpenceKonde, 2020-07-21:
This will likely be part of the fast-digital-I/O stuff, where I'll have fastShiftIn and fastShiftOut, which - if the pins are compiletime known, will do the lookup at compiletime, otherwise it will do that at runtime - and either way, call a shift implementation passing it the ports and bitmasks.

For the basic shiftIn, I don't want to automatically use the much faster implementation, as it may be used to talk to things that can't accept data at the speed that the "fast" implementations will output it at. Assuming your implementation has better runtime, I'll use yours for the basic shiftIn/shiftOut

SpenceKonde, 2020-10-10:
Okay, I think I'll just use your implementations - they are better than the default, but not so fast that they are likely to break things.

closed 2020-10-10 by SpenceKonde