SpenceKonde/megaTinyCore #199: External crystal

avandalen, 2020-06-14:
Hi Spence

Can you add the option to use a crystal instead of the internal RC oscillator?

SpenceKonde, 2020-06-15:
Per datasheet, these parts do not support external high frequency crystal. Only a 32KHz watch crystal, or an external CLOCK source. 
In the case of the former - like, sure it could be done, but you lose all timekeeping functions (for obvious reasons), and I can't imagine there is very much interest in that speed among Arduino people - a much more realistic use case (and one which is at the top of my priority list after I finish all the super-urgent things, which I am months behind on) is an investigation of what people are missing when they report high sleep state power consumption on these parts, and also leverage the realtime clock and the hooks I added in 2.0.0 to enable a library to stop, set, and restart millis, to offer a wrapper around the sleep functions which will stop millis timing, enable the RTC with external crystal or 32KULP and an interrupt to wake on that overflowing, but if that is what wakes it, it will immediately go back to sleep, until whatever the USER configured to wake it does so, at which time it will stop the RTC, copy it's timer value, use that plus cumulative overflows of RTC since it went to sleep as well as the previous millis value to set the current millis time, and hence keep time while in an extremely low power sleep. Like you know, all sorts of consumer electronics today do, without sacrificing micros while awake like the RTC millis option does (which I now regret adding because of this idea). 
Adding support for an external crystal  is a low priority new feature that has been planned for some time. The main problem with it, though, is as soon as you offer an external clock source, you are no longer tied to the frequencies that the internal oscillator can use (hey, how about 12MHz, that's popular?) Okay, but that requires an implementation of micros for TCA, TCB, AND TCD! Why is there a specific implementation? If you can't divide 64 by the speed in MHz (us/clock cycle) evenly (ie. at compiletime), micros will be wrong, and jump around every millis rollover... You can just FORGET about doing floating point division - or ANY division of a long - in a time sensitive function like micros (on ATTinyCore when I realized this, someone had complainted that micros() was way wrong at certain speeds. So then I fixed it, and someone else complained that it was taking ~100us to return. And then I wrote specific bitshift implementations to approximate the division for every supported crystal speed that's not a power of two.) But yeah, my point there is that supporting an external clock source opens a real can of worms in terms of development effort, when I currently have a ton of new issues here I haven't even triaged, a new core that I promised people two weeks ago, and a whole bunch of other cool stuff in the works that will be much bigger impact.  

The internal oscillator on these parts is actually pretty good, especially with the ~3v and ~5v oscillator adjustments for UART baud rates and the fractional baud rate generator (which makes all those wacky UART frequencies unnecessary). Heck, if people want the main timebase more accurate, at some point I will probably implement a thing to tune the internal HF oscillator from an external watch crystal connected to the RTC, like the new DA-series parts can do automatically.

avandalen, 2020-06-15:
I think the 32kHz crystal is ok, since the clock frequency of 5MHz is derived from it by an internal PLL just like the SAMD21, which use a 32kHz cystal also.
https://www.avdweb.nl/arduino/samd21/sam-15x15

From: Spence Konde (aka Dr. Azzy) 
Sent: Monday, June 15, 2020 1:05 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

Per datasheet, these parts do not support external high frequency crystal. Only a 32KHz watch crystal, or an external CLOCK source.
In the case of the former - like, sure it could be done, but you lose all timekeeping functions (for obvious reasons), and I can't imagine there is very much interest in that speed among Arduino people - a much more realistic use case (and one which is at the top of my priority list after I finish all the super-urgent things, which I am months behind on) is an investigation of what people are missing when they report high sleep state power consumption on these parts, and also leverage the realtime clock and the hooks I added in 2.0.0 to enable a library to stop, set, and restart millis, to offer a wrapper around the sleep functions which will stop millis timing, enable the RTC with external crystal or 32KULP and an interrupt to wake on that overflowing, but if that is what wakes it, it will immediately go back to sleep, until whatever the USER configured to wake it does so, at which time it will stop the RTC, copy it's timer value, use that plus cumulative overflows of RTC since it went to sleep as well as the previous millis value to set the current millis time, and hence keep time while in an extremely low power sleep. Like you know, all sorts of consumer electronics today do, without sacrificing micros while awake like the RTC millis option does (which I now regret adding because of this idea).
Adding support for an external crystal is a low priority new feature that has been planned for some time. The main problem with it, though, is as soon as you offer an external clock source, you are no longer tied to the frequencies that the internal oscillator can use (hey, how about 12MHz, that's popular?) Okay, but that requires an implementation of micros for TCA, TCB, AND TCD! Why is there a specific implementation? If you can't divide 64 by the speed in MHz (us/clock cycle) evenly (ie. at compiletime), micros will be wrong, and jump around every millis rollover... You can just FORGET about doing floating point division - or ANY division of a long - in a time sensitive function like micros (on ATTinyCore when I realized this, someone had complainted that micros() was way wrong at certain speeds. So then I fixed it, and someone else complained that it was taking ~100us to return. And then I wrote specific bitshift implementations to approximate the division for every supported crystal speed that's not a power of two.) But yeah, my point there is that supporting an external clock source opens a real can of worms in terms of development effort, when I currently have a ton of new issues here I haven't even triaged, a new core that I promised people two weeks ago, and a whole bunch of other cool stuff in the works that will be much bigger impact.

The internal oscillator on these parts is actually pretty good, especially with the ~3v and ~5v oscillator adjustments for UART baud rates and the fractional baud rate generator (which makes all those wacky UART frequencies unnecessary). Heck, if people want the main timebase more accurate, at some point I will probably implement a thing to tune the internal HF oscillator from an external watch crystal connected to the RTC, like the new DA-series parts can do automatically.

avandalen, 2020-06-16:
Hi Spence,

Thank you for your detailed explanation. It is not easy I see..

I use the ATtiny in this battery dongle:
https://www.avdweb.nl/solar-bike/batteries/battery-hack
It could be that the Yamaha motor is sensitive to frequency deviations of the 2400 baud rate with which the data is sent to the Yamaha. 
I don't know what happens if it is not excact 2400 baud, because I dont know how sensitive (future) Yamaha motors are. Therefore, the 2400 baud rate must be precise and the internal RC oscillator may not be precise enough. To be sure, I have to connect an external crystal.

Do you have an idea how that can be done?

What is the DA series?

Would you like to manufacture and sell my Dily3217 for your own account on your Tindie shop? I have no time anymore.

have a nice day

Albert

From: Spence Konde (aka Dr. Azzy) 
Sent: Monday, June 15, 2020 1:05 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

Per datasheet, these parts do not support external high frequency crystal. Only a 32KHz watch crystal, or an external CLOCK source.
In the case of the former - like, sure it could be done, but you lose all timekeeping functions (for obvious reasons), and I can't imagine there is very much interest in that speed among Arduino people - a much more realistic use case (and one which is at the top of my priority list after I finish all the super-urgent things, which I am months behind on) is an investigation of what people are missing when they report high sleep state power consumption on these parts, and also leverage the realtime clock and the hooks I added in 2.0.0 to enable a library to stop, set, and restart millis, to offer a wrapper around the sleep functions which will stop millis timing, enable the RTC with external crystal or 32KULP and an interrupt to wake on that overflowing, but if that is what wakes it, it will immediately go back to sleep, until whatever the USER configured to wake it does so, at which time it will stop the RTC, copy it's timer value, use that plus cumulative overflows of RTC since it went to sleep as well as the previous millis value to set the current millis time, and hence keep time while in an extremely low power sleep. Like you know, all sorts of consumer electronics today do, without sacrificing micros while awake like the RTC millis option does (which I now regret adding because of this idea).
Adding support for an external crystal is a low priority new feature that has been planned for some time. The main problem with it, though, is as soon as you offer an external clock source, you are no longer tied to the frequencies that the internal oscillator can use (hey, how about 12MHz, that's popular?) Okay, but that requires an implementation of micros for TCA, TCB, AND TCD! Why is there a specific implementation? If you can't divide 64 by the speed in MHz (us/clock cycle) evenly (ie. at compiletime), micros will be wrong, and jump around every millis rollover... You can just FORGET about doing floating point division - or ANY division of a long - in a time sensitive function like micros (on ATTinyCore when I realized this, someone had complainted that micros() was way wrong at certain speeds. So then I fixed it, and someone else complained that it was taking ~100us to return. And then I wrote specific bitshift implementations to approximate the division for every supported crystal speed that's not a power of two.) But yeah, my point there is that supporting an external clock source opens a real can of worms in terms of development effort, when I currently have a ton of new issues here I haven't even triaged, a new core that I promised people two weeks ago, and a whole bunch of other cool stuff in the works that will be much bigger impact.

The internal oscillator on these parts is actually pretty good, especially with the ~3v and ~5v oscillator adjustments for UART baud rates and the fractional baud rate generator (which makes all those wacky UART frequencies unnecessary). Heck, if people want the main timebase more accurate, at some point I will probably implement a thing to tune the internal HF oscillator from an external watch crystal connected to the RTC, like the new DA-series parts can do automatically.

SpenceKonde, 2020-06-18:
(posts edited to remove duplicates of my post above in order to improve readability of this issue). 

What do you mean the 5MHz clock is derived from it via internal PLL? On the tinyAVR 0/1-series and megaAVR 0-series, all the MHz-tier clock speeds (barring an external CLOCK) are derived from the internal high-speed oscillator, which runs at either 16MHz or 20MHz depending on the OSCCFG fuse, by division by the prescaler. If you use the an external 32KHz crystal as the clock source, the clock speed can only be 32KHz at most. Which really limits what you can do with it... I 

Regarding the Dily3217 - I don't have capability to manufacture doublesided boards currently, so would have to be sending it out, and I don't have time to take something like this on right now either :-P

avandalen, 2020-06-19:
Hi Spence

For other deives, e.g. SAMD21, the 48MHz clock is derived from the 32768Khz crystel:

From: Spence Konde (aka Dr. Azzy) 
Sent: Thursday, June 18, 2020 8:59 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

(posts edited to remove duplicates of my post above in order to improve readability of this issue).

What do you mean the 5MHz clock is derived from it via internal PLL? On the tinyAVR 0/1-series and megaAVR 0-series, all the MHz-tier clock speeds (barring an external CLOCK) are derived from the internal high-speed oscillator, which runs at either 16MHz or 20MHz depending on the OSCCFG fuse, by division by the prescaler. If you use the an external 32KHz crystal as the clock source, the clock speed can only be 32KHz at most. Which really limits what you can do with it... I

Regarding the Dily3217 - I don't have capability to manufacture doublesided boards currently, so would have to be sending it out, and I don't have time to take something like this on right now either :-P

avandalen, 2020-06-19:
Dear Spence

For another atmel chip, the SAMD21, the 48MHz can be derived from the 32768kHz crystal by the use of a PLL, see the datasheet of the SAMD21 (Digital Frequency Locked Loop (DFLL48M) Operation)

But I suppose the ATtiny3217 doesnt have this?
Yes I know, selling of small cheap Arduino boards takes a much of shipping work and does not make a lot of money. Even if you outsource manufacturing and testing
Cheers
Albert

From: Spence Konde (aka Dr. Azzy) 
Sent: Thursday, June 18, 2020 8:59 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

(posts edited to remove duplicates of my post above in order to improve readability of this issue).

What do you mean the 5MHz clock is derived from it via internal PLL? On the tinyAVR 0/1-series and megaAVR 0-series, all the MHz-tier clock speeds (barring an external CLOCK) are derived from the internal high-speed oscillator, which runs at either 16MHz or 20MHz depending on the OSCCFG fuse, by division by the prescaler. If you use the an external 32KHz crystal as the clock source, the clock speed can only be 32KHz at most. Which really limits what you can do with it... I

Regarding the Dily3217 - I don't have capability to manufacture doublesided boards currently, so would have to be sending it out, and I don't have time to take something like this on right now either :-P

SpenceKonde, 2020-06-19:
That's one hell of a PLL! You sure it's actually a PLL as opposed to using the crystal to autotune an RC oscillator?

A better solution here might be an external CLOCK? (like these, you give them power, and they output a clock signal that can be used as the clock source, those little shiny silvery-golden SMD parts, real short and close to the PCB that look kind of like SMD crystals - and indeed many crystals use the same package, just with 2 of the pins being GND.  https://www.digikey.com/products/en/crystals-oscillators-resonators/oscillators/172 )

So that's one approach

Another would be to use an external crystal for the RTC, and use that and a TCB to see how far off it is and adjust. For your purposes, it's probably best to tweak the UART baud value instead of using it to cal the oscillator because you have much finer granularity there thanks to the fractional baud rate generator....

But like, how far off even are you? The internal oscillator, combined with the UART error values in the sigrow for 3v and 5v, and the fractional buad rate generator.... it's pretty damned good. It's worlds apart from the classic AVRs!

The DA-series is what probably would have been the ATmega 1-series if microchip marketeers hadn't taken over. Check out the datasheet AVR128DA64 is the top end part (lean back while reading it so you don't drool into your keyboard). One of the features it has is "autotune" with a watch crystal - it uses that to time the main clock, and then automatically adjusts the calibration bits appropriately.; the version of my boards with the crystal for the 32/48-pin parts will be here next week, and I'll play with it then if I haven't gotten the 64-pin parts working yet. I'm pretty close to releasing DACore to add Arduino support :-)

avandalen, 2020-06-19:
Hi Spence,

See here:
https://microchipdeveloper.com/32arm:samd21-code-gcc-clock-system-dfll48m-init

Im thinking to switch to the STM8 and STM32. These microcontrollers are made Arduino compatible and also accept an external crystal, that is easy for me.
https://github.com/stm32duino
https://github.com/stm32duino/Arduino_Core_STM8

see for instance the Blue Pill SMT32
https://nl.aliexpress.com/item/32326304541.html?src=google&src=google&albch=shopping&acnt=494-037-6276&isdl=y&slnk=&plac=&mtctp=&albbt=Gploogle_7_shopping&aff_atform=google&aff_short_key=UneMJZVf&&albagn=888888&albcp=6459980570&albag=76980386066&trgt=743612850714&crea=nl32326304541&netw=u&device=c&albpg=743612850714&albpd=nl32326304541&gclid=CjwKCAjwxLH3BRApEiwAqX9arTwy53rD0bMycQcDffKUmKwX3C8BI4R85LXQvKMNCH0YTLyklJA-YxoCzewQAvD_BwE&gclsrc=aw.ds

Albert

STM32 "Blue Pill

From: Spence Konde (aka Dr. Azzy) 
Sent: Friday, June 19, 2020 12:55 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

That's one hell of a PLL! You sure it's actually a PLL as opposed to using the crystal to autotune an RC oscillator?

A better solution here might be an external CLOCK? (like these, you give them power, and they output a clock signal that can be used as the clock source, those little shiny silvery-golden SMD parts, real short and close to the PCB that look kind of like SMD crystals - and indeed many crystals use the same package, just with 2 of the pins being GND. https://www.digikey.com/products/en/crystals-oscillators-resonators/oscillators/172 )

So that's one approach

Another would be to use an external crystal for the RTC, and use that and a TCB to see how far off it is and adjust. For your purposes, it's probably best to tweak the UART baud value instead of using it to cal the oscillator because you have much finer granularity there thanks to the fractional baud rate generator....

But like, how far off even are you? The internal oscillator, combined with the UART error values in the sigrow for 3v and 5v, and the fractional buad rate generator.... it's pretty damned good. It's worlds apart from the classic AVRs!

The DA-series is what probably would have been the ATmega 1-series if microchip marketeers hadn't taken over. Check out the datasheet AVR128DA64 is the top end part (lean back while reading it so you don't drool into your keyboard). One of the features it has is "autotune" with a watch crystal - it uses that to time the main clock, and then automatically adjusts the calibration bits appropriately.; the version of my boards with the crystal for the 32/48-pin parts will be here next week, and I'll play with it then if I haven't gotten the 64-pin parts working yet. I'm pretty close to releasing DACore to add Arduino support :-)

SpenceKonde, 2020-06-19:
Yeah, that's the same thing the DA-series parts will have:  an automatic
mechanism to adjust OSCCAL (or equiv) based on crystal - it's not a PLL.

You can do that yourself on any "modern AVR" part with the RTC or PIT and
crystal and one of the timers derived from system clock. But I was saying,
you have less granularity that way that you do if you instead use that
timing difference to calculate the perfect BAUD divisor directly - you'd
get a factor to multiply the baud rate calculated with the standard methods
(minus the correction factors in the SIGROW, which are supposed to do
exactly this at certain known conditions.

Again I ask, have you measured how close the baud rate you need is (eg,
using a scope)? I think you are overestimating the magnitude of the error
you will have to deal with - the internal oscillators and their cal options
are WAY better than the classic AVRs. That said, if it's being run on
varying voltage and/or varying temperatures, that doesn't work as well...

avandalen, 2020-06-21:
im on holiday, and come back later

From: Spence Konde (aka Dr. Azzy) 
Sent: Friday, June 19, 2020 11:55 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

Yeah, that's the same thing the DA-series parts will have: an automatic
mechanism to adjust OSCCAL (or equiv) based on crystal - it's not a PLL.

You can do that yourself on any "modern AVR" part with the RTC or PIT and
crystal and one of the timers derived from system clock. But I was saying,
you have less granularity that way that you do if you instead use that
timing difference to calculate the perfect BAUD divisor directly - you'd
get a factor to multiply the baud rate calculated with the standard methods
(minus the correction factors in the SIGROW, which are supposed to do
exactly this at certain known conditions.

Again I ask, have you measured how close the baud rate you need is (eg,
using a scope)? I think you are overestimating the magnitude of the error
you will have to deal with - the internal oscillators and their cal options
are WAY better than the classic AVRs. That said, if it's being run on
varying voltage and/or varying temperatures, that doesn't work as well...

avandalen, 2020-07-04:
I come back soon..

From: Spence Konde (aka Dr. Azzy) 
Sent: Friday, June 19, 2020 11:55 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

Yeah, that's the same thing the DA-series parts will have: an automatic
mechanism to adjust OSCCAL (or equiv) based on crystal - it's not a PLL.

You can do that yourself on any "modern AVR" part with the RTC or PIT and
crystal and one of the timers derived from system clock. But I was saying,
you have less granularity that way that you do if you instead use that
timing difference to calculate the perfect BAUD divisor directly - you'd
get a factor to multiply the baud rate calculated with the standard methods
(minus the correction factors in the SIGROW, which are supposed to do
exactly this at certain known conditions.

Again I ask, have you measured how close the baud rate you need is (eg,
using a scope)? I think you are overestimating the magnitude of the error
you will have to deal with - the internal oscillators and their cal options
are WAY better than the classic AVRs. That said, if it's being run on
varying voltage and/or varying temperatures, that doesn't work as well...

avandalen, 2020-07-04:
I still think it is a pll / fll with the SAMD21:
17.6.7 Digital Frequency Locked Loop (DFLL48M) Operation The DFLL48M can operate in both open-loop mode and closed-loop mode. In closed-loop mode, a low-frequency clock with high accuracy can be used as the reference clock to get high accuracy on the output clock (CLK_DFLL48M). The DFLL48M can be used as a source for the generic clock generators, as described in the GCLK – Generic Clock Controller. Related Link

From: Spence Konde (aka Dr. Azzy) 
Sent: Friday, June 19, 2020 11:55 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

Yeah, that's the same thing the DA-series parts will have: an automatic
mechanism to adjust OSCCAL (or equiv) based on crystal - it's not a PLL.

You can do that yourself on any "modern AVR" part with the RTC or PIT and
crystal and one of the timers derived from system clock. But I was saying,
you have less granularity that way that you do if you instead use that
timing difference to calculate the perfect BAUD divisor directly - you'd
get a factor to multiply the baud rate calculated with the standard methods
(minus the correction factors in the SIGROW, which are supposed to do
exactly this at certain known conditions.

Again I ask, have you measured how close the baud rate you need is (eg,
using a scope)? I think you are overestimating the magnitude of the error
you will have to deal with - the internal oscillators and their cal options
are WAY better than the classic AVRs. That said, if it's being run on
varying voltage and/or varying temperatures, that doesn't work as well...

avandalen, 2020-07-05:
thank you for you good help, i will read this soon..
I was a bit overworked with my hobby projects and need to slow down.

From: Spence Konde (aka Dr. Azzy) 
Sent: Friday, June 19, 2020 11:55 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

Yeah, that's the same thing the DA-series parts will have: an automatic
mechanism to adjust OSCCAL (or equiv) based on crystal - it's not a PLL.

You can do that yourself on any "modern AVR" part with the RTC or PIT and
crystal and one of the timers derived from system clock. But I was saying,
you have less granularity that way that you do if you instead use that
timing difference to calculate the perfect BAUD divisor directly - you'd
get a factor to multiply the baud rate calculated with the standard methods
(minus the correction factors in the SIGROW, which are supposed to do
exactly this at certain known conditions.

Again I ask, have you measured how close the baud rate you need is (eg,
using a scope)? I think you are overestimating the magnitude of the error
you will have to deal with - the internal oscillators and their cal options
are WAY better than the classic AVRs. That said, if it's being run on
varying voltage and/or varying temperatures, that doesn't work as well...

avandalen, 2020-07-15:
Hi Spence

>> Again I ask, have you measured how close the baud rate you need is
I cant measure this myself, I use an ancient tektronic scope 100MHz from 40 years old.

Normally the baud rate frequency is not very critical. But the ATtiny is used in a Yamaha battery hack, where the Yamaha motor is fooled because a different battery is used than the original one. The battery sends serial data to the motor and if the motor receives erroneous data it will not work. It turns out that the baud rate in this case must be very precise, therefore a crystal is needed.

I will use another processor ATMEGA328PB-AU with crystal.

See here the baud rate errors:
https://cache.amobbs.com/bbs_upload782111/files_22/ourdev_508497.html

Have a nice day
Albert

From: Spence Konde (aka Dr. Azzy) 
Sent: Friday, June 19, 2020 11:55 PM
To: SpenceKonde/megaTinyCore 
Cc: Albert van Dalen ; Author 
Subject: Re: [SpenceKonde/megaTinyCore] External crystal (#199)

Yeah, that's the same thing the DA-series parts will have: an automatic
mechanism to adjust OSCCAL (or equiv) based on crystal - it's not a PLL.

You can do that yourself on any "modern AVR" part with the RTC or PIT and
crystal and one of the timers derived from system clock. But I was saying,
you have less granularity that way that you do if you instead use that
timing difference to calculate the perfect BAUD divisor directly - you'd
get a factor to multiply the baud rate calculated with the standard methods
(minus the correction factors in the SIGROW, which are supposed to do
exactly this at certain known conditions.

Again I ask, have you measured how close the baud rate you need is (eg,
using a scope)? I think you are overestimating the magnitude of the error
you will have to deal with - the internal oscillators and their cal options
are WAY better than the classic AVRs. That said, if it's being run on
varying voltage and/or varying temperatures, that doesn't work as well...

SpenceKonde, 2020-10-04:
External clock (eg an external clock going to the CLKIN pin) will be added in 3.0.0, coming sooner than one thinks. 

Those baud rate errors have nothing to do with clock accuracy. Those are the error assuming a perfect (100% accurate) clock, on a "classic" AVR, due to the granularity of the baud rate divisor on those parts. This just inspired me to make such a spreadsheet, if you can open .xlsx files...
[BaudRatesAVR.xlsx](https://github.com/SpenceKonde/megaTinyCore/files/5322562/BaudRatesAVR.xlsx)

That shows the same thing as the wormfood tables (on the bottom);

But like the wormfood tables, they don't account for clock accuracy - clock accuracy errors are on top of that....

SpenceKonde, 2020-10-04:
That said, with an external watch crystal, a TCB, and the PIT event source, you have all the tools to correct the USARTn.BAUD value to the optimal for what the clock is running at under any given conditions.... 

Clock the TCB off system clock, and configure for input capture using PIT_DIV_64 as the event source.... You expect 39062 or 39063... but OFC it won't be that, because the internal oscillator isn't perfect (though IME it's usually within a percent under normal conditions at least).... say you measure 39500, indicating the oscillator is running just over 1% fast.... 

Call Serial.begin as usual, before sending anything, correct the baud rate...
USART0.BAUD=(  (((unsigned long) USART0.BAUD)*(MeasuredValueFromTCB) ) + 19531)/39063; (the +19531 is to ensure it's rounded correctly)

Don't have input capture code handy, but it's easy-peasy on the TCB's - they are everything you could wish for in an input capture timer..... input capture pulse width measurement mode, all just makes sense. Very well designed timer IMO...

SpenceKonde, 2020-10-10:
External clock support is added will be in 2.10 (there's no 3.0.0 yet - that change got scrubbed)

closed 2020-10-10 by SpenceKonde

SpenceKonde, 2020-11-05:
If you're still looking at this, by any chance:
* at ~3 or ~5v and room temperature, the sigrow corrections will get the baud rate to within 1 part per 1024 (in an unknown direction), ie, +/- 0.097%. With U2X=0 (which megaTinyCore will do for you, unless you're asking for a baud rate so fast that U2X will do a better job). With target of 2400 baud, the calculation itself will be off by 7~13ppm (ie, it is negligible - all of the baud error is from the internal oscillator, which -  at v or 5v with the sigrow correction (used by default on all parts with 8k or more flash, and on all parts prior to 2.1.4, when I realized it was costing too damned much flash on 4k and 2k parts because someone complained about that).... You're +/- 0.1% under favorable conditions, They only guarantee +/-2% over a wider voltage and temperature range (see electrical characteristics on datasheet), but if you use the shape of the frequency-voltage curve from the typical characteristics section of the datasheet, you could do a lot better than that (dependence on voltage is much stronger than temperature).
* HOWEVER that is the AVERAGE.... when I was running tests this weekend, I discovered that in any given 512th of a second measured against a watch crystal.... one of them wasn't running at quite an average speed........ (measurement was done using PIT interrupt - the basic measuring scheme is shown in the attached sketch, though the attached sketch is aimed at something entirely different, namely determining the full extent of the frequency range of the internal oscillator - as usual, it can be tweaked much further than the datasheet admits - but instead of changing the cal byte, you can just take measurements at the factory calibration and calculate what the actual clock speed woks out to) 
 and it can transition - it would  on a timescale of a 256th of a second or so from being -0.1% to +0.1% from the average... that it, it changes fast enough that you can't really keep it "right on target", but using the methods I described, you could probably keep it within 0.5% if you calibrated it against a watch crystal, and 1% if you didn't.  
[OSC20MCALIBA.zip](https://github.com/SpenceKonde/megaTinyCore/files/5486661/OSC20MCALIBA.zip). Of course it is also possible that the variation I recorded was due to problems with my crystal circuit

* On a classic AVR (eg, 328p) with a crystal, assuming you get a decent one, you could have total of stability and tolerance variation of 50 ppm - say you're running at 16 mhz, per that table you linked, and my spreadsheet (which rounds less), you'd be within 0.01%. If it instead used a resonator (which is what's on nano, uno, clones thereof, and most pro mini clones, looks like a tiny crystal, and is very often mistaken for a crystal, even by people who one might expect to know better), the frequency tolerance would be +/-1-2% - much worse than an actual crystal - but also cheaper and smaller, hence what gets used on the Arduino boards.... 
* Finally, on a tinyAVR 0/1-series with an external clock for the system clock (supported as of 2.1.x versions), you could easily get the same 50 ppm accuracy on the system clock... and there the fractional baud rate generator would get a chance to shine, as it's baud rate calculations only introduce 13 ppm of deviation from the target baud if we're targeting 2400 baud and use 16 MHz clock, 10 ppm with 20 MHz clock. For 16 MHz (but not 20 - BAUD register would overflow with the number you'd need to set it to), you can:
```
Serial.begin(2400);
USART0.BAUD=53333; //calculated baud rate for CLK2X mode assuming perfect 16 MHz clock. 
USART0.CTRLB|=USART_RXMODE_CLK2X_gc; //enable CLK2X mode
```
This would bring the deviation from the calculation down to ~6 ppm, and you can get oscillators with the frequency within 20ppm....  (megaTinyCore doesn't use CLK2X unless it needs to in order to get a "pretty accurate" baud rate (it doubles the max baud in addition to halving the TX baud rate error, which gets much higher as the baud rate approaches the maximum possible at a given clock speed; the cutoff megaTinyCore uses is 38400*(speed in MHz) - the reason we don't use it whenever we can is that it makes RX twice as sensitive to baud rate error from whatever it's receiving from.... 

That all said - I think the ideal course of action here is for you to actually investigate how accurate the baud rate on the official hardware you're trying to copy is. A hardware teardown looking for a crystal would be a good start. If you don't find one,, or if all you find is a watch crystal (it's unlikely they'd derive the baud rate from that, though it's possible), then it's using an RC oscillator to generate it's baud rate, and their accuracy is probably 1-2% at best, meaning trying to do better than that is unnecessary. If you find one of those blue 2-terminal things marked as a crystal from the reference designators, that's a ceramic resonator, not a crystal, so 1-2% accuracy.

SpenceKonde, 2020-11-05:
One note for the hardware teardown: I have seen 6 MHz crystals in the "garbage can" form factor (commonly used for watch crystals) in commercial hardware, so just because the crystal you find is small and cylindrical doesn't mean it's not a MHz crystal used as the system clock.

avandalen, 2020-11-06:
Hi Spence, thank you very much for your comprehensive help, this will help solve the baud rate problem.