SpenceKonde/megaTinyCore #200: multiple definition of nvm_write_byte error

T94T, 2020-06-19:
I just tried to include your EEPROM library in a sub class. Unfortunately it can't get compiled with errors listed below. Please find a very simple test project attached. With `#include <EEPROM.h>` commented out in the `Test_Class.h` file the project gets compiled without errors.

> In function 'nvm_write_byte(unsigned int, unsigned char)':
> (.text+0x0): multiple definition of 'nvm_write_byte(unsigned int, unsigned char)'
> sketch/Test.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
> collect2: error: ld returned 1 exit status

[Test.zip](https://github.com/SpenceKonde/megaTinyCore/files/4805485/Test.zip)

SpenceKonde, 2020-06-26:
The hell? It's like the include guard isn't working... Not sure what I'm doing wrong here...

SpenceKonde, 2020-07-05:
I looked at  this again - no idea how to fix it! 

Anyone know?

Yardie-, 2020-07-06:
Not sure if this will help 
I found a small error in the EEPROM.h file though

line 61 
` EERef &operator=( uint8_t in )     { return nvm_write_byte( (uint8_t*) index, in ), *this;  }
` 
probably should be 

`  EERef &operator=( uint8_t in )     { return nvm_write_byte( (uint16_t) index, in ), *this;  }
`

T94T, 2020-07-11:
The error still remains. But I found a solution/workaround.
[TestWorking.zip](https://github.com/SpenceKonde/megaTinyCore/files/4906355/TestWorking.zip)

SpenceKonde, 2020-09-25:
It was just pointed out that there are some issues with EEPROM.h relating to workarounds (this macro, indirectly, is one of them). This is simple to sort out and will be fixed in next release. 

I may end up ripping the user row stuff out and putting it into it's own library. I think I will, that way I could have a unified API on DxCore and here for userrow programming, even though the underlying implementation has to be very different.

T94T, 2020-09-27:
Thank you. So I'm waiting for the next release.

SpenceKonde, 2020-10-03:
Fixed in github version will be in 2.0.6

closed 2020-10-03 by SpenceKonde