SpenceKonde/megaTinyCore #201: Export issue

robshep, 2020-06-25:
I get the following error when trying to Sketch > "Export compiled binary" 

```
opt/arduino-1.8.12/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.12/hardware -hardware /home/rob/.arduino15/packages -tools /opt/arduino-1.8.12/tools-builder -tools /opt/arduino-1.8.12/hardware/tools/avr -tools /home/rob/.arduino15/packages -built-in-libraries /opt/arduino-1.8.12/libraries -libraries /home/rob/Arduino/libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=3216,clock=1,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,millis=enabled,uartvoltage=5v -vid-pid=10C4_EA60 -ide-version=10812 -build-path /tmp/arduino_build_51199 -warnings=none -build-cache /tmp/arduino_cache_727724 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/home/rob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/rob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=/home/rob/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/rob/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /home/rob/Arduino/Blink/Blink.ino
/opt/arduino-1.8.12/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.12/hardware -hardware /home/rob/.arduino15/packages -tools /opt/arduino-1.8.12/tools-builder -tools /opt/arduino-1.8.12/hardware/tools/avr -tools /home/rob/.arduino15/packages -built-in-libraries /opt/arduino-1.8.12/libraries -libraries /home/rob/Arduino/libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=3216,clock=1,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,millis=enabled,uartvoltage=5v -vid-pid=10C4_EA60 -ide-version=10812 -build-path /tmp/arduino_build_51199 -warnings=none -build-cache /tmp/arduino_cache_727724 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/home/rob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/rob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=/home/rob/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/rob/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -verbose /home/rob/Arduino/Blink/Blink.ino
Using board 'atxy6' from platform in folder: /home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2
Using core 'arduino' from platform in folder: /home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2
Detecting libraries used...
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny3216 -DF_CPU=1000000L -DARDUINO=10812 -DARDUINO_attinyxy6 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERD0_A0 -DNO_EXTERNAL_I2C_PULLUP -DUSE_TIMERD0_PWM -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino/api/deprecated -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/variants/txy6 /tmp/arduino_build_51199/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny3216 -DF_CPU=1000000L -DARDUINO=10812 -DARDUINO_attinyxy6 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERD0_A0 -DNO_EXTERNAL_I2C_PULLUP -DUSE_TIMERD0_PWM -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino/api/deprecated -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/variants/txy6 /tmp/arduino_build_51199/sketch/Blink.ino.cpp -o /tmp/arduino_build_51199/preproc/ctags_target_for_gcc_minus_e.cpp
/opt/arduino-1.8.12/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_51199/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=attiny3216 -DF_CPU=1000000L -DARDUINO=10812 -DARDUINO_attinyxy6 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERD0_A0 -DNO_EXTERNAL_I2C_PULLUP -DUSE_TIMERD0_PWM -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino/api/deprecated -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/variants/txy6 /tmp/arduino_build_51199/sketch/Blink.ino.cpp -o /tmp/arduino_build_51199/sketch/Blink.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /tmp/arduino_cache_727724/core/core_37e0d9c13be6c7679909200a975cb1ef.a
Linking everything together...
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-gcc -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x0 -mmcu=attiny3216 -o /tmp/arduino_build_51199/Blink.ino.elf /tmp/arduino_build_51199/sketch/Blink.ino.cpp.o /tmp/arduino_build_51199/../arduino_cache_727724/core/core_37e0d9c13be6c7679909200a975cb1ef.a -L/tmp/arduino_build_51199 -lm
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-objcopy -O binary -R .eeprom /tmp/arduino_build_51199/Blink.ino.elf /tmp/arduino_build_51199/Blink.ino.bin
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /tmp/arduino_build_51199/Blink.ino.elf /tmp/arduino_build_51199/Blink.ino.eep
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-objcopy -O ihex -R .eeprom /tmp/arduino_build_51199/Blink.ino.elf /tmp/arduino_build_51199/Blink.ino.hex
chmod +x /home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/create_disassembler_listing.sh
/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/create_disassembler_listing.sh /home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-objdump "--disassemble --source --line-numbers --demangle --section=.text" /tmp/arduino_build_51199/Blink.ino.elf /tmp/arduino_build_51199/Blink.ino.lst
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-size -A /tmp/arduino_build_51199/Blink.ino.elf
Sketch uses 802 bytes (2%) of program storage space. Maximum is 32768 bytes.
Global variables use 10 bytes (0%) of dynamic memory, leaving 2038 bytes for local variables. Maximum is 2048 bytes.
chmod +x /home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/delete_merged_output.sh 
/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/delete_merged_output.sh false /tmp/arduino_build_51199/Blink.ino.with_bootloader.hex 
rm: cannot remove '/tmp/arduino_build_51199/Blink.ino.with_bootloader.hex': No such file or directory
Error compiling.
```

I'm not using optiboot.

Thanks for any suggestions

SpenceKonde, 2020-06-26:
Interesting. Like, I think that has always been erroring - what's weird for you is that it seems to actually be interfering with the rest of opperation. On boards with no bootloader, I see an error about not being able to remove that file go by - but everything else goes on it's merry way. I'm no Linuxwizard though - I don't know how to interpret this, but if you or someone else can figure out what needs to be done

...

...

Wait, why the fuck am I even using that script on this core at all?! The merged hex images are dumb, but there's no reason they wouldn't work - all these parts have hw bootoader support... so why do I have that debris in here?

robshep, 2020-06-26:
:)

If I scoop the hex file out of the build directories it loads and runs fine as far as I can tell. 
But I'm not familiar with what's going on and so wondering if there is something crucial not being done due to the abort.

Incidentally the `delete_merged_output.sh` file doesn't seem to have much to boast about, but switching `rm` for `rm -f` will let it continue silently if it's OK for that file not to exist at the point you want to delete it. 

Alternatively check to see if it exists before attempting to delete it :

```
#!/bin/bash
  
if [ "$1" == "false" -a  -e "$2"  ]; then
  rm "$2"
fi
```

robshep, 2020-06-26:
Using that last option resulted in a `.txy6.hex` file next to my source `.ino`
I assume that's good, I've never exported before.

SpenceKonde, 2020-06-26:
I don't think the delete merged output stuff should be in megaTinyCore at all - these boards have hardware bootloader support, so for boards that do have a bootloader, no reason not to include it.... and for ones that don't, there;s no file created to delete. Try what I did to platform.txt now, and see if it works - you should get a .hex and a .lst file in the sketch folder.

robshep, 2020-06-26:
I'm sorry could you expand a little on how I should try this.....  just run board manager and pull in upgrade? 

or something else...

SpenceKonde, 2020-06-26:
Modify platform.txt to make it match what's in the repo - from previous posts, I was under the impression that you;d done a manual installation, since you were proposing modifications to the files...

robshep, 2020-06-26:
OK,
Well it runs OK.
I don't have boards in front of me, but I'll try the HEX later.
Thanks

```
/opt/arduino-1.8.12/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.12/hardware -hardware /home/rob/.arduino15/packages -tools /opt/arduino-1.8.12/tools-builder -tools /opt/arduino-1.8.12/hardware/tools/avr -tools /home/rob/.arduino15/packages -built-in-libraries /opt/arduino-1.8.12/libraries -libraries /home/rob/Arduino/libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=3216,clock=1,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,millis=enabled,uartvoltage=5v -ide-version=10812 -build-path /tmp/arduino_build_314247 -warnings=none -build-cache /tmp/arduino_cache_346493 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/rob/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/rob/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/home/rob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/rob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/rob/Arduino/Blink/Blink.ino
/opt/arduino-1.8.12/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.12/hardware -hardware /home/rob/.arduino15/packages -tools /opt/arduino-1.8.12/tools-builder -tools /opt/arduino-1.8.12/hardware/tools/avr -tools /home/rob/.arduino15/packages -built-in-libraries /opt/arduino-1.8.12/libraries -libraries /home/rob/Arduino/libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=3216,clock=1,bodvoltage=1v8,bodmodeactive=disabled,bodmodesleep=disabled,eesave=enable,millis=enabled,uartvoltage=5v -ide-version=10812 -build-path /tmp/arduino_build_314247 -warnings=none -build-cache /tmp/arduino_cache_346493 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/home/rob/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/rob/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=/home/rob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/rob/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /home/rob/Arduino/Blink/Blink.ino
Using board 'atxy6' from platform in folder: /home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2
Using core 'arduino' from platform in folder: /home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2
Detecting libraries used...
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny3216 -DF_CPU=1000000L -DARDUINO=10812 -DARDUINO_attinyxy6 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERD0_A0 -DNO_EXTERNAL_I2C_PULLUP -DUSE_TIMERD0_PWM -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino/api/deprecated -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/variants/txy6 /tmp/arduino_build_314247/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=attiny3216 -DF_CPU=1000000L -DARDUINO=10812 -DARDUINO_attinyxy6 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERD0_A0 -DNO_EXTERNAL_I2C_PULLUP -DUSE_TIMERD0_PWM -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino/api/deprecated -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/variants/txy6 /tmp/arduino_build_314247/sketch/Blink.ino.cpp -o /tmp/arduino_build_314247/preproc/ctags_target_for_gcc_minus_e.cpp
/opt/arduino-1.8.12/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_314247/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=attiny3216 -DF_CPU=1000000L -DARDUINO=10812 -DARDUINO_attinyxy6 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERD0_A0 -DNO_EXTERNAL_I2C_PULLUP -DUSE_TIMERD0_PWM -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino/api/deprecated -I/home/rob/.arduino15/packages/megaTinyCore/hardware/megaavr/2.0.2/cores/arduino -I/home/rob/.arduino15/packages/megaTinyCore/hardwar [... cut ...] gle --section=.text" /tmp/arduino_build_314247/Blink.ino.elf /tmp/arduino_build_314247/Blink.ino.lst
/home/rob/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-size -A /tmp/arduino_build_314247/Blink.ino.elf
Sketch uses 812 bytes (2%) of program storage space. Maximum is 32768 bytes.
Global variables use 10 bytes (0%) of dynamic memory, leaving 2038 bytes for local variables. Maximum is 2048 bytes.
cp /tmp/arduino_build_314247/Blink.ino.lst /home/rob/Arduino/Blink 
```

SpenceKonde, 2020-07-05:
Since it now generates the hex without errors, this should be good. None of the board defs in megaTinyCore need to delete merged hex files anyway, so that helper script is simply removed. 

2.0.4 will also have greatly improved naming for those exports :-)

closed 2020-07-05 by SpenceKonde