SpenceKonde/megaTinyCore #202: BOD fuse issues

EclipseBoom, 2020-07-03:
Using the T1616.  I've recently run into two different applications with issues related to the BOD and startup for dirty/slow power rise.  These newer chips seem to be more sensitive to this as I never had problems like this with the older AVRs.

Basically the BODCFG.SAMPFREQ and SYSCFG1.SUT seem to have a critical effect on startup with slowly rising power.  In my application a high-impedence power supply turns on but as soon as the BOD is cleared and the CPU starts up it loads the power and drops below the BOD cutoff.  This then immediately reduces the load and the power jumps up again.  This seems to freak out the CPU and it locks up even when the power eventually climbs above the BOD limit and stays there.  With UPDI active I don't even think I can just add a classic RESET line delay circuit...

The MegaTinyCore sets (leaves) these fuses to the fastest settings of 1kHz sample and 4mS delay.  If I manually reduce them using the AVRStudio programmer tool to 125Hz sample rate and 64mS startup delay I get much more reliable startup.  I also can't set the BOD to 1.8V at all.  2.6V  or higher seems to be the best.

I'm wondering if these slower startup settings shouldn't be the default?  How many applications really need instant power-on such that 64mS is too slow?

Anyway not really a bug.  Just a different possible configuration setting that might work well for the average user.

SpenceKonde, 2020-10-03:
Okay, executive decision - will use the most conservative settings.

closed 2020-10-03 by SpenceKonde

SpenceKonde, 2020-10-10:
Change in plans, what the hell, might as well throw in menu options to set these. I combined with cleaning up the options for BOD mode to remove unreasonable options.

commit bb29cee6 (2020-10-03): Conservative settings for SUT, 125Hz BOD sample rate

Fixes #202