SpenceKonde/megaTinyCore #203: mistake in wire.h

concon, 2020-07-03:
<C:\Users\lcagl\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.2\libraries\Wire\src/Wire.h: In function 'word readValue(byte, byte)':
C:\Users\lcagl\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.2\libraries\Wire\src/Wire.h:75:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)
     uint8_t requestFrom(int, int);
             ^~~~~~~~~~~
C:\Users\lcagl\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.0.2\libraries\Wire\src/Wire.h:73:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, size_t)
     uint8_t requestFrom(uint8_t, size_t);
             ^~~~~~~~~~~

I get it when compile, do you have any idea?

SpenceKonde, 2020-07-03:
You have included neither the full error message (there are a few lines
before that), nor the code that generated it. Come on man!

Can you test whether the same code works on other "modern AVR" based boards
(what the Arduino team has at times called "megaavr", though that term is
inaccurate; the technical name for the applicable parts would be the
megaAVR 0-series and tinyAVR 0/1-series), such as the Arduino Nano Every or
Hans' (MCUDude) MegaCoreX boards? Looking at the Wire.h files for those, I
think all of them will have the same issue with the code you are attempting
to compile. The thing that's erroring is part of what I copied from the
official "megaavr" core - they changed the types of the arguments for
requestFrom() from:

    uint8_t requestFrom(uint8_t, uint8_t);
    uint8_t requestFrom(uint8_t, uint8_t, uint8_t);
    uint8_t requestFrom(uint8_t, uint8_t, uint32_t, uint8_t, uint8_t);
    uint8_t requestFrom(int, int);
    uint8_t requestFrom(int, int, int);
to
    uint8_t requestFrom(uint8_t, size_t);
    uint8_t requestFrom(uint8_t, size_t, bool);
    uint8_t requestFrom(int, int);
    uint8_t requestFrom(int, int, int);

(Interesting, the weirdo 5-argument form of requestFrom() found in the
official Arduino AVR core but not documented in the Wire reference is also
broken by the new version of the library - what that's for is unclear to
me...)

I'm not really sure what the purpose of the uint8_t requestFrom(int, int)
and uint8_t requestFrom(int, int, int) versions is - I've opened a thread
about this matter on the forums...
https://forum.arduino.cc/index.php?topic=693716

concon, 2020-07-04:
Sory,

#include <Wire.h>
#include "FDC2214.h"
#include <EEPROM.h>

#define __AVR_ATtiny1604__

#define F_CPU 1000000L
//#define TWI_FREQ 100000L

const byte SDA_PIN = 6;
const byte SCL_PIN = 7;

//variables
float k;
float CRE; 
float CRL;
float CLEV;
float CLEVbos;
float SMax;
float Max;

int HREF;
int i;
byte o;                       // sensor output type 0-4/20 ma. 
String entry;                 // String that contain the entered character(s) from serial terminal keyboard
int sensorValue = 0;          // the sensor value  
int sonraki= 0;
int gosterge = 0;

#define CHAN_COUNT 3
float cap[CHAN_COUNT]; // variable to store data from FDC

FDC2214 capsense(FDC2214_I2C_ADDR_0); // Use FDC2214_I2C_ADDR_1 

// ###
void setup() {
  Wire.begin();
  Wire.setClock(100000L);
  delay(1000);
  Serial.begin(57600);
  Serial.println("\nFDC2214 Capacitive liquid level Sensor");
// ### Start I2C 
 
  pinMode (1,OUTPUT); // dac output
  analogWrite(1,0);   
  //pinMode (3,OUTPUT); // fdc2214'ün sd pini
  //digitalWrite(3, LOW); // FDC2214 Power ON
  
  
  // ### Start FDC
  // Start FDC2214 with 4 channels init
  //bool capOk = capsense.begin(0x7, 0x5, 0x5, false); //setup three channels, autoscan with 3 channels, deglitch at 10MHz, external oscillator 
  // Start FDC2214 with 4 channels init
   bool capOk = capsense.begin(0x7, 0x5, 0x5, true); //setup three channels, autoscan with 3 channels, deglitch at 10MHz, internal oscillator 
  if (capOk) Serial.println("Sensor OK");  
  else Serial.println("Sensor Fail");                
}

SpenceKonde, 2020-07-05:
I've added code for backwards compatibility that makes this sketch work. It will be in next release.
66ae540ebe7dfa4534962ab453c191af15110fb3

SpenceKonde, 2020-07-05:
I don't know *why* they felt a need to change wire in a breaking way for "megaavr" - but I reversed that, and added compatibility calls so code written the "new" way still works too.

SpenceKonde, 2020-07-05:
Oh - and you should *NOT* define `__AVR_ATtiny1604__` in your sketch. 
You should *most definitely* not define #define F_CPU 1000000L - that won't actually change the speed it's running at! Use the tools -> Clock menu - and be sure to "burn bootloader" first too (since chips come pre-set to use 20MHz derived speeds, and 1MHz assumes 16MHz derived speeds). 
And your sketch needs a loop() function too.

closed 2020-07-05 by SpenceKonde