SpenceKonde/megaTinyCore #204: I2C what am I doing wrong

Yardie-, 2020-07-05:
Hi I have just started playing with the 3216 and am having trouble with the I2C interface.
Code that works on ATTiny85 and Arduino nanos doesn't seem to work.
Serial is fine.

I setup the 3216 as i2c master and it can't find anything.
As a slave it is not visible.

I have been doing lots of various tests. I will do more and clarify it as much as I can 
however are here any obvious gotchas

External pullups didn't seem to change anything.

ideas?

SpenceKonde, 2020-07-05:
Using the correct set of I2C pins? (see Wire.swap() )

I2C definitely should be working...

SpenceKonde, 2020-07-05:
Post minimum example that reproduces the problem?

Yardie-, 2020-07-06:
Thanks for the reply Spence.
Even the default examples don't work.
I am using the default pins with  
(D)8 SDA  10┨~     ~┠11 (D) 9 A11 SCL
pins 
I did read the docs and code wise the only thing I didn't try was adding Wire.swap() before begin
as I am using the default pins..
Perhaps it's the chip I'll try another chip and swapping the pins to alternates on that chip and get back to you.
Using the analogRead on SDA gives roughly the same results as the working ones however it ain't pretty and it's not really a good comparison. I need to source an oscilloscope to check it properly.
I was really just checking to see if anyone else had this problem.
Thanks.
As I sort it I will report back the process.

Yardie-, 2020-07-06:
OK New chip = no problems. 
I'm an idiot.
It must have just been some kind of bad solder/bad handling/operator error type problem.

Once again thanks @SpenceKonde for your help it was invaluable.

I found a small error in the EEPROM.h file though 
Yes it is a new issue but I didn't think it warranted a full bug report

line 61 
  EERef &operator=( uint8_t in )     { return nvm_write_byte( (uint16_t) index, in ), *this;  }
probably should be 
 EERef &operator=( uint8_t in )     { return nvm_write_byte( (uint8_t*) index, in ), *this;  }

closed 2020-07-06 by Yardie-