SpenceKonde/megaTinyCore #205: Add fast digital I/O functions to optimize performance for compile time known situations

SpenceKonde, 2020-07-05:
This optimization has been discussed at length elsewhere, but I realized there wasn't an issue for it here. It will drastically speed up digitalRead/digitalWrite/pinMode!

SpenceKonde, 2020-10-26:
Uhhhhhhh... guys.. Have any of y'all looked at an assembly listing for digitalWrite() with compiletime known pin and value? 

The compiler is smart enough to do most of this optimization anyway... the only catch is the check for whether it's input or output, to decide whether to switch the pullup or the output and the fact that it uses the PORT registers instead of the VPORT ones.....

SpenceKonde, 2020-11-19:
@WestfW @MCUdude @nerdralph

Do any of you know if it's possible to make it #error if one calls digitalWriteFast() with arguments that aren't compile-time known? I'm inclined to think that would be better behavior here; if someone is trying to use fast digital I/O, but they aren't specifying constants, shouldn't we stop and force them to either go fix it, or explicitly accept the fallback?
While for the normal Arduino API functions, one would never want to put in a roadblock for users like this, I'd argue that the fast digital I/O functions are a sufficiently advanced feature that people using it are qualified to understand, say, something like this:
`#error "digitalWriteFast requires the pin number to be compile-time known"` ...

nerdralph, 2020-11-19:
I was doing that with digitalWrite for picoCore, but recently disbled it
and am working on making it a warning.  Somethin like "digital functions
will run slower as pin number ia not a compile-time constant.
https://github.com/nerdralph/picoCore/blob/master/avr/cores/picocore/Arduino.h#L56

SpenceKonde, 2020-11-19:
Oh awesome! Wow... that is like... super useful, heh... 

And yeah, I would never do something like that for digitalWrite() itself (the people who use that straight up are usually not clued in enough to know what that error means anyway - no way to say that nicely) - but digitalWriteFast(), where someone is likely calling it because they expect performance... but they slipped up and are not passing a constant so it's no better than digitalWrite? they deserve the dignity of an error message ;-)

Where can I find a list of things like __builtin_constant_p and badArg? I've been wishing I could do things like that for years!

nerdralph, 2020-11-20:
Here's the common onee.
https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html
There's more target-specific  builtins too.

badArg  is a function I defined that uses the error attribute.

SpenceKonde, 2020-11-23:
Wow... this is so cool! (I am not getting the error I send printed though? I have to declare a new error function for each error I want to show, though that's easy enough). I do wish I could make the error show up right at the bottom of the output so it's impossible to miss it... 

I've always found it very frustrating that it was possible to pass blatantly bad values - values anyone who wasn't blinded by having seen the error all day convinced the problem was elsewhere could see were obviously wrong, and we could know that at compile time, yet all the core can do is just... ignore the bad call, with no way to tell the user...

MCUdude, 2020-11-23:
Here is the MegaCoreX "pin known at compile time check" implementation for digitalReadFast and digitalWriteFast:
https://github.com/MCUdude/MegaCoreX/blob/2b34004d185db90c5f115b0ae4c06fda10a1d831/megaavr/cores/coreX-corefiles/api/Common.h#L125-L131

And here is it for MicroCore:
https://github.com/MCUdude/MicroCore/blob/a274c196be805cda25b845fc5c8cae468acc0b40/avr/cores/microcore/Arduino.h#L119-L161

SpenceKonde, 2020-11-23:
Hm, when I did that, the error wasn't displayed in the compiler output... making it less than helpful for debugging...

MCUdude, 2020-11-23:
Here is my error when the pin isn't known at compile time (MegaCoreX):

```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x0 -mmcu=atmega4809 -o /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_814156/Blink.ino.elf /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_814156/sketch/Blink.ino.cpp.o /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_814156/core/core.a -L/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_814156 -lm
In function 'check_constant_pin',
    inlined from 'digitalWriteFast' at /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/wiring_digital.c:208:3,
    inlined from 'loop' at /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_modified_sketch_85025/Blink.ino:33:19,
    inlined from 'main' at /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/main.cpp:42:9:
/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/Common.h:130:5: error: call to 'badArg' declared with attribute error: 
     badArg("Digital pin must be a constant");
     ^
In function 'check_constant_pin',
    inlined from 'digitalWriteFast' at /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/wiring_digital.c:208:3,
    inlined from 'loop' at /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_modified_sketch_85025/Blink.ino:35:19,
    inlined from 'main' at /Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/main.cpp:42:9:
/Users/hans/Documents/Arduino/hardware/MegaCoreX/megaavr/cores/coreX-corefiles/api/Common.h:130:5: error: call to 'badArg' declared with attribute error: 
     badArg("Digital pin must be a constant");
     ^
lto-wrapper: fatal error: /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc returned 1 exit status
compilation terminated.
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ATmega4809.
```

SpenceKonde, 2020-12-03:
Okay, I don't know why I was having issues with this before.

SpenceKonde, 2020-12-03:
Anyone care to bang on this a bit and make sure I didn't fuck something up?

SpenceKonde, 2020-12-10:
This appears to be working. please create a new issue if it doesn't work/

closed 2020-12-10 by SpenceKonde