SpenceKonde/megaTinyCore #206: AVRISPII Programmer option gone in Arduino IDE 1.8.13

sideswiper22, 2020-07-20:
Hello, I have stumbled upon this when I updated my IDE to 1.8.13. It seems like Arduino made it so that each board family only has programmer options stated in their respective programmers.txt to reduce clutter. While this seems nice, it means that third-party boards won't be able to use the default programmer for bootloaded chips unless they modify their programmers.txt manually.

A quick fix I made is I added these lines to the core's programmers.txt:

avrispmkii.name=AVRISP mkII
avrispmkii.communication=usb
avrispmkii.protocol=stk500v2
avrispmkii.program.protocol=stk500v2
avrispmkii.program.tool=avrdude
avrispmkii.program.extra_params=-Pusb

SpenceKonde, 2020-07-20:
Does this impact megaTinyCore? These parts use UPDI, not PDI, and I didn't think the avrispmkii supported updi...? 

Did you mean to file this with megaTinyCore?

SpenceKonde, 2020-07-20:
Well, it's already in on ATTinyCore, so that wouldn't make sense?

SpenceKonde, 2020-07-20:
If you can confirm that the avrispmkii can program UPDI parts, obviously will add that to programmers.txt immediately. Are there any other programmers capable of doing UPDI that were lost?

sideswiper22, 2020-07-20:
Sorry, I was referring to parts burned with OptibootX (non-UPDI) so that it now uses a USB to serial converter like the CH340G to upload code to the chips.

With Arduino IDE 1.8.13, megaTinyCore by default only shows "jtag2updi (megaTinyCore)" and "Onboard Atmel mEDBG (xplained_mini) (megaTinyCore)" as its available programmers (unless you don't touch it, leaving it blank by default and reverts to the default IDE programmer. In which case if you choose one of the two it will never go back to using the default programmer).

SpenceKonde, 2020-07-20:
When an Optiboot board definition is selected, it will always upload using the serial port. It will attempt to upload using the selected programmer only if you choose upload with programmer... If it's not doing that, that's a godawful bug, and needs to be fixed urgently...

SpenceKonde, 2020-07-20:
I am seeing the correct behavior on my system - regardless of what programmer I select, as long as I have a serial port selected (even one that's no longer present) it's attempting to upload through a serial port with the arduino protocol, as it should. If you're seeing something different, can you enable verbose output on compile, and show me what it prints in the console?

If you do "upload using programmer", or try to upload to a board that isn't one of the optiboot ones, that will attempt to use the selected programmer (that's the normal behavior for unbootloaded boards, and "upload using programmer" explicitly overrides the board's normal upload method)

SpenceKonde, 2020-07-20:
upload, not on compile (edited)

sideswiper22, 2020-07-21:
Hello, I've checked it as well looks like it's fine too on my end. I apologize, partly because it was always my intuition that the programmer setting is always important every time code is uploaded.

closed 2020-07-21 by sideswiper22

SpenceKonde, 2020-07-21:
> it was always my intuition that the programmer setting is always important every time code is uploaded.

Yeah, you and almost everyone else. I feel bad when I hear about people with the STK500 sync error on a traditional arduino board (the one that provides about as much information as "something went wrong") who say that they tried every different option from tools -> programmer when trying to upload a sketch normally (ie, they went through the 20+ programmer options, none of which changed anything at all). 

If it were up to me, I would change the name of that submenu option to "Programmer (for Burn Bootloader) whenever the currently selected board/etc specified the upload parameters.