SpenceKonde/megaTinyCore #208: "Steal" MCUdude's Comparator library

MCUdude, 2020-07-28:
I encourage you to steal the [Comparator library](https://github.com/MCUdude/MegaCoreX/tree/master/megaavr/libraries/Comparator). The porting itself should be fairly easy; much easier than the Logic library. And the comparator on mega0/mega1 are much more comprehensive than on classic AVRs.

SpenceKonde, 2020-07-29:
Porting looks trivial

SpenceKonde, 2020-08-08:
Nope, not trivial, not for megaTinyCore nor DxCore - the main issues are that I'm not as comfortable as you are with this newfangled C++ stuff like "namespaces" and "classes"., yet I would need to go from a library designed to support a single analog comparator, to one which supports either 1 or 3 depending on the part... and with a different set of pins on each one... 

Yeah, this is not a priority, I have far more pressing things to work on....

NickNifontov, 2021-01-06:
any news about comparator support?

MCUdude, 2021-01-06:
> any news about comparator support?

I don't think it's going to happen any time soon. I've also realized I wouldn't bother trying to do it myself. I've done some contributions recently to DxCore and the release script repo that I put a lot of time and effort into, but I usually get little or no response back. It's kinda demotivating

SpenceKonde, 2021-01-27:
Target 2.3.0

NickNifontov, 2021-03-03:
> Target 2.3.0

Any news about date release of 2.3.x?

SpenceKonde, 2021-03-03:
Next project is ATTinyCore 2.0.0. I've got 3 pins_arduino.h's covering 3 parts left to clean and normalize, thrn 3 more that need to be created, though the parts that have already been done are all the weird ones.. what's left is cleaning the x8 and x313 pins_arduino to bring it up to par, there are two pin mappings for the 88. and then the 828 and 1634 need to be created from information in the second halfassed core that is being jettisoned from ATTinyCore 2.0.0. After that boards.txt needs to be reprocessed, standardized, and expanded to accomodate the new pin mappings, expose flash saving compile options in a consistent manner, provide options for reset disab le (with appropriate warnings) on non-micronucleus boards, and external clock which really should have always been an option because it is actually not really hard to revive a part that you hosed like that. Plus sometime in there there's testing of the new functionality plus regression testing.... OH! And I need to generate I think 4 or 5 micronucleus build configs for each family availble in a part with at least 8k of flash, and build micronucleus binaries for them, and test my dev boards to make sure it works on these new families....  After the new bootloaders are done, hopefully by then the other people involved in the project will be able to tell me what libraries actually work and should be part of the core, and which ones don't work and won't be made to work so we shuould forget about them and warn people that they won't work. There's a long and sordid discussion of the matter over on ATTinyCore. 

Will start working on megaTinyCore 2.3.0 after ATTinyCore 2.0.0 is available, but as you can see the to do list is really long. Porting this library should not be too hard - I do not think it requires my particular skills.

Even in the absence of a library, the comparators are *not* hard to use without the library, I encourage you to give it a try...  It's like... 3 registers? maybe 4 if you MCUdude's libraries are cute wrappers around chip peripherals, but comparators ain't like the CCL where the whole thing is a mess without the library - the code is uglier, but it isn't hard to write or anything.  in!

SpenceKonde, 2021-04-01:
This is now implemented in 2.3.0-dev (github master branch). Testing was VERY sparse. Users who care about this functionality are encouraged to test it and find all the stuff I screwed up before it goes into a release :-)

Comparator library readme:
https://github.com/SpenceKonde/megaTinyCore/tree/master/megaavr/libraries/Comparator

Open new issues for any bugs, let's let #208 rest in peace.

closed 2021-04-01 by SpenceKonde