SpenceKonde/megaTinyCore #211: ADC_REFSEL_VREFA_gc available on other devices besides just __AVR_ATtiny3216__

freemovers, 2020-07-30:
https://github.com/SpenceKonde/megaTinyCore/blob/9aa8c03f910e068f78abe6e226478897e157adef/megaavr/cores/arduino/Arduino.h#L62

Looks like you were going to define these for the following devices?:
`(defined(__AVR_ATtiny804__) || defined(__AVR_ATtiny806__) || defined(__AVR_ATtiny807__) || defined(__AVR_ATtiny1604__) || defined(__AVR_ATtiny1606__) || defined(__AVR_ATtiny1607__) || defined(__AVR_ATtiny1614__) || defined(__AVR_ATtiny1616__) || defined(__AVR_ATtiny1617__) || defined(__AVR_ATtiny3216__) || defined(__AVR_ATtiny3217__) )`

SpenceKonde, 2020-07-30:
Yup, thanks...

SpenceKonde, 2020-07-31:
OH
I remember this.....

SpenceKonde, 2020-07-31:
The datasheets and IO headers don't match

commit 5bfd8c12 (2020-07-31): Fix #211

Corrected copy-paste error, updated docs.
closed 2020-07-31 by SpenceKonde