SpenceKonde/megaTinyCore #215: One Wire Library Issue

heinrich321, 2020-07-31:
Please investigate possible issue with OneWire library  on ATtiny3216. Pins toggling fine, but when using with OneWire, normal example such as [here](https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Simple/Simple.ino) there is nou output on the pin.

SpenceKonde, 2020-07-31:
Please link to the version of OneWire you are using. You see how many issues I have open, and that doesn't even include the other core that I haven't released yet and have people asking me for! I do not have time to go chasing down libraries, make it easy for me.

heinrich321, 2020-07-31:
Hey brother sure thing. Its the newest version 2.3.5 [standard one wire lib](https://www.pjrc.com/teensy/td_libs_OneWire.html) . I tried it on all of the pins, let me know if I can provide any other info. In the meantime, ill do a bit-bang :)

SpenceKonde, 2020-07-31:
Please confirm that this works: https://github.com/SpenceKonde/OneWire
If so will submit PR with that change to Paul.

SpenceKonde, 2020-07-31:
Can't say I was much impressed by line 18 of: https://github.com/PaulStoffregen/OneWire/blob/master/util/OneWire_direct_gpio.h

heinrich321, 2020-07-31:
Works like a charm! Thanks for that, pretty hardcore picking that up so fast! Cheers :)

SpenceKonde, 2020-08-01:
Actually, @heinrich321 can you test one more thing?
1. Record sketch size with the version you say works from above. 
2. Pull latest changes from
3. Test with those https://github.com/SpenceKonde/OneWire/
4. Comment here reporting whether it works, and whether the sketch size decreased (and by how much)

Thanks!

heinrich321, 2020-08-01:
Hi @SpenceKonde [First Fix](https://github.com/SpenceKonde/OneWire/commit/f0129cc41860dcfd456690662d56469f38fd703e#diff-c8d273aeb987f835206d8ec3ca3b9a92)

Sketch uses 6275 bytes (19%) of program storage space. Maximum is 32768 bytes.
Global variables use 204 bytes (9%) of dynamic memory, leaving 1844 bytes for local variables. Maximum is 2048 bytes.

(Works)

With the latest optimisation seems like theres a syntax error:

> Arduino: 1.8.13 Hourly Build 2020/06/03 04:33 (Windows 10), Board: "ATtiny3216/1616/1606/816/806/416/406, ATtiny3216, 20 MHz, 1.8V (5 MHz or less), Disabled, Disabled, EEPROM retained, Enabled (default timer), Closer to 5v"

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::reset()':

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:178:36: error: expected ')' before ';' token

  } while ( !DIRECT_READ(reg, mask));

                                    ^

In file included from C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:144:0:

C:\Users\heinr\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:22:64: error: lvalue required as left operand of assignment

 #define DIRECT_WRITE_LOW(base, mask)    ((*(base)+1)) &= ~(mask))

                                                                ^

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:181:2: note: in expansion of macro 'DIRECT_WRITE_LOW'

  DIRECT_WRITE_LOW(reg, mask);

  ^~~~~~~~~~~~~~~~

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:188:29: error: expected ')' before ';' token

  r = !DIRECT_READ(reg, mask);

                             ^

In file included from C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:144:0:

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp: In member function 'void OneWire::write_bit(uint8_t)':

C:\Users\heinr\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:22:64: error: lvalue required as left operand of assignment

 #define DIRECT_WRITE_LOW(base, mask)    ((*(base)+1)) &= ~(mask))

                                                                ^

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:205:3: note: in expansion of macro 'DIRECT_WRITE_LOW'

   DIRECT_WRITE_LOW(reg, mask);

   ^~~~~~~~~~~~~~~~

C:\Users\heinr\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:22:64: error: lvalue required as left operand of assignment

 #define DIRECT_WRITE_LOW(base, mask)    ((*(base)+1)) &= ~(mask))

                                                                ^

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:213:3: note: in expansion of macro 'DIRECT_WRITE_LOW'

   DIRECT_WRITE_LOW(reg, mask);

   ^~~~~~~~~~~~~~~~

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp: In member function 'uint8_t OneWire::read_bit()':

C:\Users\heinr\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:22:64: error: lvalue required as left operand of assignment

 #define DIRECT_WRITE_LOW(base, mask)    ((*(base)+1)) &= ~(mask))

                                                                ^

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:234:2: note: in expansion of macro 'DIRECT_WRITE_LOW'

  DIRECT_WRITE_LOW(reg, mask);

  ^~~~~~~~~~~~~~~~

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:238:28: error: expected ')' before ';' token

  r = DIRECT_READ(reg, mask);

                            ^

In file included from C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:144:0:

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp: In member function 'void OneWire::write(uint8_t, uint8_t)':

C:\Users\heinr\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:22:64: error: lvalue required as left operand of assignment

 #define DIRECT_WRITE_LOW(base, mask)    ((*(base)+1)) &= ~(mask))

                                                                ^

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:260:2: note: in expansion of macro 'DIRECT_WRITE_LOW'

  DIRECT_WRITE_LOW(baseReg, bitmask);

  ^~~~~~~~~~~~~~~~

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp: In member function 'void OneWire::write_bytes(const uint8_t*, uint16_t, bool)':

C:\Users\heinr\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:22:64: error: lvalue required as left operand of assignment

 #define DIRECT_WRITE_LOW(base, mask)    ((*(base)+1)) &= ~(mask))

                                                                ^

C:\Users\heinr\Documents\Arduino\libraries\OneWire\OneWire.cpp:271:5: note: in expansion of macro 'DIRECT_WRITE_LOW'

     DIRECT_WRITE_LOW(baseReg, bitmask);

     ^~~~~~~~~~~~~~~~

exit status 1

Error compiling for board ATtiny3216/1616/1606/816/806/416/406.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

heinrich321, 2020-08-01:
I think there is just a few bracket errors, think I fixed it:

As well as de-referencing before incrementing on line #22

#define PIN_TO_BASEREG(pin)             (((digitalPinToPort(pin))<<2))
#define DIRECT_READ(base, mask)         ((*((base+2)) & (mask)) ? 1 : 0)
#define DIRECT_MODE_INPUT(base, mask)   ((*(base)) &= ~(mask))
#define DIRECT_MODE_OUTPUT(base, mask)  ((*(base)) |= (mask))
#define DIRECT_WRITE_LOW(base, mask)    ((*((base)+1)) &= ~(mask))
#define DIRECT_WRITE_HIGH(base, mask)   ((*((base)+1)) |= (mask))

heinrich321, 2020-08-01:
Optimisation:

Fix #1 Sketch uses 6275 bytes (19%)
Fix #2 Sketch uses 6227 bytes (19%)

0.7% size reduction & working

SpenceKonde, 2020-08-02:
Nice, thanks for going the extra mile and fixing my messed up parens too!

SpenceKonde, 2020-08-02:
I test compiled it - no good, it gets an -fpermissive warning on my core and not with stock core. It needs an explicit cast in there somewhere

SpenceKonde, 2020-08-02:
I can't for the life of me figure out why the stock core doesn't, honetly...

SpenceKonde, 2020-08-05:
Okay - it now builds without the warning. I didn't screw anything up right?  (I don't have hardware to verify behavior against)

heinrich321, 2020-08-06:
Sorry been a bit hectic - will test tonight :) and let you know

SpenceKonde, 2020-08-15:
https://github.com/PaulStoffregen/OneWire/pull/94 PR submitted. Closing this one as my fork can be used by megaTinyCore users who need to interact with OneWire devices.

closed 2020-08-15 by SpenceKonde

BigGene, 2021-05-19:
Glad I found this post and fix. Your OneWire fork works properly with the ATtiny1614 on your development board :) What's the best way to get Paul to take action on the PR from last August?
I can also report that the ss_oled.h (ver 4.2.0), RF24.h (ver 1.4.0), and DallasTemperature.h (ver 3.9.0) libraries seem to be working. Thank you!

SpenceKonde, 2021-05-19:
Hmm, seeing as I have been unsuccessful for 9 months, I can';t say I think I'm well placed to advise ;-)

BigGene, 2021-12-13:
It looks like the OneWire library was updated to version 2.3.6. Do you know if that includes your fix for megaTinyCore users? I should be able to check it myself, but I don't want to mess up my current working setup. Sorry for being such a newb!

SpenceKonde, 2021-12-14:
Does not look like he has addressed the issue.

BigGene, 2021-12-14:
Thanks. That's what I was thinking :(