SpenceKonde/megaTinyCore #216: Adding MFRC522 library support?

sideswiper22, 2020-08-01:
I think this issue has been opened before but I'm not sure. All I know is that it's probably an issue with the F() macro but I don't know how to solve this problem. Using the Mifare RFID (MFRC522) library with megaTinyCore (and probably all megaAVR parts) causes a lot of compile errors in the console meaning it cannot be used with these parts. Any solution?

Here's the error log:
`Arduino: 1.8.13 (Windows 10), Board: "ATmega4809, Nano Every, Reset, BOD 4.2V, Internal 16 MHz, No bootloader"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Clyde Corpuz\Documents\Arduino\libraries -fqbn=MegaCoreX:megaavr:4809:pinout=nano_every,resetpin=reset,BOD=4v2,clock=internal_16MHz,bootloader=no_bootloader -ide-version=10813 -build-path C:\Users\CLYDEC~1\AppData\Local\Temp\arduino_build_378166 -warnings=none -build-cache C:\Users\CLYDEC~1\AppData\Local\Temp\arduino_cache_61330 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose C:\Users\CLYDEC~1\AppData\Local\Temp\arduino_modified_sketch_116674\sketch_aug01a.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Clyde Corpuz\Documents\Arduino\libraries -fqbn=MegaCoreX:megaavr:4809:pinout=nano_every,resetpin=reset,BOD=4v2,clock=internal_16MHz,bootloader=no_bootloader -ide-version=10813 -build-path C:\Users\CLYDEC~1\AppData\Local\Temp\arduino_build_378166 -warnings=none -build-cache C:\Users\CLYDEC~1\AppData\Local\Temp\arduino_cache_61330 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose C:\Users\CLYDEC~1\AppData\Local\Temp\arduino_modified_sketch_116674\sketch_aug01a.ino

Using board '4809' from platform in folder: C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.4

Using core 'coreX-corefiles' from platform in folder: C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.4

Detecting libraries used...

"C:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\cores\\coreX-corefiles/api/deprecated" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\cores\\coreX-corefiles" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\variants\\nano-every" "C:\\Users\\CLYDEC~1\\AppData\\Local\\Temp\\arduino_build_378166\\sketch\\sketch_aug01a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for MFRC522.h: [MFRC522-1.4.6@1.4.6]

ResolveLibrary(MFRC522.h)

  -> candidates: [MFRC522-1.4.6@1.4.6]

"C:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\cores\\coreX-corefiles/api/deprecated" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\cores\\coreX-corefiles" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\variants\\nano-every" "-IC:\\Users\\Clyde Corpuz\\Documents\\Arduino\\libraries\\MFRC522-1.4.6\\src" "C:\\Users\\CLYDEC~1\\AppData\\Local\\Temp\\arduino_build_378166\\sketch\\sketch_aug01a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for SPI.h: [SPI@1.0]

ResolveLibrary(SPI.h)

  -> candidates: [SPI@1.0]

"C:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\cores\\coreX-corefiles/api/deprecated" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\cores\\coreX-corefiles" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\variants\\nano-every" "-IC:\\Users\\Clyde Corpuz\\Documents\\Arduino\\libraries\\MFRC522-1.4.6\\src" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\libraries\\SPI\\src" "C:\\Users\\CLYDEC~1\\AppData\\Local\\Temp\\arduino_build_378166\\sketch\\sketch_aug01a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

"C:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_ATmega4809 -DARDUINO_ARCH_MEGAAVR "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\cores\\coreX-corefiles/api/deprecated" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\cores\\coreX-corefiles" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\variants\\nano-every" "-IC:\\Users\\Clyde Corpuz\\Documents\\Arduino\\libraries\\MFRC522-1.4.6\\src" "-IC:\\Users\\Clyde Corpuz\\AppData\\Local\\Arduino15\\packages\\MegaCoreX\\hardware\\megaavr\\1.0.4\\libraries\\SPI\\src" "C:\\Users\\Clyde Corpuz\\Documents\\Arduino\\libr [... cut ...]
C:\Users\Clyde Corpuz\Documents\Arduino\libraries\MFRC522-1.4.6\src\MFRC522.cpp:1341:24: note: in expansion of macro 'F'

   default:      return F("Unknown type");

                        ^

Using library MFRC522-1.4.6 at version 1.4.6 in folder: C:\Users\Clyde Corpuz\Documents\Arduino\libraries\MFRC522-1.4.6 

Using library SPI at version 1.0 in folder: C:\Users\Clyde Corpuz\AppData\Local\Arduino15\packages\MegaCoreX\hardware\megaavr\1.0.4\libraries\SPI 

exit status 1

Error compiling for board ATmega4809.

`

SpenceKonde, 2020-08-01:
uuuugh - link me to the github of the library.

SpenceKonde, 2020-08-01:
(to be clear, I know exactly what the problem is, and the fix is not hard, but needs to be done in the library)

SpenceKonde, 2020-08-01:
Also, your terminology is wrong. "megaAVR" according to Microchip refers to any ATmega parts. The (Arduino-only) name "megaavr" refers to the post-2016 parts which I refer to as "Modern AVR", or explicitly by their product lines (tinyAVR 0/1-series, megaAVR 0-series, and the Dx-series - though the latter has additional complications). Don't use "megaAVR" when you mean "megaavr" - the only place in megaTinyCore that should use that word is in the directory name, because that's what the official cores call this architectures, and it's required for library compatibility. And unfortunately it's alluded to by the name, because I was unaware of that distinction when I wrote the core!) 

But yeah, refer me to the library, and I'll fork it and implement the fix, refer to you for testing, and if it works, submit as PR to library maintainer.

SpenceKonde, 2020-08-01:
For that matter - PLEASE do that for any other libraries with that F() macro problem. Simple fix in the library, but impossible to fix in the general case from the core without significant adverse impacts

sideswiper22, 2020-08-02:
Alright [https://github.com/miguelbalboa/rfid](url)

SpenceKonde, 2020-08-02:
I'll take a stab at fixing it - I hope I'm able to convince him to merge in my changes (since they will be fully transparent, and have no impact other than adding support for a huge number of parts) - he stated that he is no longer maintaining the library and will not add support for new parts. I don't want to have to keep my own fork just to support the new parts via like, a four-line change wrapped in #ifdefs!

TomWS1, 2020-08-05:
I submitted an issue to the MFRC522 library: https://github.com/miguelbalboa/rfid/issues/523 as I ran into this independently of sideswiper22.

TomWS1, 2020-08-05:
This was closed with the comment: 
"Must be fixed by the framework of the board.
The simple solution was to typedef __FlashStringHelper as char and define F() macro which does not modify the string. But then the RAM consumption is high."
I responded with "Not quite as simple as that as all functions that attempt to overload with ___FlashStringHelper then fail as duplicate overloads.  I suppose one way around it is to implement __FlashStringHelper so that it beningly returns a char*"  Which, in fact, may be the way to deal with this.

SpenceKonde, 2020-08-05:
Can you provide an example of a library that does that overloading for analysis?

TomWS1, 2020-08-05:
You don't need a library.  Methods in Print and String both fail on compile if I modify String.h as:
//TWS class __FlashStringHelper;
typedef char __FlashStringHelper;
#define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))

Both of these files include both const char * and const __FlashStringHelper in their overloaded methods.  For example:
	String & operator = (const char *cstr);
	String & operator = (const __FlashStringHelper *str);  <== this one fails during compiler

My brain dead workaround was to comment out all the duplicate overloaded methods.  It got rid of the error messages but not very pretty.

Rotzbua, 2020-08-05:
`Arduino.h` of megatinycore 2.0.5 does redefine `F()` with type `char`:
```cpp
#undef F
#define F(str) (str)
```
`String.h` has a different definition of `F()` with type `__FlashStringHelper`.

That is basically the problem.
---

I tried something simple: added in `Arduino.h`:
```cpp
#define __FlashStringHelper char
```
Worked. But no guarantee for safe usage.

Best.

TomWS1, 2020-08-06:
> 
> 
> Can you provide an example of a library that does that overloading for analysis?

I realized you might mean from the MFRC522 library, so here is the method declaration:
`	// old function used too much memory, now name moved to flash; if you need char, copy from flash to memory
	//const char *GetStatusCodeName(byte code);
	static const __FlashStringHelper *GetStatusCodeName(StatusCode code);
	static PICC_Type PICC_GetType(byte sak);
	// old function used too much memory, now name moved to flash; if you need char, copy from flash to memory
	//const char *PICC_GetTypeName(byte type);
	static const __FlashStringHelper *PICC_GetTypeName(PICC_Type type);
`

sideswiper22, 2020-09-18:
Any progress on this?

TomWS1, 2020-09-18:
I worked around the problem by:
1. putting:
```
//TWS class __FlashStringHelper;
typedef char __FlashStringHelper;
```
it String.h and
2. commenting out the overloaded methods in Print (.cpp & .h) and String (.cpp & .h) as in:
`	//String(const __FlashStringHelper *str);
`
This was, of course, only in the megaTinyCore package files.

SpenceKonde, 2020-09-18:
I am investigating a few approaches to fixing the __FlashStringHelper issue
in a universal manner

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SpenceKonde, 2020-10-10:
I took the easy way out, for now, and just reimlemented F() macro as it normally is. It's slower and less efficient, but compatible with all libraries. 

The alternatives are like, really ugly :-(

closed 2020-10-10 by SpenceKonde

sideswiper22, 2020-10-13:
Tested and working. Thank you!