SpenceKonde/megaTinyCore #217: Hardware reset ?? huh

Yardie-, 2020-08-02:
I have been trying to get my head around how to actually use the reset/upi pin as a hardware reset.
I have used your software trick to great effect but I was wondering how to just enable the pin directly.

attaching an interrupt or watch during loop for low on the pin == LOW then calling the reboot function would work 
But the datasheet seems to say by setting the  RSTPINCFG to 0x2 would enable it.
I only use the Arduino IDE to set the fuses and it doesnt seem to have it as an option.
I can understand why too many choices = too many problems. :1234: 

Is this just something that we need to do directly??

SpenceKonde, 2020-08-02:
The problem is that if you set the UPDI pin to work as a true hardware
reset, then you can't program it via UPDI. Because you no longer have a
UPDI pin, you have a reset pin. You need to use a HV programmer to reset
the pin to UPDI so you can program it again. That is why it is not an
option - because people would set it, not have HV programmer, and complain
that they had bricked their board (this is the same reason I don';t support
disabling reset on ATTinyCore.

It's an option with the optiboot (serial bootloader) board defs, so that
you can wire up the "autoreset" circuit after burning bootloader like that
(without that, optiboot UX is pretty crap). Judging from interest on my
Tindie store, very few people are doing that :-P

There is an open issue for adding support for HV UPDI programming. Some
guys claim to have gotten it to work using a PWM pin and cap/diode voltage
multiplier, and I think a zener to get the 12v, and IIRC a fet and a
transistor to apply it to the pin - so it's like a buck in parts tops
to make a nano into it. Nobody has gotten a PR to me or made clear exactly
what I need to do to get the core to work with it; it sounds like just some
different parameters for uploading to the board? Since many of my other
priorities are super urgent, and nobody seems to care about this very much
(nobody has mentioned it to me in months!), I'm not going to work on that
until I finish the action items that people are asking about.

Yardie-, 2020-08-02:
Hi Spence
Good call I thought that was probably the case. 
If I want to take the risk I will do the research and burn the appropriate bits and watch the bricks fly. 
Your software solution is clean and simple and setting an interrupt to trigger it isn't that hard.
Thanks heaps these are very cool little chips.

closed 2020-08-02 by Yardie-

Yardie-, 2020-08-05:
Just for anyone following who wants some very simple code that just works.
It uses another pin. as reset pin.
Yeah it's easy enough to write yourself but cut and paste works for me :)

```
/*
both work just uncomment the vesion you want
*/
#define LED_PIN (int) 13
#define RESET_ME_PIN (int) 10

void reset_me()
{
   _PROTECTED_WRITE(RSTCTRL.SWRR,1);
}

void setup()
{
   pinMode(LED_PIN, OUTPUT);
   digitalWrite(LED_PIN, LOW); // at start turn off LED to show reboot
   pinMode(RESET_ME_PIN, INPUT_PULLUP);
   delay(1000);
   digitalWrite(LED_PIN, HIGH); // then turn LED on after one second
   //~ attachInterrupt (digitalPinToInterrupt(RESET_ME_PIN), reset_me, FALLING); // with interupts
}

void loop()
{
   if(digitalRead(RESET_ME_PIN) == LOW) reset_me(); // in the loop
}

```