SpenceKonde/megaTinyCore #218: TW as both slave and master is it possible

Yardie-, 2020-08-05:
After lots of battling I have finally noticed something that doesn't seem to happen on other Arduinos I have dealt with.
Calling the begin as Wire.begin(); Is the only way to run as a Master
giving any other parameters forces the avr into slave mode.
Is this the correct behaviour

To use a unit in both do you have to do the following?
1
Wire.begin(address....)
respond to calls
Wire.end()
Wire.begin()
make calls to slaves
Wire.end();
goto 1

Yardie-, 2020-08-05:
Well actually at the moment I can't get the 3216 to work as a master anyway. Even just trying to set a DS3231 RTC 
I need to step back for a while.
They work as wonderful obedient TW slaves 
BUT master I dunno 
no luck here 
yes I added 10K pull-ups

SpenceKonde, 2020-08-06:
Well, they are most certainly supposed to work as masters!

Yardie-, 2020-08-06:
Yeah I'm a bit over it at the moment so I will take a break and come back to it. 
Everything else just works so I don't know what I'm doing wrong. 
The question I am most interested in answering is in the code somewhere that I haven't yet seen.

Can the chip handle being both a master and a slave at the same time. 
At first glance the code implies no.

SpenceKonde, 2020-08-06:
Huh, wait, I didn't post this before your reply 6 hrs ago?!

> doesn't seem to happen on other Arduinos

That would be because you have not tried it with an Arduino Uno WiFi Rev. 2, Nano Every, or other parts with the new peripherals. Wire library has received only very minimal modifications from the version in that package. 

> Is this the correct behaviour

I suspect that the Arduino developers have given precisely zero thought to the possibility that a device could be both a master and a slave! I had only noticed that it was possible while looking through a datasheet the other day, I had always assumed it was one or the other.

That said, it is based on twi.c and twi.h, which (according to the copyright notice at the top) were written by Microchip, which is keenly aware of the details of TWI operation on their parts. It appears that that library was intentionally written to support one or the other only, and intentionally disables the initialization in the event that it sees that it's already enabled. 

> is it possible

Yes, datasheet makes abundantly clear that operating as both master and slave is possible without disabling the TWI interface between those two modes. 

> (what I think you meant) Is it possible with Wire.h without major redesign of that library?

Good question! 

Quick impression is that the naive approach might actually work...

Yardie-, 2020-08-06:
First and foremost I am once again an idiot.
Going back to basic Master to RTC code 
YES IT DOES WORK as a master I am in love once again.

as far as the code goes 
Yes corporate coders only do the bare minimum "I don't get paid to be creative. I work to the design specs. full stop."

but an old bloke like me
who does this stuff just for fun 
the naive approach is me all the way.
Bash away and get it up then fix the broken bits at the end.

I'll have a proper look soon. disabling the disabling shouldn't be that hard.
Probably not what you would want in a release but it could be useful in a thread like this for others who want the functionality and of the both at once.

I'm basically setting up a TWI network over longer cables so it's necessary for me. I want everyone to talk to each other when they need to. 
plug something in and it can ask for the time then go back to being a slave.
I had hoped that the RTC onboard was accurate enough but without some sort of drift calculations based on regular updates or a crystal on each unit.
Now we can broadcast that reduces the need but I would still like it.

Anyway I'll have a play and post back what I find.
Thanks again Spence

:) Greg

SpenceKonde, 2020-08-06:
Okay. cool, so it DOES do master correctly! Phew. 

Okay, yeah, what I'd be inclined to try is just commenting out the checks on line 65 and 101 of twi.c - and seeing first if it works in a simple situation, and then if I could make it actually work in practice, being on the lookout for TWI getting into borked states... and checking for (MSTATUS & (TWI_ARBLOST_bm | TWI_BUSERR_bm)) compulsively....

except for those checks against it, twi_mode is a "write-only variable" :-P I suspect you'll find cases where it breaks, but I'll be damned if I could tell you what it would be :-P

SpenceKonde, 2020-08-06:
And to be clear, TWI_ARBLOST is not necessarily a bad sign if it doesn't otherwise trash behavior - though I am pretty sure the libraries aren't smart enough to use that to let the user code know that the transaction didn't happen - it just means two things tried to send something at the same time, and that one discovered the winner wasn't him... (at this point they should - and hopefully are smart enough to actually - give up and let whoever did win do their thing. That's what they standard says they're supposed to do)

Yardie-, 2020-08-06:
No from what I saw TWI_ARBLOST just fails and maybe tries again.
twi_mode swaps all over the place so setting it on init doesn't seem to be set in concrete.
I think calling Master init each from each init or at lest setting the bus speed 
The bus speed was the first thing that twigged me to it. 
It is zero unless it's a master.
It should be easy to see if it works and even arbitrates just with two masters playing tennis with an RTC or maybe rugby.
:smile:

Yardie-, 2020-08-06:
I would like to add some way of alerting user to TWI_ARBLOST or at least have some sort of whatif secario

Yardie-, 2020-08-06:
Progress so far 

I commented out those two lines then added
then basically  using RTCLib did

```
void data_received(int number_of_bytes)
{
   ;
}
void data_requested()
{
   ;
}

void be_master()
{
  if (! rtc.begin()) {
    Serial.println("Couldn't find RTC");
    while (1);
  }
}

void be_slave()
{
   Wire.begin(12);
   Wire.onReceive(data_received);         // register event
   Wire.onRequest(data_requested);  // send data in specific buffer
}

loop()
{

be_master();
read the rtc  
be_slave();
delay(1000);
}

```
And it works cleanly
even fails when I try and read as slave but works again once I am master again.

I have to do some more hookups to see if it's still responding as a slave though
but it looks like it is actually a nice simple solution.
it actually clarifies the process as well
I will have to leave this for a while but it looks very promising so far

Yardie-, 2020-08-06:
I have to leave this but I just found a nasty piece of code in line 334 of twi.c

 ```
 // in case of arbitration lost, retry sending
      if (master_result == TWIM_RESULT_ARBITRATION_LOST) {
         goto trigger_action;
      }

```

don't fail just keep smashing away forever  
I don't think that's much of an idea.

I will remove that for my code and just throw an error then pick it up 

I have to leave this but that was a little distressing.
This could be so much better.
The core is all there.

SpenceKonde, 2020-08-06:
Well, actually that should be **good** shouldn't it?

In the event of lost arbitration, all it should need to do to succeed
is to wait until the other device is done, and retry sending... And
that's what it's doing!
Maybe we don't even have to worry about losing arbitration! That would
be pretty sweet...

Though if it was hanging up, the first thing I would want to do is put
a pin-flip in there (eg,  say we do PC2, since it's a pretty useless
pin on 3216 - pinMode(PIN_PC2,OUTPUT); in setup, and a
VPORTC.IN|=(1<<2); (VPORTx.IN works like the old PINx register,
writing it toggles the pin - unlike the PORTx.IN)  right before the
goto, and led+resistor on that pin so you can see if it suddenly
starts outputting a squarewave when the device locks up :-P )

Yardie-, 2020-08-06:
or just add a delay before the goto it wouldn't need to be that long

Yardie-, 2020-08-06:
I know it aint pretty but it doesn't break anytin' (I think )
 ```
// in case of arbitration lost, retry sending
      if (master_result == TWIM_RESULT_ARBITRATION_LOST) {
         uint8_t ok = 0;
         bool was = LOW;
         while (ok < 10)
         {
            if ( digitalRead(SCL) == HIGH )
            {
               if(was == HIGH)
                  ok ++;
               was = HIGH;
            }
            else
            {
               was = LOW;
               ok = 0;
            }
         }
         goto trigger_action;
      }
```

Yardie-, 2020-08-07:
actually and personally I would rather that my code knows that it lost arbitration and the connection failed.
rather than banging away in the background.
This protocol is hard enough to debug as it is. 
It needs a simply available error reporting system rather.
Anyway thanks Spence

Yardie-, 2020-08-07:
Ok I finally found the definitive answer to my original question.
from your original statements about the datasheet

> 
> >    is it possible
> 
> 
> Yes, datasheet makes abundantly clear that operating as both master and slave is possible without disabling the TWI interface between those two modes.
> 

and line 57 of Wire.cpp

> 
> // Special for megaAVR 0-series: Select which pins to use for I2C interface
> // True if pin specification actually exists
> // Note that we do not currently support the dual TWI mode
> 
So for those that follow.
The hack works and allows you simply to call 
Wire.begin()
to become master
Then 
Wire.begin(address)
to be a slave again

I have not yet tested the Slave side so I am not sure if you need to re register the  Wire.onReceive and Wire.onRequest but as you are not calling Wire.end between these calls the function pointers should remain.
As long as you start as a slave and set that up correctly.
then call Wire.begin();
before you make any master functions
then Wire.begin(address...) 
to revert back to a slave.

Whilst you are master you will not be a slave. 

This is where the arbitration and error info would be handy but that would require changes in Wire.cpp to make available

static register8_t  master_trans_status;                       /*!< Status of transaction */
static register8_t  master_result;                             /*!< Result of transaction */

cpp eludes me though

SpenceKonde, 2020-08-07:
What stops it from being both at the same time? I am pretty sure that the master baud doesn't need to be set to 0 to make slave mode work.....

Yardie-, 2020-08-07:
It hangs basically on the state of 
twi_mode variable 
line 288 
if(twi_mode != TWI_MODE_MASTER) return false;
rejects calls if you are not a master.
but also 
twi_mode is used as a data state indicator as well

```
/*! TWI Modes */
typedef enum __attribute__((packed)) TWI_MODE_enum {
	TWI_MODE_UNKNOWN = 0,
	TWI_MODE_MASTER = 1,
	TWI_MODE_SLAVE = 2,
	TWI_MODE_MASTER_TRANSMIT = 3,
	TWI_MODE_MASTER_RECEIVE = 4,
	TWI_MODE_SLAVE_TRANSMIT = 5,
	TWI_MODE_SLAVE_RECEIVE = 6
} TWI_MODE_t;

```
So as you suggested a better solution would require a slightly different approach.
or at least a a serious investigation of the process.

As a total aside

what is the point of 
    = (0x00<<0),
isn't that the same as 
   = 0x00

SpenceKonde, 2020-08-08:
Sure, but unless they zero out TWI0.MBAUD....? Ah, yup, there, in TWI_Disable().

0x00 << 0

Yeah, that's 0. (You do often see constructions like that where they use constants defined in the header files to be 0x00, and then leftshift them by another constant (which may be 0) - this is just because it's good practice to use the names and bitwise & and | to build values you write to registers - since the names provide a reminder of what they mean (I've had to wade through code that did this, and it sucked. The worst part was I wrote it, so there was nobody else to blame!

Yardie-, 2020-08-08:
Haha thanks.
Yep going back a couple of years is a fun process. I now use rediculously long variable names for exactly that reason. 
I'm on 'holidays' for a week but
I have had a couple of ideas on how to make it work whilst keeping the same code base. I'll play with it a bit and see what you think.

Yardie-, 2020-08-08:
Ok I think I have it but I can't test it for a week.
If you could have a glance at it that would be great.
only two functions in twi.c need changes
Everything happens in TWI_MasterWriteRead 
so if we arrive as a slave we swap to master for the duration of the function
TWI_MasterInit just needs a minor change so it doesn't change the baud rate when it is re-initialised.

the previous changes are not now required and can stay to catch other potential weird states.

All you need to in the user code is 
Wire.begin( slave_address)
setClock(speed)

make call to another slave as master

respond to call from another master.

I can't see anywhere where the slave function pointers are unset so they should remain valid.
I could be wrong though.

So voila

```
void TWI_MasterInit(uint32_t frequency)
{
   if(twi_mode != TWI_MODE_UNKNOWN) return;

   // Enable pullups just in case, should have external ones though
#ifdef NO_EXTERNAL_I2C_PULLUP
   pinMode(PIN_WIRE_SDA, INPUT_PULLUP);
   pinMode(PIN_WIRE_SCL, INPUT_PULLUP);
#endif
/* This is done in wiring.c if this is needed for the part.
#if defined(TWISPIROUTEA)
   PORTMUX.TWISPIROUTEA |= TWI_MUX;
#endif
*/
   twi_mode = TWI_MODE_MASTER;

   master_bytesRead = 0;
   master_bytesWritten = 0;
   master_trans_status = TWIM_STATUS_READY;
   master_result = TWIM_RESULT_UNKNOWN;

   TWI0.MCTRLA = TWI_RIEN_bm | TWI_WIEN_bm | TWI_ENABLE_bm;
   if(frequency > 255 || (uint8_t) frequency != TWI0.MBAUD ) // still doesn't cover sending a zero value as frequency but it never did
      TWI_MasterSetBaud(frequency);
   TWI0.MSTATUS = TWI_BUSSTATE_IDLE_gc;
}

```

and 

```
uint8_t TWI_MasterWriteRead(uint8_t slave_address,
                         uint8_t *write_data,
                         uint8_t bytes_to_write,
                         uint8_t bytes_to_read,
                   uint8_t send_stop)
{
   TWI_MODE_t mode = twi_mode;
   if(twi_mode == TWI_MODE_SLAVE)
   {
      twi_mode = TWI_MODE_UNKNOWN;
      if ( TWI0.MBAUD > 0 )
         TWI_MasterInit((uint32_t) TWI0.MBAUD&0xFF); // if baud is already set just maintain thatTWI_MasterInit((uint32_t)
      else
         TWI_MasterInit(100000); // just hard coded default here
   }

   if(twi_mode != TWI_MODE_MASTER) return false; // not slave or master ?? bye

   /*Initiate transaction if bus is ready. */
   uint8_t ret = 0;
   if (master_trans_status == TWIM_STATUS_READY) {

      master_trans_status = TWIM_STATUS_BUSY;
      master_result = TWIM_RESULT_UNKNOWN;

      master_writeData = write_data;

      master_bytesToWrite = bytes_to_write;
      master_bytesToRead = bytes_to_read;
      master_bytesWritten = 0;
      master_bytesRead = 0;
      master_sendStop = send_stop;
      master_slaveAddress = slave_address<<1;

trigger_action:

      /* If write command, send the START condition + Address +
       * 'R/_W = 0'
       */
      if (master_bytesToWrite > 0) {
         twi_mode = TWI_MODE_MASTER_TRANSMIT;
         uint8_t writeAddress = ADD_WRITE_BIT(master_slaveAddress);
         TWI0.MADDR = writeAddress;
      }

      /* If read command, send the START condition + Address +
       * 'R/_W = 1'
       */
      else if (master_bytesToRead > 0) {
         twi_mode = TWI_MODE_MASTER_RECEIVE;
         uint8_t readAddress = ADD_READ_BIT(master_slaveAddress);
         TWI0.MADDR = readAddress;
      }

      else if (master_bytesToWrite == 0 && master_bytesToRead == 0) {
         twi_mode = TWI_MODE_MASTER_TRANSMIT;
         uint8_t writeAddress = ADD_WRITE_BIT(master_slaveAddress);
         TWI0.MADDR = writeAddress;
      }

      /* Arduino requires blocking function */
      while(master_result == TWIM_RESULT_UNKNOWN) {}

      // in case of arbitration lost, retry sending
      if (master_result == TWIM_RESULT_ARBITRATION_LOST) {
         //~ uint8_t ok = 0;
         //~ bool was = LOW;
         //~ while (ok < 10)
         //~ {
            //~ if ( digitalRead(SCL) == HIGH )
            //~ {
               //~ if(was == HIGH)
                  //~ ok ++;
               //~ was = HIGH;
            //~ }
            //~ else
            //~ {
               //~ was = LOW;
               //~ ok = 0;
            //~ }
         //~ }
         goto trigger_action;
      }

      if (master_bytesToRead > 0) {
         // return bytes really read
         ret = master_bytesRead;
      } else {
         // return 0 if success, >0 otherwise
         ret = (master_result == TWIM_RESULT_OK ? 0 : 1);
      }
   } else {
      ret = 1;
   }

   if(mode == TWI_MODE_SLAVE) //revert back to slave mode if that's how we entered
   {
      twi_mode = TWI_MODE_UNKNOWN;
      TWI_SlaveInit(TWI0.SADDR>>1,TWI0.SADDR &0X01,(uint8_t)TWI0.SADDRMASK);
   }

   return ret;
}

```

I think I got the bits correct.

SpenceKonde, 2020-08-08:
Nice!

I will see if I have a chance to poke at this myself - I certainly won't
get to actually testing it myself - but I think what I'd ideally like to do
is hack togerther an implementation that would also work on a megaAVR
0-series or AVR DA-series part (see: https://GitHub.com/SpenceKonde/DxCore
- but fair warning - if you start reading that datasheet, lean back so you
don't drool into your keyboard... those are the parts that every AVR
microcontroller wishes it was)

And what, you don't bring your electronics with you on vacation?What's the
point of time off if you can't do something meaningful with it?! When I
visit my parents' place at the beach, I turn my poor mothers living room
table into an electronics bench (my father of course is abetting me
enthusiastically - it runs in the family). I spent a month at the beach and
only touched the ocean once!

Yardie-, 2020-08-09:
I have not drastically changed the code just manipulated stuff that is already there so I if the original code works in the other cores it should just work.
It basically just does a temporary swap to master whilst any master calls are made.
checking and leaving the bus speed as it was or to the default and also leaving the 2 address variables and broadcast values as they were.

the only real issue should be that the default bus speed is set in Wire.h and twi.c doesn't have one which I actually think is a flaw as it is possible to set this to 0 and still act as master with no feedback to debug. 

I will be interested to hear how it goes with other avrs

Yardie-, 2021-03-19:
Hi Spence
Just coming back to this after moving to ESP32-picos for a while. Wow bloat city there. All that memory it's pretty hedonistic. :)

Anyway! 
On reflection I guess the cleanest way to do this is simply just accept that it's one or the other and that's it.
With that in mind do the following
uint8_t i2c_address = 0x99;

void be_master()
{
  Wire.end();
  Wire.begin();
}

void be_slave()
{
  Wire.end();
  Wire.begin(i2c_address);
}

then in the setup just do
Wire.begin(i2c_address);

Then it doesn't matter unless some other master tries to talk to the avr whilst it is a master as well.
The other solution also had that problem anyway.

SpenceKonde, 2021-06-28:
We now have a much more graceful implementation that we should be able to use, (needs a bit more testing and adjustment, though. - see https://github.com/SpenceKonde/DxCore/issues/54 ). This would give us support for multimaster, and for dual mode on the Dx-series parts. Very exciting; I wish I didnt have a mountain of critical issues to dig through before this one!

Yardie-, 2021-06-28:
Wow I just had a quick look at this and this is something Linux kernel developers didn't even tackle.
It is so exciting.

SpenceKonde, 2021-07-17:
Planned for 2.4..n - where n is 1 + the number of emergency fix releases I need to put out for 2.4.0

SpenceKonde, 2021-09-06:
Take the latest github version 2.4.3, for a spin. (the changes are not inthe "released" 2.4.2)

SpenceKonde, 2021-09-12:
Github version of Wire now compiles cleanly with the rewritten code (thanks to @MX682X for the new version of Wire!). Major flash savings too, as promised. Now we need people to see if it works. Note the new tools menu option to select master-or-slave or master-and-slave configuration, the former uses less flash and much less RAM than the latter.

Yardie-, 2021-09-12:
Very timely Spence thanks I have a couple of uses for this and will test it in the next couple of weeks. I think the extra RAM of master-and-slave is worth it when you need it. 
I'll reinstall the GIT version.
Thanks heaps for all your work and  @MX682X

SpenceKonde, 2021-11-29:
This is in essentially it's final form (a memory use improvement is coming sometime likely before the first actual "release")

closed 2021-11-29 by SpenceKonde