SpenceKonde/megaTinyCore #219: Provide secured connection for package file

Rotzbua, 2020-08-05:
Currently the webserver does not support https. 

> http://drazzy.com/package_drazzy.com_index.json

Would be nice to see an secured connection, to avoid sniffing and manipulation by third parties (Free wifi..). 

> https://drazzy.com/package_drazzy.com_index.json

SSL certificates are for free if https://letsencrypt.org/ is used.

Test is possible with  https://www.ssllabs.com/ssltest/ .

Best.

SpenceKonde, 2020-08-05:
See https://github.com/SpenceKonde/ATTinyCore/issues/394

As i said then, I am willing to do this if someone will make it as simple as copy-paste to set up. However, if not, it's basically never going to happen unless it is forced on me: Count how many issues I have open here, and on my other core, and be aware that I have a third core in the works, and an online store, and several personal projects, plus a day job. And learning how to configure my webserver for SSL is BORING, while every one of the things I mentioned except my day job is fascinating and compelling.

closed 2020-08-08 by SpenceKonde

Rotzbua, 2020-08-08:
Sorry for the double issue, but it is **a shame the way security is treated.**

Security is one of the most challenging topics for devs and ops. If you have no motivation for security, stop running a webserver and use another solution.

Another solution is to host it completly with github like digistump does (did) it:
https://raw.githubusercontent.com/digistump/arduino-boards-index/master/package_digistump_index.json

Harsh words but best wishes.

SpenceKonde, 2020-08-08:
Well, like I said, if someone will tell me in very simple, copy-paste terms how to do it, sure I will do it, but I have a huge number of outstanding issues, all of which are more fun to work on than farting around with this crap.