SpenceKonde/megaTinyCore #220: readTempVCC.ino example not working properly

Nbelles, 2020-08-07:
Hey Spence,

I'm trying to read both the temperature sensor and the supply voltage using the main ADC on the ATtiny1614 but it seems to be giving very erratic results. When I run it, I get a very small voltage (< 100mV) for the first supply voltage reading and then every subsequent supply voltage reading is a little bit bigger but still not a valid reading (250mV < 260mV). It seems whenever I make any progress on getting a supply voltage reading that is correct, it makes my temperature reading more incorrect. Would you be able to look over the example and see what is going on with it? Maybe it's just my chips that are having the problem but I've tested it on all the ones I have and they all produce a similar result. 

A couple of related questions:
- Is there a limit to how frequently the temperature sensor and supply voltage measurements can be made?
- Is there a limit to how quickly you can change the analog references and associated registers? (Do I need some delay between when I read from the temperature sensor to the supply voltage?)
- Is it possible to request more documentation on using the secondary ADC on the parts that support it?

Thanks for all your great work!

SpenceKonde, 2020-08-07:
I will see if I can reproduce your issue with the sketch for reading Vcc and Temperature. I know it used to work!

1. No.
2. There is always some settling time, but it's on the order of the length of time it takes to take a reading or two - people often just take a couple of readings and throw away the first few. Not able to quickly find the citation for the settling time in he datasheet, you might have better luck. 
3. If there is a specific deficiency in the advanced ADC guide here, please report via issue. a bigger longer guide with more complete example code, no. Expanding that further is not a priority that justifies delaying fixes and releases that people are asking me for. I would suggest that for a much more in-depth description of the operation of the ADC that you refer to wiring_analog.c in the core, the datasheet, and the io header file for that part (to get the names of registers with correct capitalization, spelling, etc)

SpenceKonde, 2020-08-08:
Fixed with b4744ba7cc5b6350e10c4ae7d8cd437f8236bdc8

And discovered that I really ought to have an init delay of non-0 by default, so the first reading on a new reference isn't garbage. 

What probably broke it was the change from ~150 kHz ADC clock to ~1 MHz. Also corrected this sketch to account for the fact that SAMPLEN is set to 14 by the core on startup, to compensate for the faster ADC clock (so the duration of the sampling is about the same as it used to be - only the processing is 5x faster). 

And that, by the way, is why analogRead() is more than twice as fast here than it is on both a classic AVR or the official megaavr core :-)

closed 2020-08-08 by SpenceKonde

Nbelles, 2020-08-08:
Thank you so much. Works great!

Nbelles, 2020-08-08:
Not sure if you’ll still get a notification for this because the issue has been closed, but I was doing some looking through the datasheet and found Section 18.5.2 (VREF) in the ATtiny1614 version that mentions a register ADCn Reference Force Enable which, if I’m reading it correctly, forces the voltage reference to be running even if it has not been requested. Maybe this would help fix the issue with the first reading being garbage. 

I’ll try it out tonight and see if it changes anything and report back. 
[image]

SpenceKonde, 2020-08-08:
Yeah, I know about that - it does work (and I tried it), though it needs a very brief delay to work - you'll notice that although I do still throw away the first value. on readings after the first, I almost don't have to, other than that that function could be pulled out and used without the temp part, by someone who is using a version of megaTinyCore without that change yet... You'll notice the reading only increases by 1-2 LSB. beween first and second reading, that's thanks to INITDLY16 left in place by the temp read (and to match what the core will set by default in 2.0.6). With INITDLY32, the reading doesn't change at all!

It's worth noting that forcing the ADC reference on has two problems- datasheet warns against it if you use an external analog reference, and it wastes a small amount of power, which is likely an issue if you're running on battery in deep sleep most of the time (and running on battery is the usual time that you want to measure Vcc).

Nbelles, 2020-08-08:
Yeah, I feel dumb because just two minutes after I posted about that, I read the comments on your sketch and noticed you had at least tried it and commented it out.

I have been messing around with it and it seems like you have some magic you have in your head that somehow told you to print out the right number of characters that takes the exact right amount of time for the voltage to settle. It seems that if you wait too long you end up with the same problem. I tried adding a really long delay (like 1 second) and it ended up giving the same (garbage) reading as when it was first sampled. But if you take the delay away entirely you end up with the same problem again and it gives basically the same value. It seems as though you have to have some specific amount of delay (not too big and not too small) otherwise you don't get a good reading.

I agree that most people won't want to keep it on all the time. For my current application, it isn't going to be running on battery so I was curious if it would be a viable solution. 

I haven't read up much about the INITDLYx, but would it be silly (in my case) to set that value to be a higher value if I don't care about how quickly I get a reading and instead just want the first reading to be pretty close to accurate?

SpenceKonde, 2020-08-08:
It's covered in the ADC chapter of the datasheet, ex, for the 3216/3217, it's 30.5.4 (page 440) - not that that tells you anything I haven;t said here. One thing you'll never find in the ADC section of a Microchip datasheet is any sort of guidance about how to pick the right setting for a given use case. Actually, that applies to every chapter, but the ADC has a particularly large number of mysterious options, all of which have some impact on accuracy, but... how much? Under what conditions do I need to be concerned? What values should I choose by default, and when should I pick something different? You ain't gonna get that, not in their datasheets. 

You particularly wonder when you see some whack job feature in the datasheet, where you **know** they had some specific use case in mind - but god only knows what...I'd love to hear some Microchip/Atmel engineer telling the backstory of AVR design decisions, which features were put in because of a specific customer needing it, whether they knew the chip they were shipping had a given silicon bug when they shipped that they didn't have time/budget to fix or if it was a surprise when customers complained about it - that kind of thing...

SpenceKonde, 2020-08-08:
Oh, and those options are called (from the io.h

```
typedef enum ADC_INITDLY_enum
{
    ADC_INITDLY_DLY0_gc = (0x00<<5),  /* Delay 0 CLK_ADC cycles */
    ADC_INITDLY_DLY16_gc = (0x01<<5),  /* Delay 16 CLK_ADC cycles */
    ADC_INITDLY_DLY32_gc = (0x02<<5),  /* Delay 32 CLK_ADC cycles */
    ADC_INITDLY_DLY64_gc = (0x03<<5),  /* Delay 64 CLK_ADC cycles */
    ADC_INITDLY_DLY128_gc = (0x04<<5),  /* Delay 128 CLK_ADC cycles */
    ADC_INITDLY_DLY256_gc = (0x05<<5),  /* Delay 256 CLK_ADC cycles */
} ADC_INITDLY_t;
```

SpenceKonde, 2020-08-08:
> I have been messing around with it and it seems like you have some magic you have in your head that somehow told you to print out the right number of characters that takes the exact right amount of time for the voltage to settle

Huh? No (and serial prints are damned close to instant, unless you print so much at a time that you fill the output buffer - serial print gets stuffed into a buffer, and they turn on the DRE interrupt which fires every time a char is done sending to grab the next one... the delay produced by a Serial.print is very small....)

I think you are doing something weird and wrong. 

If you set the `ADC0.CTRLD=ADC_INITDLY_DLY32_gc; //safe to assign rather than |= since nothing else will be set here ` in your sketch, say - under what conditions do you get garbage values? Do you ever get garbage values other than the first one after changing the reference? 

My understanding is that the answer is supposed to be "never". 

Note that unless you pulled from github today, your copy of the core does not touch that register; it will not be in release until 2.0.6.

Nbelles, 2020-08-08:
I just tested it again to make sure I wasn't crazy and sleep deprived last night. If I open the sketch with the changes you made in b4744ba, it works great (less than 100mV of total error). Now, if I just comment out lines 23 and 24 (the two Serial prints), the readings are bad again (200mV to 300mV of total error). Why is that the case? Why do those two lines of Serial printing make such a huge difference?

SpenceKonde, 2020-08-09:
Huh. Well then, it sounds like what they said in the datasheet about 50uS is actually accurate. And counting overhead and inefficiency, stuffing those charachters into the buffer just might take long enough that the 32us initdelay is long enough? See if 64 makes it always work...

Nbelles, 2020-08-09:
It doesn't seem to change much of anything when I change it to `ADC0.CTRLD|=ADC_INITDLY_DLY64_gc;`, still about 200-300mV of error. 

Also, I haven't really been paying much attention to the temperature during these tests but I have noticed it is a good bit higher than reality now (it is saying it is 90+ degrees Fahrenheit while sitting in my 68 degrees Fahrenheit house). I think it used to be pretty close (like within 5 degrees Fahrenheit) but now I'm not so sure. I figured that the factory calibration would help with fixing the offset/gain of the sensor. 

As a side note, could you explain the _gc and _gm naming scheme I've seen you use? I figured out that _bm is bit mask but haven't been able to figure out the others.