SpenceKonde/megaTinyCore #221: typo in pins_arduino.h

Yardie-, 2020-08-13:
Hi Spence
I'm not sure of this 
I was trying to compile some e-paper libraries that are new to me and they threw 
~/Arduino/hardware/megaTinyCore/megaavr/variants/txy6/pins_arduino.h:81:79: note: suggested alternative: 'PIN_SPI_SCK'
no other code throws this but the library is fairly complex and has somehow triggered the error

I'm sort of guessing that 

```
 #ifdef PIN_SPI_SCK_PINSWAP_1
#define SS ((uint8_t) (PORTMUX.CTRLB&PORTMUX_SPI0_bm?PIN_SPI_SS_PINSWAP_1:PIN_SPI_SS))
#define MOSI ((uint8_t) (PORTMUX.CTRLB&PORTMUX_SPI0_bm?PIN_SPI_MOSI_PINSWAP_1:PIN_MOSI_SCK))
#define MISO ((uint8_t) (PORTMUX.CTRLB&PORTMUX_SPI0_bm?PIN_SPI_MISO_PINSWAP_1:PIN_MISO_SCK))
#define SCK ((uint8_t) (PORTMUX.CTRLB&PORTMUX_SPI0_bm?PIN_SPI_SCK_PINSWAP_1:PIN_SPI_SCK))
#else
```
probably should be 

 ```
#ifdef PIN_SPI_SCK_PINSWAP_1
#define SS ((uint8_t) (PORTMUX.CTRLB&PORTMUX_SPI0_bm?PIN_SPI_SS_PINSWAP_1:PIN_SPI_SS))
#define MOSI ((uint8_t) (PORTMUX.CTRLB&PORTMUX_SPI0_bm?PIN_SPI_MOSI_PINSWAP_1:PIN_SPI_MOSI))
#define MISO ((uint8_t) (PORTMUX.CTRLB&PORTMUX_SPI0_bm?PIN_SPI_MISO_PINSWAP_1:PIN_SPI_MISO))
#define SCK ((uint8_t) (PORTMUX.CTRLB&PORTMUX_SPI0_bm?PIN_SPI_SCK_PINSWAP_1:PIN_SPI_SCK))
#else
```
it compiles fine after those changes.

SpenceKonde, 2020-08-14:
Thaaaanks

Yardie-, 2020-08-14:
> Closed #221 via 88745d7b4f151edb3daea1ebdddba69b0d53d755.

No worries.
It was a simple typo easily missed.
I have no idea how the library triggered it. So I can not really claim any credit for it. Haha.

Yardie-, 2020-08-14:
hey just a heads up Spence it is in all the variants

SpenceKonde, 2020-08-15:
Oh krutz

Yardie-, 2020-08-15:
LoL

SpenceKonde, 2020-08-15:
Thanks. That has been broken for a while. The sort of thing that makes you wonder if anyone is using the damned core (though, IIRC, I do recommend using other ways to refer to the pins,) (don't I? I sure HOPE I am, because those macros are ghastly and should be strongly deprecated; they're only in there at all to make sure libraries that rely on them can be used. I wish I could give a deprecated message.... :-/

closed 2020-08-15 by SpenceKonde

Yardie-, 2020-08-15:
Yeah Ok that would be why it triggered it using MOSI as a pin id.
and don't worry they will be using them they are so cool.
They just need someone to write heaps of stupid examples and stick them on hackaday :)
I love them they are easy to use easy to solder into stuff(as long as you put your glasses on).
I already have 20 waiting for me to finish the code. :)
You have sold me.

commit 88745d7b (2020-08-14): Correct names of SPI pin macros

Fixes #221 - not sure how I missed this. I swear I compiled it on 0 and 1 series parts...