SpenceKonde/megaTinyCore #223: TWI begin() does not fully implement swapped pins

Taragorm, 2020-08-24:
In twi.c#63

```
void TWI_MasterInit(uint32_t frequency)
{
	if(twi_mode != TWI_MODE_UNKNOWN) return;

	// Enable pullups just in case, should have external ones though
#ifdef NO_EXTERNAL_I2C_PULLUP
	pinMode(PIN_WIRE_SDA, INPUT_PULLUP);
	pinMode(PIN_WIRE_SCL, INPUT_PULLUP);
#endif
```

...

The pinMode() calls for pullups refer to the default pins, unless you mess with defining macros. This broke my code, as it assigned an ouput back to an input...

[IMO this shouldn't be done anyway, as the WPU is no where near enough, and this has the possibility of making it work unreliably, as opposed to not. That said, I assume it's for Arduino compatibility].

SpenceKonde, 2020-09-28:
Good catch. 

I'm with you 100% on "we shouldn't try to use the wimpy internal pullups asif they were suitable for use as I2C pullups, as, obviously, they're not. Really, I ought to include pads so the user can easily install pullups on the I2C pins. For Rev. C of my dev boards, I'm considering swapping out the UPDI and LED resistors for a single 1206 4-element resistor network (I love those things - fewer parts to place, cheap, and solder very reliably... say I swapped the autoreset resistor for a 4-element 4.7k one. Two in series gets you 10k for the R in the RC that generates the reset pulse, and the other two could be optionally connected to I2C lines with solder bridge jumpers (I don't know if you've seen the latest versions of my breakouts, but I've become very enamored of solder bridge jumpers to add little-used functionality for those who want it.... not only is there the classic reset enable one, there's a default-connected spot where you can reversibly disconnect every place where power can come into Vcc (regulator - for eliminating the quiescent current of a regulator not being used, UPDI 3-pin header, and serial header, plus there's one that connects the XDIR pin of USART0 to the CTS line, so you can use the FTDI header to conveniently connect an RS485 transciever... or if autoreset isn't enabled, there's... DIO I think I called it? that connects the DTR line to an I/O pin too - with that and CTXD bridged, you've got all you need to rig up some sort of flow control using that 6-pin header....

(also as a random aside - next month I'm hopefully going to have a small number of epic serial port adapters for sale - 4-position switch to choose 5v, 3.3v, "auxiliary" (it will ship with blank pads for a regulator, so if you need 2.5 or 1.8v serial, there you go - the last option is the most important though: "target" - whatever voltage the target applies to Vcc, that's the voltage levels it will use. Uses the Holtek serial chip, which has proper VDDIO pin for output levels... Which reminds me - did you see those new DB (and upcoming DD) series parts? Those are the latest "full size" AVR parts - DB's have 128k flash, 16k ram, generally same baller specs as the DA series they released back in april... plus support for external crystal (FINALLY!), 2 or 3 on-chip opamps with software controlled mux's for the inputs and a feedback resistor network to make a programmable gain amplifier. I can't say I've ever wanted to do that, but I'm sure it's very useful :-P - but the thing I'm thinking of is their "MVIO - multivoltage I/O. They have the 4 or 8 pins of PORTC optionally running from the voltage supplied at the designated power pin - those parts have one right in the middle of PORTC on the larger parts; the 28/32 pin ones instead steal PD0 and turn it into the "VDDIO2" pin, but it's like a built in level shifter.... And while looking around on digikey.... I found an LDO where you choose the output voltage by letting a few pins either float or pulling them up or down.... And you could have the MCU do that.... imagine doing "PowerSupply.On(3.3v);" on your 5v AVR128DB, and poof, not only do you have a 3.3v supply, since you tied it to the VDDIO pin, you've also got a port full of pins that can communicate comfortably at 3.3v ).... How the hell did I get that far off track.... 

Anyway,, yeah, I probably should just gun that. entirely, since those pullups are pretty much guaranteed not to work (at least not reliably, which is much worse than if it just didn't work :-P

SpenceKonde, 2020-09-28:
To be clear, I didn't write that; it was like that when we ganked the libraries from the official core.

commit d9248cd3 (2020-10-10): Remove USE_TIMERD0_PWM from boards.txt and NO_EXTERNAL_PULLUPS

Fixes #223,
closed 2020-10-10 by SpenceKonde