SpenceKonde/megaTinyCore #226: We need an SD card library - anyone ported this one yet

SpenceKonde, 2020-09-18:


CommonGrounds, 2020-09-18:
What coincidence, I'm just working with SD modifications, below is a rough version of the modified standard SD library that need to be placed in the hardware megatiny libraries folder ( only left what is needed for Mega tiny in Sd2PinMap.h, I deleted what is unnecessarily ). 

As well as a much modified version of PetitFs [http://elm-chan.org/fsw/ff/00index_p.html](url) (works even on attiny 412 i include some exampes ) but only for 412( spi is on swap(1) and 3216( standard spi swap(0) ) (goes to sketchbook/libraries folder) I also have an interesting lowLatency logger but only for Attiny85 ( using USI SPI). 

And modified Greiman SdFat that also goes in the hardware megatiny libraries folder. 
And I'm currently working on the full FatFs version [http://elm-chan.org/fsw/ff/00index_e.html](url)

[SD.zip](https://github.com/SpenceKonde/megaTinyCore/files/5246631/SD.zip)
[PetitFS.zip](https://github.com/SpenceKonde/megaTinyCore/files/5246681/PetitFS.zip)
[SdFat.zip](https://github.com/SpenceKonde/megaTinyCore/files/5246719/SdFat.zip)
[PetitFs_LowLatency_logger.zip](https://github.com/SpenceKonde/megaTinyCore/files/5246737/PetitFs_LowLatency_logger.zip)

SpenceKonde, 2020-09-24:
Aaah, thanks for reminding me that I need to do the SPI thing!

This is wonderful.

SpenceKonde, 2020-10-09:
Can you explain the PetiteFS library? Where should it go - Can it be part of the core, or does it need to be installed separately?

SpenceKonde, 2020-10-09:
Also, @CommonGrounds you say that these libraries have been adapted to support megaTinyCore. The included readme files do not reflect this, can you give a brief description of the changes you made to them and the name which I should credit that to in the library readme files?

SpenceKonde, 2020-10-10:
Okay, I studied it further, that makes sense. 

Code which goes in Sketchbook/Libraries is not my domain. Suggest submitting  a PR with your fixes to the original repos for petitefs and SDFat

closed 2020-10-10 by SpenceKonde

SpenceKonde, 2020-10-12:
Okay I need to get more of a basic grounding in the state of SD libraries on Arduino...

Abutx, 2021-06-03:
> What a coincidence, I'm just working with SD modifications, below is a rough version ( not using standard SD ( SdFat and PetitFs is much better ) of the modified standard SD library that need to be placed in the **sketchbook\hardware\megaTinyCore\megaavr\libraries** folder ( only left what is needed for Mega tiny core in Sd2PinMap.h I deleted from that file everything else ). 
> 
> As well as a some modified code by me from [https://github.com/greiman/PetitFS](url) wrapper of PetitFs original code [http://elm-chan.org/fsw/ff/00index_p.html](url) ( works even on attiny 412 i include some examples ) but only for 412 ( spi is on swap(1) ( MOSI1 PA6 ,MISO1 PA7, SCK PA3, SS PA2, TXD1 PA1 )  and 3216 ( standard spi swap(0) MOSI ,MISO, SCK, SS ) and **goes to sketchbook/libraries folder**. I also have an interesting lowLatency logger - goes to sketchbook folder, but only for Attiny85 ( using USI SPI ). This modified PetitFs library works for Attiny85, Attiny841, Atmega328 and MegaTinyCore (412 and 3216 for now but modifications for other can be included i just have only this 2 ).
> 
> And modified Greiman SdFat so it can work with MegaTinyCore, that also goes in the **sketchbook\hardware\megaTinyCore\megaavr\libraries** folder. 
> 
> I'm currently working on wrapper for the full FatFs version [http://elm-chan.org/fsw/ff/00index_e.html](url)
> 
> SPI bug in megaTinyCore library must be corrected for this to work, if is not implemented yet i include correction for 412 and 3216 )
> .
> [SD.zip](https://github.com/SpenceKonde/megaTinyCore/files/5246631/SD.zip)
> [PetitFS.zip](https://github.com/SpenceKonde/megaTinyCore/files/5246681/PetitFS.zip)
> [SdFat.zip](https://github.com/SpenceKonde/megaTinyCore/files/5246719/SdFat.zip)
> [PetitFs_LowLatency_logger.zip](https://github.com/SpenceKonde/megaTinyCore/files/5246737/PetitFs_LowLatency_logger.zip)
> [SPI_Correction_for_412_and_3216.zip](https://github.com/SpenceKonde/megaTinyCore/files/5246913/SPI_Correction_for_412_and_3216.zip)
> 
> Gnuplot image from  lowLatency logger example ( using PetitFS ) on attiny85 ( using PB5 as analog pin and reset pin in same time ( so it have some noise ), PB0 MISO, PB1 MOSI, PB2 SCK, PB4 SS, PB3 TX serial output ) , it can do 40000 Samples/sec on 8Mhz depending on card you use. I will do MegaTiny compatibile version or somebody can if you liked.
> 
> [image]

@CommonGrounds can you please update the github links they seem not to work properly. Thank uou

technoblogy, 2022-05-31:
Test