SpenceKonde/megaTinyCore #227: ATtiny1614 PA03 write LOW locks chip up with I2C running

SimonMerrett, 2020-09-23:
As usual, I'm clutching at straws here but I thought I'd ask in case there's something you are aware of that might be causing the issues.

**Setup**:
ATtiny1614 @ 10MHz, 3v3, no bod, no millis(), megatinycore v2.05, no serial, no bootloader.
Si7051 temperature sensor attached on I2C PB01 and PB02 with 10k pullups, green led and series 1k resistor attached on PA03 to ground. TWI code based on http://www.technoblogy.com/show?2QYB

**Issue observed**:
Reading from temp sensor works fine normally and I can verify this with an I2C oled attached (and then removed for further tests). I can write the led pin high and turn it on fine - the loop still runs - but if I try to change the led pin to an input after turning the led on, or write it low, the first time through the loop works fine but the second time the microcontroller never returns from attempting to read the Si7051. 

Why on earth would PA03 have anything to do with TWI? All I can think of is that EXTCLK must somehow have been activated but none of the peripherals I have activated use EXTCLK (only TWI in this stripped down case).

All speculation gratefully received.

SpenceKonde, 2020-09-24:
Please post the exact code that reproduces this issue!

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SimonMerrett, 2020-09-24:
```
#include <TinyMegaI2CMaster.h>

#define SI7051      0x40  // Slave address
#define RESET       0xFE  // Reset command
#define TEMP_NOHOLD 0xF3  // Temperature read command

int16_t nowTemp = 0; // store temperature reading

void setup() {
  buzzer_init();          // configure timers and set port pin direction etc
  sensor_init();          // power up the sensor
  delay(100);
  PORTA.DIRSET |= PIN3_bm; // set  LED pin to output
}

void loop() {

  // read the temperature sensor over I2C
  nowTemp = readTemp();

  // beep to signal exiting the temperature sensor reading
  buzzer_activate(true);
  delay(10);
  buzzer_activate(false);
  delay(10);

  // now attempt to flash the LED
  //  PORTA.DIRSET |= PIN3_bm; // set  LED to output
  PORTA.OUTSET |= PIN3_bm; // set high to power LED on
  delay(100);
  PORTA.OUTCLR |= PIN3_bm; // set low to power LED off
  //  PORTA.DIRCLR |= PIN3_bm; // set  LED to input

  // Pause so the beeping isn't too annoying if this works
  delay(4000);
}

void sensor_init(void)
{
  PORTA.DIRSET |= PIN7_bm; // output. To power Si7051 and pull I2C lines up
  PORTA.OUTSET |= PIN7_bm; // set high to power Si7051 and pull I2C lines up
  // initialise the temperature sensor
  TinyMegaI2C.init();
  TinyMegaI2C.start(SI7051, 0);
  TinyMegaI2C.write(RESET);
  TinyMegaI2C.stop();
}

void buzzer_init(void)
{
  TCA0.SINGLE.CTRLA = 0; //disable TCA0 and set divider to 1
  TCA0.SINGLE.CTRLESET = TCA_SINGLE_CMD_RESET_gc | 0x03; //set CMD to RESET to do a hard reset of TCA0 and enable for both halves.
  TCA0.SINGLE.CTRLD = 0; //make sure split mode is off
  TCA0.SINGLE.CMP0 = 0x44;
  TCA0.SINGLE.CTRLA = TCA_SINGLE_CLKSEL_DIV64_gc; // set the clock divisor but don't enable until called from function

  TCA0.SINGLE.CTRLB = TCA_SINGLE_CMP2EN_bm
                      | TCA_SINGLE_WGMODE_FRQ_gc;
  TCA0.SINGLE.EVCTRL &= ~(TCA_SINGLE_CNTEI_bm); // explicitly set the counter to use clock ticks instead of event counts
  PORTB.DIRSET |= PIN2_bm; // output PB2. Buzzer waveform OUT
}

void buzzer_activate(boolean onOff)
{
  if (onOff)
  {
    PORTB.DIRSET |= PIN2_bm; // make pin output
    TCA0.SINGLE.CTRLA |= TCA_SINGLE_ENABLE_bm; // turn buzzer on!
  }
  else
  {
    TCA0.SINGLE.CTRLA &= ~TCA_SINGLE_ENABLE_bm; // turn buzzer off!
    PORTB.DIRCLR |= PIN2_bm; // make pin input
  }
}

int16_t readTemp(void)
{
  uint8_t _a = 0;
  uint8_t _b = 0;

  TinyMegaI2C.start(SI7051, 0);
  TinyMegaI2C.write(TEMP_NOHOLD);
  delay(10); // allow reading to take place
  TinyMegaI2C.restart(SI7051, 2);
  _a = TinyMegaI2C.read();
  _b = TinyMegaI2C.read();
  TinyMegaI2C.stop();

  uint16_t _c = (_a << 8) | _b;
  int16_t numb = static_cast<int16_t>(10.0 * (175.72 * (float)_c / 65536 - 46.85));
  return numb;
};
```

SimonMerrett, 2020-09-24:
The buzzer has no perceivable effect on the presentation of the issue - commenting it out produces the same symptoms but makes it easier to check if anything has changed while not looking at the board. I need/want the buzzer in the final system so its as well to leave the code in but comment it out for testing.

SpenceKonde, 2020-09-24:
I don't suppose it works if you disconnect the LED? If it does, then add more caps to the board, it's a decoupling probem.

Also, got the schematic handy?

SimonMerrett, 2020-09-24:
[image]

I cut the solder jumper to the LED and now it won't even get through the loop once (using the buzzer as confirmation). In order for loop to run now I have to comment out `sensor_init()' as well as `readTemp(void)`. Bizarrely, commenting out all the GPIO writes to PA03 for the LED does not allow the loop to run, as it did when the LED was connected. The Si7051 has been running with a full oled I2C display and gui button inputs and buzzer outputs without issue. It's only when I tried to light the LED that crashes began.

The board is being powered from Arduino Nano 3.3v rail, not the cr2032 in the picture but I am also testing on battery power in case that makes a difference.

SpenceKonde, 2020-09-24:
The heck....

Uhm, I have to ask, why tinymegai2c, rather than Wire? Do we trust that
that library isnt doing something dumb?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

SimonMerrett, 2020-09-24:
Cancel my report of 

> In order for loop to run now I have to comment out sensor_init() as well as readTemp(void). Bizarrely, commenting out all the GPIO writes to PA03 for the LED does not allow the loop to run, as it did when the LED was connected.

... there may have been a P.I.C.N.I.C. with connections. I chose tinymegai2c because it frees up significant memory over Wire and was successful as I built up the rest of my program but I will test wire as I have memory space free at this stage.

will report back. I'm also going to try lowering the I2C bus speed. I was reluctant to drop from 400kHz because of the display but needs must.

SpenceKonde, 2020-09-24:
I can't describe how much it frustrates me that the Arduino API is so goddamned bloated, particularly things like Wire. And that's *after* I changed them to just be a stream, instead of a HardwareI2C - Frankly, all the "arduino API" stuff they did was the product of the mind of a "good programming design" CS guy who had never written embedded C; that's all well and good.... EXCEPT WHEN YOU ARE DESIGNING A PROGRAMMING PLATFORM FOR EMBEDDED SYSTEMS!! But I think honestly from their perspective, they're not, they're designing an educational aid. Did you see the madness they wrought upon the digital I/O on the official megaavr core? (don't go look now, stay on this problem - but keep it in the back of your mind; if you ever realize you've ingested something poisonous and need to induce vomiting post-haste, just give that a read. For a really good time, misspell pinMode as PinMode. It's NOT A COMPILE ERROR IN THE OFFICIAL MEGAAVR CORE because they added an enum named PinMode to contain the valid arguments to pinMode. I spent like an hour trying to figure that out (my core was copied from that - and I hadn't noticed it because I habitually used direct port writes), then another hour ranting about it to anyone who would listen, and then ripped every trace of that awful idea from the core... Some of the decisions the Arduino team has made in the past 5 years or so have been so bad that I'm just like, you know what, I'm not going to make that compatible. Another example is how digitalRead() turns off PWM. Could someone tell me why?! That dates back to the early days, but even from the educational tool perspective, like, why shouldn't someone be able to read their own PWM? Why shouldn't they be able to see for themselves how PWM is just turning a pin on and off?! I mean, granted with how bloody slow digitalRead() is, they might be hard pressed to see it, but still (when I get this board order i'm working on done, and put out a few other fires.... we're getting fastDigitalWrite() and it's friends.

SpenceKonde, 2020-09-24:
I wonder how hard it would be to add a call to Wire to bump up the speed...

SimonMerrett, 2020-09-24:
Tried tinymegai2c at 100kHz and Wire, neither with success.
Tried changing from PA03 through PA04, 05 and 06 with the same unsuccessful result (none of them are connected to anything but are intended to be GPIOs on my board).

SpenceKonde, 2020-09-24:
How about, ah, Wire.setClock()?

SpenceKonde, 2020-09-24:
but that doesn't address your current problem

SpenceKonde, 2020-09-24:
This is also beside the point, but don't use |= for DIRSET and DIRCLR and it's homologs. That results in a read modify write cycle, because those pins aren't in the IO space - and they always read as 0, but are volatile so have to be read anyway.  Just assign to them. Do use |= for setting and clearing bits in the AVRxt parts those are ONLY the VPORT registers and the GPIO registers (on classic AVRs, it's monkey/dartboard which ones are in the IO space. The fastest way to write pins is VPORTx.OUT |= (1<<bit) or &=~(1<<bit), same goes for direction (VPORTx.DIR) and clearing INTFLAGs. (also, VPORTx.IN can be written and acts like the PINx register on classic AVRs - but the normal PORTx.IN doesn't). CBI and SBI are also single clock instructions not two clocks on these, too.

SimonMerrett, 2020-09-24:
Interesting you should mention that. I have just got it running through the loop using Wire but only if I cycle the LED pin like this:
```
  PORTA.DIRSET |= PIN3_bm; // set  LED to output
  PORTA.OUTSET |= PIN3_bm; // set high to power LED on
  delay(100);
  PORTA.OUTCLR |= PIN3_bm; // set low to power LED off
  PORTA.DIRCLR |= PIN3_bm; // set  LED to input
```
every loop. I will now try with your better suggestion.

SpenceKonde, 2020-09-24:
Oh! HERE IS YOUR PROBLEM!!!!

SpenceKonde, 2020-09-24:
"Reading this bit field will return the value of PORTx.DIR"
"Reading this bit field will return the value of PORTx.OUT."
(also added some needless asides above)

SpenceKonde, 2020-09-24:
PORTA.DIRCLR |= (1<<bit) will actually SET ALL BITS OF PORT A TO INPUTS
And the same with the others...

SpenceKonde, 2020-09-24:
Actually wait, that may not actually do anything other than disable the buzzer?

SpenceKonde, 2020-09-24:
But, I suspect that that wacky behavior is may be part of the issue here....

SimonMerrett, 2020-09-24:
but maybe it disabled the power to the Si7051 too...

SpenceKonde, 2020-09-24:
Bam!

SimonMerrett, 2020-09-24:
And if tinymegai2c didn't have some nice timeout feature (doesn't) then it could get locked waiting for the i2c interrupt to fire that never will.

SimonMerrett, 2020-09-24:
Just to reference best practice - I'm now using 
```
  PORTA.OUTSET = PIN3_bm; // set high to power LED on
  delay(100);
  PORTA.OUTCLR = PIN3_bm; // set low to power LED off
```
from https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/DirectPortManipulation.md

So is the VPORT better and what would it look like?
```
  VPORTA.OUT |= PIN3_bm;
  delay(100);
  VPORTA.OUT &= ~PIN3_bm;
```

SpenceKonde, 2020-09-24:
Yup looks right. I actually added a section on that, as it seemed like a good chance to give useful advice.

SimonMerrett, 2020-09-24:
That's brilliant and a helpful addition. The "Getting started..." series of guides for the ATtiny 0 and 1 series do not follow the best practice you have clearly explained. They are still in `|=` mode for `PORTx.OUTSET` etc.

closed 2020-09-24 by SimonMerrett

SpenceKonde, 2020-09-24:
Wait, on official microchip ones? If they ever show OUTCLR |=, you should open ticket with microchip support.

SimonMerrett, 2020-09-24:
Just went and checked and they do not - I was casting erroneous accusations. El Tangas did raise an issue in the Microchip repo in support of using `OUTCLR` but he did not suggest using `|=` with it and they did not accept the suggestion. I must have let this bad habit slip in by assuming OUTCLR can be used as OUTSET or something along those lines